FastLED/FastLED #114: Broken on ATmega8

dwillmore, 2015-01-02:
I use some ATmega8 AVR chips in a UNO board.  When I tried to use FastLED, I got a whole bunch of:
/home/willmore/Arduino/libraries/FastLED/clockless_trinket.h: In member function ‘void ClocklessController<DATA_PIN, T1, T2, T3, RGB_ORDER, XTRA0, FLIP, WAIT_TIME>::showAdjTime(const uint8_t*, int, CRGB&, bool, int) [with unsigned char DATA_PIN = 14u, int T1 = 4, int T2 = 10, int T3 = 6, EOrder RGB_ORDER = GRB, int XTRA0 = 0, bool FLIP = false, int WAIT_TIME = 50]’:
/home/willmore/Arduino/libraries/FastLED/clockless_trinket.h:277: error: impossible constraint in ‘asm’

My first guess was right.  The ports aren't defined for that chip.  I added "|| defined(__AVR_ATmega8__)"  To the line for the mega328 and 168 and all was well.

I'm not sure if that's correct.  Could someone with more knowledge of the AVR hardware review this and submit a fix?

'cause, you know, ATmega8 support is super important. ;)

dwillmore, 2015-01-02:
It would probably be helpful to know I made the change in fastpin_avr.h

focalintent, 2015-04-05:
I'll add this line in - however not having any ATmega8 hardware here to develop/test against I'm not going to call it officially supported.

closed 2015-04-05 by focalintent

dwillmore, 2015-04-05:
That's, cool. Thank you. I just stick a mega8 in an UNO to test it.  If I
had a few more spare, I would send some your way.