FastLED/FastLED #115: using 16 bit timers on ARM

Daniel-dk, 2015-01-05:
I am running Fastled 3.1 on a ATSAM4S4A ( 120MHz  cortex M4 ) with code ported form arduino Due.
One of the *gotchas* from this is that the SAM4S only has 16 bit timers where DUE ( 3X ) has 32 bit.
I made "next_mark" wrap around ( as if it were a 16 bit value ) and then reset the timer after every byte written otherwise the timer would overflow before too long and i would only get a few seconds of LED updates and then updates would stop ( i think the overflowing timer causes some issue ? ( this was done in clockless_arm_sam.h inside showRGBInternal(PixelController<RGB_ORDER> & pixels) method )

<code>
TC_Start(DUE_TIMER,DUE_TIMER_CHANNEL);
			next_mark = (DUE_TIMER_VAL + (TOTAL))% UINT16_MAX ; wrap around
			b = pixels.loadAndScale1();
			writeBits<8+XTRA0>(next_mark, port,b);
<\code>

the timer is running at 60MHz ( 120/2 ) ie. TC_CMR_TCCLKS_TIMER_CLOCK1

Daniel-dk, 2015-01-05:
it seems that was just one of the symptoms, but not the final solution :(
working on it

focalintent, 2015-01-05:
If you have a reference for a place where I can get a dev board for the 4s I can take a swing at fixing it.

Daniel-dk, 2015-01-05:
i got mine from EBV for around $25 :) ( http://www.atmel.com/tools/SAM4S-XPLD.aspx  and http://store.atmel.com/PartDetail.aspx?q=p:10500321#tc:description ), it's almost the same hardware as the 3S so that should also work.
The 4S16C is a larger footprint version of the 4S4A ( our PCB is only 10mm x 100 mm )
[image]

The 4S is also one of the cheapest SEGGER J-Link tools you can get ( if you solder to some test pads it works as a Segger J-link tool like the SAM-ICE )

focalintent, 2015-01-05:
Well, as I'm sure you've discovered, FastLED does enough low level work that "almost the same hardware as" is mostly meaningless ;) 

That said, while I can get some of the hardware to work with, there's at least 2 (if not three) platforms on the "finish porting" list first (the NRF51822 which is mostly there, which works with the rfduino and the STM32 which is used in the sparkcore)

Daniel-dk, 2015-01-05:
I'm gonna keep going on this anyway :)
I have an RFduino somewhere that has only ever been made into a iBeacon as a demo  (it is sad )

Daniel-dk, 2015-01-07:
My Timer issue seems to happen if i have more than 16 LEDs, so I reset the timer and recalculate "next_mark" before writing each byte ( in clockless_arm_sam.h , showRGBInternal method
<code>
		while(pixels.has(1)) {
			pixels.stepDithering();

			#if (FASTLED_ALLOW_INTERRUPTS == 1)
			cli();
			if(DUE_TIMER_VAL > next_mark) {
				if((DUE_TIMER_VAL - next_mark) > ((WAIT_TIME-INTERRUPT_THRESHOLD)*CLKS_PER_US)) { 
					sei(); 
					TC_Stop(DUE_TIMER,DUE_TIMER_CHANNEL); 
					return DUE_TIMER_VAL; 
					}
			}
			#endif
			TC_Start(DUE_TIMER,DUE_TIMER_CHANNEL);
			next_mark = (DUE_TIMER_VAL + (TOTAL));
			writeBits<8+XTRA0>(next_mark, port, b);
			
			TC_Start(DUE_TIMER,DUE_TIMER_CHANNEL);	
			b = pixels.loadAndScale1();
			next_mark = (DUE_TIMER_VAL + (TOTAL)) ; 
			writeBits<8+XTRA0>(next_mark, port,b);
			
			TC_Start(DUE_TIMER,DUE_TIMER_CHANNEL);
			b = pixels.loadAndScale2();
			next_mark = (DUE_TIMER_VAL + (TOTAL));
			writeBits<8+XTRA0>(next_mark, port,b);

			b = pixels.advanceAndLoadAndScale0();
			#if (FASTLED_ALLOW_INTERRUPTS == 1)
			sei();
			#endif
		};
			
		TC_Stop(DUE_TIMER,DUE_TIMER_CHANNEL);
		return DUE_TIMER_VAL;
	}

I also had issues running at 120MHz, (using CLOCK1 for the timer,thus the timer runs at 60MHz) where my assumed timings were off ( way too long )
the "NS()" macro assumes that the timer runs at F_CPU and not at "F_CPU/2" which is what CLOCK1 does, so i had to change "NS(320), NS(320), NS(640)" to "NS(160), NS(160), NS(320)" to get the 320, 320 and 640ns times
also changed the NS() definition in "delay.h"
to 
<code>
	#if (F_CPU < 96000000) // removed the "1" which would be all cases 
	...

did you also have issues when running at higher clock speeds ?

focalintent, 2015-01-07:
No - the NS macros are for converting times in ns to _native_ clock ticks.  If you're using a clock source that's running at a different resolution, that adjustment should be made in the code, not by fucking with the NS macro definition.  

A much older version of the library had an issue with some teensy 3.1's where F_CPU was being reported as 96Mhz but the actual clock speed was really 48Mhz - I've since corrected/worked around those issues, which is why I put the 1 in there for "all cases" - and in fact will be completely removing that code in a new version.

focalintent, 2015-04-05:
Closing this off, as this isn't going to be a hardware platform that I support right now.

closed 2015-04-05 by focalintent

Jorgen-VikingGod, 2018-02-15:
any plans to support the ATSAM4Sxx in future release?
@Daniel-dk any progress on your work?