FastLED/FastLED #116: Green error with Arduino 1.5.8

TheNumenorean, 2015-01-07:
All the colors displayed when using Arduino 1.5.8 and FastLED 2 or 3 on APA104 has green on all the time. Black is green, all colors are adjusted to have a green tint. Newest versions used.

The two below examples are using identical code.

[image]
With arduino 1.5.8

[image]
With 1.0.6

focalintent, 2015-01-07:
What platform are you compiling on (windows, Mac, Linux?).  Also grab the latest 3.1 code - and alternatively try ws2811 and ws2812 as LED options.  The apa104 timings may be slightly off.

TheNumenorean, 2015-01-07:
Win 7 and 8.1, and I had the same issue with ws2812, there wasn't a noticeable difference between the two. The code I tested was downloaded directly from github for both 2 and 3.1, so it is up to date.

focalintent, 2015-01-07:
What about ws2811 - the latest 3.1 branch uses separate timings for ws2811 and ws2812 (and I don't remember off hand which I had apa104 mapped to). 

Also - what are you running these on?  Leonardo?  Uno? Mega? Due?

TheNumenorean, 2015-01-07:
I originally did all my testing with a mega, and it has the same issues with ws2811 and ws2812. However, I grabbed an uno and it does not have any of the same problems, and appears to work fine under any condition.

TheNumenorean, 2015-01-07:
Also, I believe APA104 is 800khz

focalintent, 2015-01-07:
There's more to the protocol than just its data rate - which mega were you trying it on?  And what pin on the mega?

TheNumenorean, 2015-01-07:
That was just all I remembered from my read-through of FastLED.h. 2560 R2, and I'm using pin 42.

focalintent, 2015-01-07:
Ah - a stupid number of the mega pins use GPIO ports that are not accessible w/the in/out opcodes, which messes with the timing for writing to them.  I bet there's something specific off with the mega timing as a result.  I'll have to get a hold of a mega 2560 here to actually test it out and work out what's happening w/the compiler & timing issue.

focalintent, 2015-04-05:
This appears to only affect pins on PORT L - 42,43,44,45,46,47,48, and 49 - and I don't have a good way to work around that, so I would say don't use those pins.

closed 2015-04-05 by focalintent

NicoHood, 2015-04-05:
Is this documented in the wiki?

focalintent, 2015-04-05:
Given that I just discovered this right now, I haven't updated the wiki yet, but I have updated the README file for the FastLED3.1 branch.