FastLED/FastLED #118: millis drift on AVR when 'microsTaken' < 1024

kriegsman, 2015-01-13:
This is a placeholder to discuss whether we want to do anything about the MS_COUNTER drift that occurs when NUM_LEDS is small, resulting in microsTaken being less than '1024'.  The code currently increments MS_COUNTER by one in this case, which makes the millis() clock run up to about 20-25% fast.  

There are already existing workarounds, and I have an idea for how to make the whole thing a bit more precise, but I want to discuss it first.  At some point.  And if we don't make code change, let's at least document the situation and give a couple of workarounds (e.g., lie to the system and tell it you have 100 LEDs).

Also, in the clockless_trinket code, we MS_COUNTER += microsTaken / 1024, but in all the other clockless code, we MS_COUNTER += microsTaken / 1000.  Am I correct in assuming this is to avoid the cost and size of doing division (by 1000) on the ATtiny, vs the cost and size of a simple shift?

focalintent, 2015-01-13:
I don't want to do anything about this - yes there are games I could start playing with fractional accumulators and such, but I'd rather suggest to people that they get an rtc or move to apa102/lpd8806 :) - and yes the ATTiny is done to avoid the divide, as much to keep div code from being compiled in as the time involved. (That was done when I was trying to

focalintent, 2015-01-13:
Done when I was trying to greatly reduce code size on the ATTiny.

kriegsman, 2015-01-13:
OK.  And I'm still noodling on this.

My thinking goes like this: a two-byte error accumulator would do the trick, as you point out. (Could probably make it be a single byte accumulator in units of 4micros each or 16micros each or something -- but lets assume that we're talking about using up two bytes of SRAM.)

I'm not sure that I want to tell people (or have to myself) to go buy, connect, configure, poll, and generally deal with a hardware realtime clock, or to go buy all new LEDs -- which may not be available (yet) in the form factor they want -- when using two bytes of RAM would make the clock run much closer to correct.

Simplest workaround in code is just tell FastLED that you have (roughly) a hundred LEDs, and the timing works out much better and the clock runs much more like "right".  The problem there is that if you have (say) 24 real LEDs, then this workaround is costing you an additional 228 bytes of RAM -- which you may not have if you're on an ATtiny to start with anyway.

kriegsman, 2015-01-13:
Actually, if we have a two byte accumulator whose units are 4micros, then the code looks something like this:
  acc4us += (microsTaken >> 2);
  MS_COUNTER += acc4us.highbyte;
  acc4us.highbyte = 0;

Since the microsTaken computation has a "24 * " term in it, we could actually reduce that to "6 *" and not have to do the "/ 4" bit shift afterwards.  I'd rant to comment the hell out of that though, or at the very least express it as "((24) / 4) * ..." because otherwise the meaning gets deeply lost.

kriegsman, 2015-01-13:
Maybe I'll play around, try it, and see what it actually does to code size, clock behavior, etc.  I'll report back here if I actually get around to that.

kriegsman, 2015-01-15:
Improved it some.  It's sort of within a percent or two now.  Closing.

closed 2015-01-15 by kriegsman