FastLED/FastLED #119: Support ESP8266 / Xtensa LX3

kriegsman, 2015-01-13:
The ESP8266 is a very inexpensive ($5) wifi-enabled SOC with an Xtensa LX3 CPU core (read: it's NOT ARM, and it's NOT AVR).  It's typically provided pre-flashed with code that makes it act like a Serial-to-Wifi bridge with an AT command set for Arduinos to talk to, but people are starting to realize that they can reflash it with their own code and run applications directly on it.

http://esp8266.net/
http://hackaday.com/tag/esp8266/

"Peter Scargill" has been doing a TON of work on it 
  https://plus.google.com/+PeterScargill/posts 
  http://scargill.wordpress.com/category/esp8266/
and he's (now) a FastLED user, too:
  http://scargill.wordpress.com/2015/01/13/esp8266-temporary-rgb-solution/

Anyway, low priority, might be "fun."  I like the zero-overhead LOOP option that the LX3 has. Woo.

kriegsman, 2015-01-13:
Sample hardware on the way.

focalintent, 2015-01-13:
Yeah, I've been watching the progress on this one for a while, now.  It's pretty low on my radar, mostly because I still have 3 other platforms higher up on the list to get through (NRF51822, STM32, LPC810).

Jorgen-VikingGod, 2015-03-22:
@focalintent these nRF51822 are awesome - I just order some from china - looking forward to play with.
Also ordered some ESP8266-07 and ESP8266-12

flokli, 2015-05-03:
Is there any progress on ESP8266 support in the FastLED library? Would really love it!

kriegsman, 2015-05-03:
Thanks for the inquiry. The ESP8266 is a nifty little chip.  We probably won't start any work on porting to it until the tool chain settles down a little. It's getting closer, but it's not there yet. 

In the meantime, you can totally use it as a wifi serial gateway, with your LED code running on any supported AVR or ARM based chip. 

Believe me, when we start work on the ESP8266, you'll know! (And also, if you're someone whose families with  Xtensa assembly language and their SOC architecture, by all means, you're welcome to start the port yourself!)

Jorgen-VikingGod, 2015-05-04:
@kriegsman You're right. I hooking the ESP8266 to my teensy 3.1 and it works perfect. Running FastLED directly on ESP8266 will not bring as good performance as expected (I think)...

flokli, 2015-05-07:
@kriegsman, @Jorgen-VikingGod: It seems like the esp is fast enough to drive the ws2812. However, it is a little bit tricky to get the timing correct.

There is already [some code](https://github.com/nodemcu/nodemcu-firmware/blob/master/app/modules/ws2812.c) for the nodemcu-firmware to drive the ws2812, which however uses loops to do the waiting.

This could be changed to assembler instructions to be safer.

kriegsman, 2015-05-07:
Yes, the chip is capable of it. 
It's really just a "simple matter of programming"-- which neither Daniel nor I have started on yet, despite our own great interest in it. We both have pretty demanding jobs working for a startup company; we love doing this, but have limited time. 

If someone else (you!) with Xtensa expertise wants to get a true FastLED port up and running, that would be useful. 

We know it's possible, we want to get to it, we'd welcome source code contributions, and other than that, everyone please just be super patient.  And if you get impatient before then, I guess it's time for you to start learning Xtensa LX3 assembly language and help out with the port!

Torxgewinde, 2015-07-06:
Hi, so far I am using the https://github.com/Makuna/NeoPixelBus with good results on ESP8266-12 (NodeMCU hardware,  programmed in Arduino-IDE)

blackketter, 2015-07-08:
Please excuse my ignorance, but wouldn't a non-optimized, pure C, bit-banged port be relatively straightforward?  It would take the Fast out of FastLED, but should provide a good base for some non-timing-dependent LEDs like the APA102.  I've just started looking at this, but have a bit of work to get up to speed...  Any tips would be much appreciated...

kriegsman, 2015-07-08:
That's the plan for Step 1, and we've already done a bunch of work. Right now the hassle is that the underlying SDK is still pretty raw; the latest SDK changes were dropped... this morning. 

So yes, the plan is for initial support for four-wire LEDs first.

blackketter, 2015-07-08:
Awesome.  Has any of the basic work been committed?  I don't see it in the main or 3.1 branch..

kriegsman, 2015-07-08:
Well yes and no. Most recent commits by me were in support of non-AVR, non-ARM platforms like this one. I have some other uncommitted changes that are now obsolete and need retesting because of SDK changes. After that, I do have some initial code that compiles but doesn't work. I really want to get on this-- but it's slow going.

forkineye, 2015-07-22:
I have WS2811 code working in my E1.31 library, pulled from Adafruit's NeoPixel implementation for the ESP8266.  Not FastLED standards, but it works.  Implementation is here if interested - https://github.com/forkineye/E131/tree/master/examples/ESP8266_WS2811.  I noticed that platforms are split out in the 3.1 branch, but I need to get familiar with the FastLED code architecture before I attempt to implement something.

focalintent, 2015-07-22:
There's a lot more that fast led does while writing out the data beyond just raising pins high and low - there's also the dithering and the scaling, and there's also the fact that it does this for a dozen different three wire chipsets, all with their own timing requirements :)

feilimb, 2015-08-22:
Just wondering if any progress has been made on FastLed support for ESP8266 in the past 4 - 6 weeks?  I have the Adafruit Dotstar library working on the ESP8266 via Arduinio IDE with an APA-102 strip of 60 leds, but I would love to see the FastLed library support at some point to give much better control.

focalintent, 2015-08-22:
Not on my end - I have a pretty large list of other work that I'm tackling right now.

HorstBaerbel, 2015-10-21:
Can I vote for this? ;)
Having the ESP8266 supported directly would be sweet, because you wouldn't need an extra Arduino etc. for many projects. The ESP is very capable on its own and already has ADCs/GPIOs etc.

Jorgen-VikingGod, 2015-10-21:
ESP8266 is really powerful SoC with just under 2 USD - having FastLED on it would be the best combination out there!

kriegsman, 2015-10-21:
Just a non-update here: it's "on the list" and we want to do it, but it's not coming soon -- mostly because Daniel and I both have pretty demanding "day jobs" at a startup company. 

Also, it's all Open Source! Anyone can jump in.   So if someone gets it working, we'd definitely appreciate the contribution of their code!  In the meantime, please believe us that it's on the eventual to-do list already.

Diaoul, 2015-11-30:
How much work would this require if based on esp8266/Arduino? Can you give some pointers? I'd like to run my strip directly on the esp rather than mixing esp + arduino but don't have the resource to dig in asm right now.

focalintent, 2015-11-30:
A fair bit of work - there's getting the low level gpio stuff going, the hardware spi access, and then re-creating the timing code for the 3-wire LEDs.   It's going to get done - there's just a bunch that needs to get done (both for this and others).  Esp8266/Arduino doesn't really buy us a whole lot, as FastLED bypasses most of the Arduino libraries anyway.

forkineye, 2015-12-02:
I came up with a way to bend the ESP8266's UART to my will for WS2811 support, thanks to support for logic inversion on the interface.  Code is in my ESPixelStick project - https://github.com/forkineye/ESPixelStick and has been ported into Makuna's NeoPixelBus - https://github.com/Makuna/NeoPixelBus.  Tested up to 720 Pixels from an E1.31 source - https://www.youtube.com/watch?v=perR5NHttQ4 and I'm running 16 modules in my Christmas show.

mkeyno, 2015-12-18:
Thanks forkineye for sharing the code and note , but It seems library its not well configured and in the fork not indicated which file belong the core library and which file belong the  example sketch , however after the compiling the sketch my Arduino (1.6.5) cease to compile at following step

AppData\Local\Temp\build8463375364921779935.tmp\ESPixelStick\bitbang.c.o
AppData\Local\Temp\build8463375364921779935.tmp\ESPixelStick\ESPixelDriver.cpp.o 
AppData\Local\Temp\build8463375364921779935.tmp\ESPixelStick\_E131.cpp.o 
AppData\Local\Temp\build8463375364921779935.tmp/core.a -lm -lgcc -lhal -lphy -lnet80211 -llwip -lwpa -lmain -lpp -lsmartconfig -lwps -lcrypto -Wl,--end-group -L
AppData\Local\Temp\build8463375364921779935.tmp 
I've tried to comments lines for Pixel objects , but not success come out
any help is really appreciated

forkineye, 2015-12-18:
@mkeyno The entire project is meant to be used as a sketch (not a library) for a ESP-01 based board I developed. If you're looking for a library to incorporate into your own sketch, I'd checkout Makuna's NeoPixelBus UartDriven branch - https://github.com/Makuna/NeoPixelBus/tree/UartDriven.  They ported my UART based 2811 code into it.

tablatronix, 2016-01-09:
:+1: 
Is there at least a branch started or a milestone or roadmap of what needs to be done ?

kriegsman, 2016-01-09:
The ESP8266 / Arduino v2.0 code was just released; I'm hoping to take another look now that the platform is stabilizing a little bit more.  There will probably be a branch when that happens, but doesn't exist yet.

lp0, 2016-01-11:
I have a modified version of NeoPixelBus (https://github.com/lp0/NeoPixelBus/blob/UartDriven/NeoPixelEsp8266.c) that:
* Starts by calculating the data for the whole UART1 FIFO buffer and fills it in one go
* Waits for the FIFO buffer to empty to determine when it has finished transmitting

(It's not possible to use the GPIO interface instead UART1 because the OS doesn't work properly if interrupts are disabled for too long: https://github.com/esp8266/Arduino/issues/654#issuecomment-134263799 and when interrupts occur (which happens frequently) the GPIO communication will not be timed correctly.)

kriegsman, 2016-01-11:
Thank you for the notes -- I was just looking at that code and the ensuing notes about interrupts, etc., this morning!    As I've said, this is definitely in the queue for development, but we want to have it properly integrated, with the whole setup: brightness control, color balance, etc., and we want it to be relatively maintainable for us, since we're already supporting a bunch of different CPU architectures and there are only two of us, and we both have day jobs working for a startup...

SO!  So, thank you again for the notes and links!  It will definitely help make things go more quickly when we get there.

focalintent, 2016-01-11:
On platforms with a high enough clock speed, that give me access to the right kinds of cycle counters, I can actually support having interrupts enabled while writing out WS281x data (e.g. the teensy 3.x and the arduino due are two platforms where I do this).  As long as the system is ok with waiting up to 30µs to service an interrupt (which pretty much all of them are) - the flip side to the trick tends to be whether or not the system can do it's interrupt servicing foo in under 5-10µs :)

focalintent, 2016-01-11:
(Also - the downside to the UART solution is I have to make sure that I can get UART timings that work for the dozen or so 3-wire led chipsets that FastLED supports, not just the WS281x -- and, as it looks like someone in that ticket has discovered, you can get away with certain types of timing being off, but the longer the strip gets, the more danger you get in of the timing being off starting to cause corruption :)

lp0, 2016-01-13:
The comment https://github.com/esp8266/Arduino/issues/654#issue-99059617 isn't referring to UART timing, it's the length of time required to communicate with (what looks like) 700+ LEDs over GPIO.

It may be possible to turn interrupts off only for each LED as long as the inter-LED delay isn't too great. I haven't tried to time the interruption that the interrupts cause. There's a cycle count function in https://github.com/adafruit/Adafruit_NeoPixel/blob/master/esp8266.c if you want to implement GPIO support.

focalintent, 2016-01-13:
My comment about UART timing is more about supporting multiple types of 3-wire chipsets (I don't think the UART trick is flexible enough to support all of them) - and less about the interrupt side of things.

On interrupts:

When I ported FastLED to the nrf51822 last year I very quickly became aware of the joys of running in an environment that gets really pissed at you if you don't allow it to handle interrupts for milliseconds at a time :)  The first version of "allow interrupts while writing ws2812 data out" was for the nrf51822 which only has a 16Mhz clock, so I suspect that the esp8266's 80mhz clock will more than handle many cases (and there's workarounds for the places where it doesn't - there may still be a maximum length however because, basically, what I do is between each pixel I enable interrupts, then disable them again, and check how many cycles have passed - if too many have passed then I stop writing the frame, and make 1 (or 2, depending on the platform) re-attempts to write the frame out.  With the nrf51822's bluetooth firmware, I found that for up to a couple hundred leds, immediately writing the frame out a second time would often allow the entire frame to write out before the next interrupt occurred.  

So - where this gets ugly is if there is a _regularly occurring_ interrupt that runs longer than 5µs.  Because that means even the second frame write will also be interrupted if the string of leds gets too long.  Let's say there's an interrupt that fires 30 times a second, and takes 100µs to run.  So, basically, every 33ms this interrupt will hit.  Since we'd be restarting the frame right after the long interrupt happens, we'd have almost the entirety of the 33ms before we get interrupted again.  At 30µs per ws2812 led, that means we could write over 1000 leds before we got hit with a second interrupt - which would effectively mean that we couldn't have any one strand of WS2812's longer than 1000 pixels (at which point you start splitting strips and using multiple pins of output (and yes, I'm planning on seeing whether or not I can also have parallel output going with the esp8266 :).  Of course, if the interrupt runs more frequently, then the maximum usable length of leds drops rapidly.  At 10ms you're capped at just over 300 leds off of a single pin.  At 5ms, more like 150, etc... 

(Earlier this week I got enough information to do the ESP8266 implementation of the 3-wire code with interrupt allowances and supporting all of our current 3-wire chipsets.  I also have the hardware to start playing with this. Now all I need is time - next week I'm on a work business trip - I often get lots of library work done in the evenings in my hotel :)

focalintent, 2016-01-13:
> I haven't tried to time the interruption that the interrupts cause.

This is an important piece of this puzzle as well - and I wanted to highlight it separately.  What made the interrupts on the nrf51822 really really ugly is the bluetooth interrupt handler would often run for _milliseconds_ at a time (seriously, folks?  Interrupts are supposed to be in and out as fast as possible - move data, set flags, that's it - and then something later in "user space" would do any necessary heavy lifting) - which is why I started doing the re-transmit attempt.  Since the nrf51822 was running at 16Mhz, i figured this trick would work on the 16Mhz avr.... 

...which is how I discovered that Arduino's systick interrupt takes more than 5µs to run, and it fires once a millisecond.  So, I could enable interrupts on AVR for the WS2812, but you'd never be able to have a strip longer than 30 leds :)

m0bygit, 2016-01-19:
I have FastLED 3.1 up and running on my ESP8266 12E with a 150 pixel APA102. Wanted to use FL for a project and saw that there was no ESP8266 implementation available so I did a crude port to Sming (https://github.com/SmingHub/Sming, a nice framework built on top of Espressifs SDK btw) myself. The code is using the hardware SPI/APA102 and the default C implementations in FL. Did some quick benchmarks on the ESP8266 running at 160Mhz and I got ~260fps with wifi enabled but idle and ~460FPS with wifi disabled and manually reseting the watchdog. The benchmark is just 10 red pixels running down the strip, called from a software timer in Sming. I decreased the timer interval until i got watchdog resets. I'm no pro at this but I'd like to help out if I can. Thanks for a nice library!

robotiko, 2016-01-19:
Hi @m0bygit,
great!
could you please share the Sming example?
can you make a PR of the FastLed port or leave it in you repo to see it working?

Thanks

alonewolfx2, 2016-01-19:
@m0bygit thanks :) can you share your code with us?

m0bygit, 2016-01-20:
@robotiko @alonewolfx2 Sure, no problem. Although I have to figure out how git/github works and clean up the code a bit before I can share it. :) The SPI speed is currently fixed at a value that works for my setup so I'll try to fix that as well. Hopefully this week. I'll write a message here with the github link when it's all done.

robotiko, 2016-01-20:
@m0bygit 
great!  thank you for that. 
I would suggest that you simply add the content to your github repo.. so we can also review and help there. 
If you want .. you can stop by Sming gitter  (you see it in Sming main page) chat to help you with git and how to do it ;)

Thanks

rfeisel, 2016-01-21:
Very excited to see progress!  Keep it up :)

m0bygit, 2016-01-21:
Ok, so anyone interested can find it at https://github.com/m0bygit/FastLED-ESP8266.git . The DATA_RATE_MHZ macro is now working (the default 26 was way to high for my strip which tops out at about 9). I haven't had any issues with it, but if you do, pop over to the repo and submit an issue. If and when I run into performance issues in my project I'll try to look into doing some ESP8266 specific optimizing. Don't hold your breath though, last time i did assembly-code was mid 90s at university (MIPS btw). ;-)

cyberreefguru, 2016-01-23:
@m0bygit what controller did you use to initialize FastLED with?  NEOPIXEL and WS2812 are not working for me.  In the interest of full disclosure, I'm compiling in Eclipse using the native sdk not Sming.

-Tom

m0bygit, 2016-01-23:
@cyberreefguru I initialize FastLED with:
```c++
FastLED.addLeds<CHIPSET, DATA_PIN, CLOCK_PIN, COLOR_ORDER, DATA_RATE_MHZ(SPI_MHZ)>(_leds, NUM_LEDS).setCorrection( TypicalLEDStrip );
```
Please note that I have only implemented the hardware SPI path here. And I only have APA102 to test with. Maybe we should move this discussion to my github page (https://github.com/m0bygit/FastLED-ESP8266.git)? File an issue and I'll try to help you out.

focalintent, 2016-01-24:
master@head now has full ESP8266 support - it doesn't have hardware SPI support, but it does have the pin access, which allows bit-bang'd spi support to work, and it also has interrupt-allowable 3-wire led/WS2811 support as well.  Which effectively means that the entirety of FastLED's functionality and support is available.  

@m0bygit if you're game, I'd be happy to take a pull for the hardware SPI support.  Honestly - don't worry about optimizing the hardware SPI support - as you've found, even without optimization, it can run too quickly (I just fixed a problem on the photon where the _bitbang_ SPI support was driving APA102's too fast :)

closed 2016-01-24 by focalintent

m0bygit, 2016-01-24:
@focalintent Sure thing, i'll adapt it to your changes and make a pullrequest asap. I was thinking more about esp8266 specific lib8tion optimization, but i hope i wont have to running at 160Mhz. :)

tablatronix, 2016-01-24:
Works!
I had to rearrange demoreel a bit to get it to compile in arduino ide with all the scope order nonsense.

HLK21, 2016-01-24:
I can confirm that Demoreel works after scope reorder with Datapin 13 (NodeMCU ESP-12E) and 60 APA104. Thanks for your hard work!

Jorgen-VikingGod, 2016-01-25:
Thank you guys to add FastLED for ESP8266 :+1:

feilimb, 2016-01-25:
I second that, many thanks to all for the trojan work :+1: I will give this a tryout on my ESP-8266 sometime this week.

HorstBaerbel, 2016-01-25:
Wow! Nice :+1: Thanks! Will give this a shot with some WS2812s when I find time.

Diaoul, 2016-01-25:
I'm eager to test this, any chance you can make a release with it?

focalintent, 2016-01-25:
@Diaoul it needs some more testing before I'll call it a release - you can just grab the code directly from here on github if ou don't want to wait.

tablatronix, 2016-01-25:
Yeah it needs stress and stability testing, I did let it run for most of the day but I only have 8 pixels in dev.

cyberreefguru, 2016-01-25:
Hi all - this may be better for the forums, but...I can't seem to get the new implementation working with some ws2812 LEDs.  I do the basic initialization: FastLED.addLeds<NEOPIXEL, 4>(leds, D7).setCorrection(TypicalLEDStrip); - I have the control signal attached to D7 which appears to be pin 13 on the nodemcu.  Then a command from MQTT I call:
            leds[0] = CRGB::Red;
            leds[1] = CRGB::Black;
            leds[2] = CRGB::Black;
            leds[3] = CRGB::Black;
            FastLED.show();

I'm on a nodemcu runing native arduino lib like code (I'm actually using eclipse).  Maybe I don't understand the nuances but I don't get anything on my LEDs.  Do I have to slow the SPI down?  It previously was working with NeoPixelBus, but I didn't want to port my extensive library from LastLED to "the bus".

m0bygit, 2016-01-25:
@focalintent So, i tried to merge your esp8266 update with my SPI code. Since I'm not using the Arduino dev myself (Sming) I thought I would do a bare Espressif IOT SDK port. That on the other hand meant re-implementing a bunch of Arduino style functions used in FastLED (which are defined in Sming), so I gave up after a while. Sorry about that. If you still want my implementation then I'll happily give you the code somehow.

HLK21, 2016-01-26:
@cyberreefguru 
Try this:
FastLED.addLeds(leds, 13).setCorrection(TypicalLEDStrip);

Actually D7 is just another name for GPIO 13.
https://alexbloggt.com/wp-content/uploads/2015/10/nodemcu_pinout_700.png

cyberreefguru, 2016-01-27:
Sorry - totally operator error; I didn't study the data sheet enough to figure out what pin is what.  Totally working now (with WIFI and MQTT)!

emcniece, 2016-01-27:
Tested demo reel out tonight... mad props!!

https://www.youtube.com/watch?v=IGXInPlnvYc

NL-Splinter, 2016-01-28:
To get Demoreal working I have to 'reorder scope', confirmed by this IDE error: 'WS2811' was not declared in this scope'. But what's to re-order? How can I get the demoreel sketch working in ArduinoIDE1.6.7? Thanks!!!

HLK21, 2016-01-28:
#include "FastLED.h"

FASTLED_USING_NAMESPACE

// FastLED "100-lines-of-code" demo reel, showing just a few 
// of the kinds of animation patterns you can quickly and easily 
// compose using FastLED.  
//
// This example also shows one easy way to define multiple 
// animations patterns and have them automatically rotate.
//
// -Mark Kriegsman, December 2014

#if defined(FASTLED_VERSION) && (FASTLED_VERSION < 3001000)
#warning "Requires FastLED 3.1 or later; check github for latest code."
#endif

#define DATA_PIN    13
//#define CLK_PIN   4
#define LED_TYPE    APA104
#define COLOR_ORDER GRB
#define NUM_LEDS    60
CRGB leds[NUM_LEDS];

#define BRIGHTNESS         200
#define FRAMES_PER_SECOND  120

void setup() {
  delay(3000); // 3 second delay for recovery
  
  // tell FastLED about the LED strip configuration
  FastLED.addLeds<LED_TYPE,DATA_PIN,COLOR_ORDER>(leds, NUM_LEDS).setCorrection(TypicalLEDStrip);
  //FastLED.addLeds<LED_TYPE,DATA_PIN,CLK_PIN,COLOR_ORDER>(leds, NUM_LEDS).setCorrection(TypicalLEDStrip);

  // set master brightness control
  FastLED.setBrightness(BRIGHTNESS);
}
uint8_t gCurrentPatternNumber = 0; // Index number of which pattern is current
uint8_t gHue = 0; // rotating "base color" used by many of the patterns

void rainbow() 
{
  // FastLED's built-in rainbow generator
  fill_rainbow( leds, NUM_LEDS, gHue, 7);
}

void addGlitter( fract8 chanceOfGlitter) 
{
  if( random8() < chanceOfGlitter) {
    leds[ random16(NUM_LEDS) ] += CRGB::White;
  }
}
void rainbowWithGlitter() 
{
  // built-in FastLED rainbow, plus some random sparkly glitter
  rainbow();
  addGlitter(80);
}

void confetti() 
{
  // random colored speckles that blink in and fade smoothly
  fadeToBlackBy( leds, NUM_LEDS, 10);
  int pos = random16(NUM_LEDS);
  leds[pos] += CHSV( gHue + random8(64), 200, 255);
}

void sinelon()
{
  // a colored dot sweeping back and forth, with fading trails
  fadeToBlackBy( leds, NUM_LEDS, 20);
  int pos = beatsin16(13,0,NUM_LEDS);
  leds[pos] += CHSV( gHue, 255, 192);
}

void bpm()
{
  // colored stripes pulsing at a defined Beats-Per-Minute (BPM)
  uint8_t BeatsPerMinute = 62;
  CRGBPalette16 palette = PartyColors_p;
  uint8_t beat = beatsin8( BeatsPerMinute, 64, 255);
  for( int i = 0; i < NUM_LEDS; i++) { //9948
    leds[i] = ColorFromPalette(palette, gHue+(i*2), beat-gHue+(i*10));
  }
}

void juggle() {
  // eight colored dots, weaving in and out of sync with each other
  fadeToBlackBy( leds, NUM_LEDS, 20);
  byte dothue = 0;
  for( int i = 0; i < 8; i++) {
    leds[beatsin16(i+7,0,NUM_LEDS)] |= CHSV(dothue, 200, 255);
    dothue += 32;
  }
}

// List of patterns to cycle through.  Each is defined as a separate function below.
typedef void (*SimplePatternList[])();
SimplePatternList gPatterns = { rainbow, rainbowWithGlitter, confetti, sinelon, juggle, bpm };

#define ARRAY_SIZE(A) (sizeof(A) / sizeof((A)[0]))
void nextPattern()
{
  // add one to the current pattern number, and wrap around at the end
  gCurrentPatternNumber = (gCurrentPatternNumber + 1) % ARRAY_SIZE( gPatterns);
}

  
void loop()
{
  // Call the current pattern function once, updating the 'leds' array
  gPatterns[gCurrentPatternNumber]();

  // send the 'leds' array out to the actual LED strip
  FastLED.show();  
  // insert a delay to keep the framerate modest
  FastLED.delay(1000/FRAMES_PER_SECOND); 

  // do some periodic updates
  EVERY_N_MILLISECONDS( 20 ) { gHue++; } // slowly cycle the "base color" through the rainbow
  EVERY_N_SECONDS( 10 ) { nextPattern(); } // change patterns periodically
}

[30 more replies]