FastLED/FastLED #120: Help with updating of driver library to incorporate new chip UCS1904

LingQL, 2015-01-15:
Hi,

We recently purchased this LED pixel with UCS 1904 chip, according to the supplier from China, they stated that UCS1904 and UCS1903 are compatible. However, we tried with your library and it seems to have issues with UCS1903, e.g we can change colour of the LED, but there is random blinking occurring and random colour switching. Is it possible to include a new library with UCS1904? Comparing both UCS 1903 and UCS1904 datasheet, we realise the difference could be in the "Timing Waveforms", but not sure whether this is the only difference if you were to update the library for it.

Have attached the link to download the UCS1904 datasheet (http://www.icdemi.com/viewManual.aspx?p=manual%2FUCS1904.pdf) Unfortunately the datasheet only comes in Chinese, but i can translate it for you if you need any further information. 
 
Thank you very much for your help in advance!

focalintent, 2015-01-15:
You should try the UCS1903B setting - the UCS1903 was a 400khz data signal setup while the UCS1904 is an 800khz data signal.  

However, the UCS1904 data timings are still slightly different from the UCS1903B timings, so I'm also checking in an update for the UCS1904 on the FastLED3.1 branch.  Please try this branch and let me know whether or not it worked!

LingQL, 2015-01-15:
Thanks for your fast reply! We have tried UCS1903B as well, it has the same issues. In fact we actually tried on all the 3 channel LEDs settings, and none of them seems to be compatible.

focalintent, 2015-01-15:
Grab the newest version of the FastLED3.1 branch - I just added UCS1904 timings as well.

LingQL, 2015-01-15:
Awesome, thanks a lot for your help!

LingQL, 2015-01-16:
Hi,

We tried with the newest version of FastLED3.1 branch, it seems to be compatible with the new library, but with a few glitches. Any suggestion? Have recorded videos to show you the problems( sent to FastLED@noreply.github.com via we transfer: http://we.tl/p2SBrss0DP) For example: 

(1) we run the LED with Arduino based on modifying your Blink mode to make it consistent display of Red/Blue colour.
Problem: At delay(50), blinking occur at shorter intervals. At delay (500), blinking occur at a longer intervals
refer to Video 1: delay(50), display Red
refer to Video 2: delay(500), display Red
refer to Video 3: delay(50), display Blue
refer to Video 4: delay(500), display Blue

Code below:
______________________________________________________________
#include "FastLED.h"
#define NUM_LEDS 2
#define DATA_PIN 3
CRGB leds[NUM_LEDS];

void setup() { 
  FastLED.addLeds<UCS1904, DATA_PIN, RGB>(leds, NUM_LEDS);
}

void loop() { 
  // Turn the LED on, then pause
  leds[0] = CRGB::Red; // we test with Blue as well
  leds[1] = CRGB::Red; 
  FastLED.show();
  delay(500); // we test with 50 and 500
}

_______________________________________________________________

(2) We run the LED with Arduino based on modifying your ColorPalette mode to make it display a hue of different colours only.
Problem: It was running smoothly for about 2 min, after that blinking occur at short intervals
refer to Video 5 

Code below:
______________________________________________________________
#include <FastLED.h>

#define LED_PIN     3
#define NUM_LEDS    2
#define BRIGHTNESS  255
#define LED_TYPE    UCS1904
#define COLOR_ORDER GRB
CRGB leds[NUM_LEDS];

#define UPDATES_PER_SECOND 100

CRGBPalette16 currentPalette;
TBlendType    currentBlending;
extern CRGBPalette16 myRedWhiteBluePalette;
extern const TProgmemPalette16 myRedWhiteBluePalette_p PROGMEM;

void setup() {
  FastLED.addLeds<LED_TYPE, LED_PIN, COLOR_ORDER>(leds, NUM_LEDS).setCorrection( TypicalLEDStrip );
  FastLED.setBrightness(  BRIGHTNESS );
  
  currentPalette = RainbowColors_p;
  currentBlending = BLEND;
}

void loop()
{
  //ChangePalettePeriodically();

  static uint8_t startIndex = 0;
  startIndex = startIndex + 1; /* motion speed */

  FillLEDsFromPaletteColors( startIndex);

  FastLED.show();
  FastLED.delay(1000 / UPDATES_PER_SECOND);
}

void FillLEDsFromPaletteColors( uint8_t colorIndex)
{
  uint8_t brightness = 255;
  
  for( int i = 0; i < NUM_LEDS; i++) {
    leds[i] = ColorFromPalette( currentPalette, colorIndex, brightness, currentBlending);
    colorIndex += 3;
  }
}

void SetupTotallyRandomPalette()
{
  for( int i = 0; i < 16; i++) {
    currentPalette[i] = CHSV( random8(), 255, random8());
  }
}

const TProgmemPalette16 myRedWhiteBluePalette_p PROGMEM =
{
  CRGB::Red,
  CRGB::Gray, // 'white' is too bright compared to red and blue
  CRGB::Blue,
  CRGB::Black,

  CRGB::Red,
  CRGB::Gray,
  CRGB::Blue,
  CRGB::Black,

  CRGB::Red,
  CRGB::Red,
  CRGB::Gray,
  CRGB::Gray,
  CRGB::Blue,
  CRGB::Blue,
  CRGB::Black,
  CRGB::Black
};
______________________________________________________________

focalintent, 2015-01-16:
What hardware are you building for (arduino uno, teensy 3.1, something else?), also what version of the arduino software are you using, and what platform are you building on (windows, os x, linux).

LingQL, 2015-01-16:
Im using arduino 1.0.6, on arduino uno,  OS x platform

focalintent, 2015-01-16:
I'm not entirely sure what's off with things here, and I don't have any UCS1904 strips locally to test.  You can try nudging the values in this line in chipsets.h:

template <uint8_t DATA_PIN, EOrder RGB_ORDER = RGB>
class UCS1904Controller800Khz : public ClocklessController<DATA_PIN, 3 * FMUL, 3 * FMUL, 4 * FMUL, RGB_ORDER> {};

The 3*/3*/4* - maybe try 2/4/4?

focalintent, 2015-04-05:
Closing this due to lack of activity (and as far as I can tell, no one else is using UCS1904s)

closed 2015-04-05 by focalintent