FastLED/FastLED #122: FastLED 3.1 with IntervalTimers on Teensy 3.1

jamescarruthers, 2015-01-16:
I get corrupted output (only the first three LEDs showing out of 24) if I use IntervalTimer on the Teensy platform.

James

focalintent, 2015-01-16:
It takes 30us to write out a ws2812 led, if you are firing an interrupt every 100us, and if your interrupt takes more than 10us to run the library is going to punt writing out the current frame (which is why you still see cut off a frame with higher intervals), hence your only 3 leds showing up. 

You need to make your function called by the timer as lightweight as possible, or switch to an LED chipset that doesn't have issues with being interrupted (apa102, lpd8806), or if you don't mind the memory hit, use FastLED's OctoWS2811 (then your leds have to be on one of the 8 OctoWS2811 pins).

focalintent, 2015-01-16:
Alternatively, If you show me the code you're running on your interval timer I can help you trim down its overhead.

jamescarruthers, 2015-01-16:
Ahhh, of course!

Thanks Dan, I'm being an idiot.

closed 2015-02-10 by jamescarruthers