FastLED/FastLED #124: Is it possible to invert the output?

reconbot, 2015-01-18:
I have a circuit where for voltage difference reasons I need to invert the signal to my LED strip. (I can revert it back but I don't have a lot of room.) Is there a way I can invert it in software to compensate?

kriegsman, 2015-01-18:
We're appy to help answer this over on the Google Plus  group for FastLED; here's a shortcut http://fastled.io/+
This bug tracker is just for library bugs, not general help questions-- we handle those on the Google Plus group. (And spoiler alert: the answer to your question is probably yes!)

See you over on google plus!

closed 2015-01-18 by kriegsman

reconbot, 2015-01-18:
Whoops sorry!

L0laapk3, 2020-07-09:
This is the first result when googling "fastled invert output". It seems to hint that the answer is yes, but I've been struggling to find how to do this in the documentation. Any pointers? :)