FastLED/FastLED #125: Compiling errors for 16u2

NicoHood, 2015-01-21:
Hi, I got the following error with the current FastLED dev branch. 3.1 (multiple times).

/cygdrive/c/Users/Nico/AppData/Local/Temp/ccv0SbHx.s:4892: Error: illegal opcode or for mcu avr35

I am using the current dev version of HoodLoader2 board definition files and HID project 2.1 (als the dev branch).
https://github.com/NicoHood/HoodLoader2/tree/dev
https://github.com/NicoHood/HID/tree/Official-Integration
TL;TR: put all these files into sketchbook/hardware and remote the -dev ending.

I am using IDE 1.6 rc2

FastLED was compiling some time before for the 16u2. I have no chance or time to test it with real LEDs right now. Please also make sure that it works for 8u2,32u2 and at90usb162 as well.

focalintent, 2015-01-21:
The limit that I will be able to do is check compilation - it is unlikely that I will be able to test whether or not it actually runs.  (Also, it will be a little bit before I can even take a quick look at that - but one thing I'd check is make sure that you are referencing a valid pin.)

Also, I'll point out that the library has never hard target/pin definitions for any of the *u2 platforms, that it was compiling for you previously may have been a quirk and/or you may have had definitions for the 16u2 that were never contributed back to this tree and were overwritten when you updated the library.

focalintent, 2015-02-17:
Ok - fixed building.  Clockless chipsets aren't compiling right now - but that may just be because there aren't pin definitions.  I'm not sure that the timings/oeprations would work for it anyway, though.

closed 2015-02-17 by focalintent

focalintent, 2015-02-17:
Ok, even if I define pin definitions, I still get impossible constraints for the clockless chipsets, which makes me think that something I'm doing in the ASM doesn't play nicely with the 16u2's opcode constraints.  Which makes sense, it's pretty tightly coupled to the avr5 platform (vs. the 16u2's avr35).  Still, I'll check in the pin definitions for hoodloader.

NicoHood, 2015-02-17:
For Hoodloader2 you basically need pin 1-7 which is PB1-7. The other pins are not really usable on an Arduino board but on other custom boards like from mattairtech who has the same pinout. But pin 1-7 (with native SPI) would be nice though.

What I don't really understand now: You linked a fix and said that there is still something wrong with the clock?

I will test that today. Thank you for taking the time :)

focalintent, 2015-02-17:
Yes, if you try to build using ws2811 chips you will still get a compile error (problem is with clock less chips).  I'm unlikely to be able to fix that though, so for hood loader you can only run chips like the apa102/ws2801/lpd8806.

NicoHood, 2015-02-17:
That sounds good to me (only have those). But there is no way to fix it or just a not so simple or clear solution yet?

Does it also support ws2801 on any pin? Would you like to add the fast pi mapping? Does the shortcut to spi pins still work here (if you dont enter data and clock pin)?

Will try that out now!

focalintent, 2015-02-17:
There is likely no way to fix it. I was pushing the limits of what I could do with the asm already.  

As for the SPI - tel me what pins the SPI clock/data line are on and I'll turn those on (as long as the 16u2 hardware SPI is configured the same as the 32u4).  

Finally, I've already added the fast pin definitions for d0-d20 as defined on your wiki.

NicoHood, 2015-02-17:
```
D1  - PB1 PCINT1 SCLK
D2  - PB2 PCINT2 MOSI
```

Its pin1 for clock and pin 2 for data it seems. Thanks again. Will test it the weekend then...

focalintent, 2015-02-17:
Ok, checked in with the hardware pins, and it appears to compile.

NicoHood, 2015-02-20:
Downloaded FastLed 3.1 branch today.
Leds are working but the shortcut doesnt (compile).

```
sketch_feb20a.ino: In function 'void setup()':
sketch_feb20a.ino:27:48: error: no matching function for call to 'CFastLED::addLeds(CRGB [20], int)'
sketch_feb20a.ino:27:48: note: candidates are:
In file included from sketch_feb20a.ino:1:0:
C:\Users\Nico\Dropbox\Arduino\sketchbook\libraries\FastLED/FastLED.h:167:129: note: template<ESPIChipsets CHIPSET, unsigned char DATA_PIN, unsigned char CLOCK_PIN, EOrder RGB_ORDER, unsigned char SPI_DATA_RATE> CLEDController& CFastLED::addLeds(CRGB*, int, int)
  template<ESPIChipsets CHIPSET,  uint8_t DATA_PIN, uint8_t CLOCK_PIN, EOrder RGB_ORDER, uint8_t SPI_DATA_RATE > CLEDController &addLeds(struct CRGB *data, int nLedsOrOffset, int nLedsIfOffset = 0) {
                                                                                                                                 ^
C:\Users\Nico\Dropbox\Arduino\sketchbook\libraries\FastLED/FastLED.h:167:129: note:   template argument deduction/substitution failed:
sketch_feb20a.ino:27:48: note:   couldn't deduce template parameter 'CLOCK_PIN'
In file included from sketch_feb20a.ino:1:0:
C:\Users\Nico\Dropbox\Arduino\sketchbook\libraries\FastLED/FastLED.h:179:95: note: template<ESPIChipsets CHIPSET, unsigned char DATA_PIN, unsigned char CLOCK_PIN> static CLEDController& CFastLED::addLeds(CRGB*, int, int)
  template<ESPIChipsets CHIPSET,  uint8_t DATA_PIN, uint8_t CLOCK_PIN > static CLEDController &addLeds(struct CRGB *data, int nLedsOrOffset, int nLedsIfOffset = 0) {
                                                                                               ^
C:\Users\Nico\Dropbox\Arduino\sketchbook\libraries\FastLED/FastLED.h:179:95: note:   template argument deduction/substitution failed:
sketch_feb20a.ino:27:48: note:   couldn't deduce template parameter 'CLOCK_PIN'
In file included from sketch_feb20a.ino:1:0:
C:\Users\Nico\Dropbox\Arduino\sketchbook\libraries\FastLED/FastLED.h:191:113: note: template<ESPIChipsets CHIPSET, unsigned char DATA_PIN, unsigned char CLOCK_PIN, EOrder RGB_ORDER> static CLEDController& CFastLED::addLeds(CRGB*, int, int)
  template<ESPIChipsets CHIPSET,  uint8_t DATA_PIN, uint8_t CLOCK_PIN, EOrder RGB_ORDER > static CLEDController &addLeds(struct CRGB *data, int nLedsOrOffset, int nLedsIfOffset = 0) {
                                                                                                                 ^
C:\Users\Nico\Dropbox\Arduino\sketchbook\libraries\FastLED/FastLED.h:191:113: note:   template argument deduction/substitution failed:
sketch_feb20a.ino:27:48: note:   couldn't deduce template parameter 'CLOCK_PIN'
In file included from sketch_feb20a.ino:1:0:
C:\Users\Nico\Dropbox\Arduino\sketchbook\libraries\FastLED/FastLED.h:238:25: note: template<template<unsigned char DATA_PIN, EOrder RGB_ORDER> class CHIPSET, unsigned char DATA_PIN, EOrder RGB_ORDER> static CLEDController& CFastLED::addLeds(CRGB*, int, int)
  static CLEDController &addLeds(struct CRGB *data, int nLedsOrOffset, int nLedsIfOffset = 0) {
                         ^
C:\Users\Nico\Dropbox\Arduino\sketchbook\libraries\FastLED/FastLED.h:238:25: note:   template argument deduction/substitution failed:
C:\Users\Nico\Dropbox\Arduino\sketchbook\libraries\FastLED/FastLED.h:244:25: note: template<template<unsigned char DATA_PIN, EOrder RGB_ORDER> class CHIPSET, unsigned char DATA_PIN> static CLEDController& CFastLED::addLeds(CRGB*, int, int)
  static CLEDController &addLeds(struct CRGB *data, int nLedsOrOffset, int nLedsIfOffset = 0) {
                         ^
C:\Users\Nico\Dropbox\Arduino\sketchbook\libraries\FastLED/FastLED.h:244:25: note:   template argument deduction/substitution failed:
C:\Users\Nico\Dropbox\Arduino\sketchbook\libraries\FastLED/FastLED.h:250:25: note: template<template<unsigned char DATA_PIN> class CHIPSET, unsigned char DATA_PIN> static CLEDController& CFastLED::addLeds(CRGB*, int, int)
  static CLEDController &addLeds(struct CRGB *data, int nLedsOrOffset, int nLedsIfOffset = 0) {
                         ^
C:\Users\Nico\Dropbox\Arduino\sketchbook\libraries\FastLED/FastLED.h:250:25: note:   template argument deduction/substitution failed:
C:\Users\Nico\Dropbox\Arduino\sketchbook\libraries\FastLED/FastLED.h:286:25: note: template<template<EOrder RGB_ORDER> class CHIPSET, EOrder RGB_ORDER> static CLEDController& CFastLED::addLeds(CRGB*, int, int)
  static CLEDController &addLeds(struct CRGB *data, int nLedsOrOffset, int nLedsIfOffset = 0) {
                         ^
C:\Users\Nico\Dropbox\Arduino\sketchbook\libraries\FastLED/FastLED.h:286:25: note:   template argument deduction/substitution failed:
C:\Users\Nico\Dropbox\Arduino\sketchbook\libraries\FastLED/FastLED.h:292:25: note: template<template<EOrder RGB_ORDER> class CHIPSET> static CLEDController& CFastLED::addLeds(CRGB*, int, int)
  static CLEDController &addLeds(struct CRGB *data, int nLedsOrOffset, int nLedsIfOffset = 0) {
                         ^
C:\Users\Nico\Dropbox\Arduino\sketchbook\libraries\FastLED/FastLED.h:292:25: note:   template argument deduction/substitution failed:
sketch_feb20a.ino:27:48: error: wrong number of template arguments (2, should be 1)
Error compiling.
```

NicoHood, 2015-02-20:
I think you have to add () here, since the && is stronger than the ||
https://github.com/FastLED/FastLED/blob/a6adcd550b48ea09f97eea3385c77bafb5d500e7/platforms/avr/fastpin_avr.h#L69

focalintent, 2015-02-20:
Ok - it is possible that the hardware SPI on the 16u2 is different from the hardware SPI used on the atmega328/32u4, so I'm just going to disable hardware SPI support for the 16u2 and friends for now.  Bit banging should be decent enough for now.

NicoHood, 2015-02-21:
Sure. Thats totally awesome.

Maybe its just a simple bug you've missed. For now its working with bitbang. Due to the usb stuff and Serial bridge (at 2M baud) I am working on, i'd really like to add a FastLed interface to the bridge. But then I need hardware SPI I think.

I cannot reopen the issue. Not sure how you handle your todo list, but as a reminder you could reopen it.

focalintent, 2015-02-21:
Open an issue for hardware SPI on the 16u2

commit 416ce981 (2015-02-17): Fix #125 - building for the 16u2.  Note that the clockless chips don't work on here yet.  Might be an issue of missing pin definitions, however.