FastLED/FastLED #127: Issue Driving NeoPixels When Running on an AVR at 8 Mhz

emed795, 2015-01-30:
When FastLED v3.0.3 is used to drive NeoPixels on an ATmega328P at 8 MHz @ 5V the NeoPixels behave sporadically with only the first few pixels lighting up. The issue appears to be related to the speed the chip is running at as the same code works fine run at 16 MHz.

bkolbe, 2015-01-31:
exactly same here

HironoriAbe, 2015-02-04:
me, too.

focalintent, 2015-04-05:
I don't have any 8Mhz avr setups here that can be easily tested with.  Please retest with the latest FastLED3.1 branch and arduino 1.6.1, and re-open this if there is still an issue.

cloudwalking, 2015-04-22:
I ran the following test program on an Arduino Uno and an 8Mhz Adafruit Gemma. It runs as expected on the Uno, but shows only white at full brightness for all LEDs on the Gemma.

Tested branch FastLED3.1, Arduino 1.6.3; test program: https://gist.github.com/cloudwalking/e03dacd31df63449980d

 (@8ee392ace11da654a99d980abb94fc7a9d568157)

focalintent, 2015-04-22:
Reopened, I'm going to have to get some 8mhz hardware I think, to try and track what is going on.

NicoHood, 2015-04-22:
You can run it ad 16mhz and use:
#include <avr/power.h>

and put this into setup:
clock_prescale_set(clock_div_2),

damonwitz, 2015-04-23:
I'm having the same problem with fire fx sketches.  Only white solid LED.  I'm very new at writing code, how do I tell the Adafruit 5v Trinket to run @ 16mhz?

cloudwalking, 2015-04-23:
Do as NicoHood writes, above. You also need to set your board to Trinket 16Mhz instead of Trinket 8Mhz.

That said, setting my Gemma's clock to 16Mhz did not fix the problem.

damonwitz, 2015-04-23:
Thanks, did anything work?  I have a Trinket Pro, maybe that will work better?

SeanGau, 2015-04-28:
I got the same problem. Is there any solution now?

damonwitz, 2015-04-28:
I got it to work by downloading Arduino v1.0.5. I had read another thread that mentioned FastLED 3.3 has some issues w Arduino later than 1.5.7

focalintent, 2015-05-04:
Those issues were fixed for AVR a while ago on the FastLED 3.1 branch (I believe you are referring to 3.0.3 - there is no 3.3 version yet).

The reason for things not working at 8Mhz is still unknown and something that I need to spend some time digging into.

cloudwalking, 2015-05-04:
I am eagerly awaiting your verdict re: 8Mhz. Thanks for taking a look.

damonwitz, 2015-05-04:
I will try it out and post my results. I sent off the neopixels to a job and will have them back next week. Thanks for all the advice.

liyanage, 2015-06-09:
For what it's worth, when running my board at 8MHz I initially also had problems with Adafruit's NeoPixel library. To make it work at 8MHz, I had to downgrade the speed to 400kHz using the library's NEO_KHZ400 option (instead of NEO_KHZ800).

focalintent, 2015-07-14:
Bunch of fixes checked in for 8Mhz gemma/trinket on the 3.1 branch.

closed 2015-07-14 by focalintent

cloudwalking, 2015-07-14:
You are a true hero.

Diaoul, 2016-11-26:
I'm using FastLED on Arduino Pro Mini 3.3V 8MHz and this fails:

```c
#include <FastLED.h>

#define DATA_PIN  4
#define NUM_LEDS  60

CRGB leds[NUM_LEDS];

void setup() {
  FastLED.addLeds<NEOPIXEL, DATA_PIN>(leds, NUM_LEDS);
}

void loop() {
  fill_solid(leds, NUM_LEDS, CRGB::White);
  FastLED.show();
}
```
Important things to note are:
* Works fine on Arduino Uno
* Works fine for less LEDs (NUM_LEDS < 50)
* Works fine for other colors, only `CRGB::White` is causing me trouble
* Still doesn't work with the trick to change clock to 16MHz

Running `"FastLED version 3.001.003"`.

It was pretty hard to troubleshoot all the way to FastLED. Do you have any suggestions or anything else I could try?

Diaoul, 2016-11-27:
OK, so I've solved this. It was actually really simple: not enough current... Sorry for the trouble