FastLED/FastLED #131: Does not work with RedBear Labs Blend boards

yepher, 2015-02-02:
The library does not work with the RedBear Blend board because the board files and this library both define "BLEND". I propose this library redefines that #define to BLEND_COLOR and NBLEND_COLOR.

## Redbear Lab Blend Board
http://redbearlab.com/blend/

Output from sketch verification:
`````
In file included from  BlinkTM1803.ino:5:
In file included from libraries/24244_cb_personal_lib_FastLED-FastLED3.1/FastLED.h:46:
libraries/24244_cb_personal_lib_FastLED-FastLED3.1/colorutils.h:781:32: error: expected identifier

typedef enum { NOBLEND=0, BLEND=1 } TBlendType;
                               ^

libraries/24244_cb_personal_lib_FastLED-FastLED3.1/colorutils.h:786:49: error: expected expression

                      TBlendType blendType=BLEND);
                                                ^

libraries/24244_cb_personal_lib_FastLED-FastLED3.1/colorutils.h:791:50: error: expected expression

                       TBlendType blendType=BLEND);
                                                 ^

libraries/24244_cb_personal_lib_FastLED-FastLED3.1/colorutils.h:801:50: error: expected expression

                       TBlendType blendType=BLEND);
                                                 ^

libraries/24244_cb_personal_lib_FastLED-FastLED3.1/colorutils.h:828:28: error: expected expression

        TBlendType blendType=BLEND)
                                  ^

`````

### Redbare hardware BLEND #64 
`````
[hardware]$ grep -r BLEND *
blend/variants/Blend/pins_arduino.h:#define BLEND
blend/variants/BlendMicro-16MHz/pins_arduino.h:#define BLEND_MICRO
blend/variants/BlendMicro-16MHz/pins_arduino.h:#define BLEND_MICRO_16MHZ
blend/variants/BlendMicro-8MHz/pins_arduino.h:#define BLEND_MICRO
blend/variants/BlendMicro-8MHz/pins_arduino.h:#define BLEND_MICRO_8MHZ
`````

kriegsman, 2015-02-03:
Let me think about how I want to name this, but I totally agree that BLEND is too generic and pollutes the global namespace unnecessarily.

kriegsman, 2015-05-14:
Changed BLEND to LINEARBLEND in FastLED as of https://github.com/FastLED/FastLED/commit/7e058dcaea741d4705bcfd12bb97df5092282597
Should be good to go now; thanks for the ticket.

closed 2015-05-14 by kriegsman