FastLED/FastLED #132: rgb2hsv_rainbow?

Swap-File, 2015-02-03:
I really like using the HSV Rainbow spectrum included with FastLED for effects, but I need to display RGB Values coming in from a color sensor, which means I need to convert RGB values into the non-standard HSV hue map.

This isn't an optimized solution, but it seems to work well for the moderately boosted (Y1 = 1;) rainbow, and could be modified for use with the strong boosted rainbow.

Are there any plans to include a function like this in the library?

    CHSV rgb2hsv_rainbow(CRGB input_color){

        CHSV output_color;

        uint8_t maximum = max(input_color.r, max(input_color.g, input_color.b));
        uint8_t minimum = min(input_color.r, min(input_color.g, input_color.b));

        output_color.v = maximum;
        uint8_t delta = maximum - minimum;

        output_color.s = (maximum == 0) ? 0 : (255 * delta / maximum);

        if (maximum == minimum) {
            output_color.h = 0;
        }
        else {
            if (maximum == input_color.r){
                //Serial.println("red is max");
                if (input_color.g >= input_color.b){
                    if ((input_color.r - input_color.g) <= (delta / 3)){
                        //Serial.println("32-64");
                        output_color.h = map((input_color.r - input_color.g), delta / 3, 0, 32, 64);
                    }
                    else{
                        //Serial.println("0-32");
                        output_color.h = map((input_color.r - input_color.g), delta, delta / 3, 0, 32);
                    }
                }
                else{
                    //Serial.println("208-255");
                    output_color.h = map((input_color.r - input_color.b), 0, delta, 208, 255);
                }
            }
            else if (maximum == input_color.g){
                //Serial.println("green is max");
                if (input_color.r >= input_color.b){
                    //Serial.println("64-96");
                    output_color.h = map((input_color.g - input_color.r), 0, delta, 64, 96);
                }
                else{
                    if (input_color.g - input_color.b <= delta / 3){
                        //Serial.println("96-128");
                        output_color.h = map((input_color.g - input_color.b), delta, delta / 3, 96, 128);
                    }
                    else{
                        //Serial.println("128-132?");
                        output_color.h = map((input_color.g - input_color.b), delta / 3, delta, 128, 160);
                    }
                }
            }
            else if (maximum == input_color.b){
                //Serial.println("blue is max");
                if (input_color.g > input_color.r){
                    //Serial.println("132?-160");
                    output_color.h = map((input_color.b - input_color.g), -delta / 3, delta, 128, 160);
                }
                else{
                    //Serial.println("160-208");
                    output_color.h = map((input_color.b - input_color.r), delta, 0, 160, 208);
                }
            }
        }

        return output_color;
    }

focalintent, 2015-02-03:
There's a rgb2hsv_approximate in the library already on the FastLED3.1 branch

closed 2015-02-03 by focalintent

Swap-File, 2015-02-03:
That looks like exactly what I was looking for!

Thanks!

Swap-File, 2015-04-21:
I'm having an odd problem with rgb2hsv_approximate.  Here is an example, I'm reading in ASCII Data and converting it from RGB to HSV:

			CRGB color1temp;
			color1temp.r = (ctoi(buffer[0]) * 100) + (ctoi(buffer[1]) * 10) + ctoi(buffer[2]);
			color1temp.g = (ctoi(buffer[3]) * 100) + (ctoi(buffer[4]) * 10) + ctoi(buffer[5]);
			color1temp.b = (ctoi(buffer[6]) * 100) + (ctoi(buffer[7]) * 10) + ctoi(buffer[8]);
			Serial.print("c1rgb ");
			Serial.print(color1temp.r);
			Serial.print(" ");
			Serial.print(color1temp.g);
			Serial.print(" ");
			Serial.println(color1temp.b);
			color1 = rgb2hsv_rainbow(color1temp);
			CHSV temp1 = rgb2hsv_approximate(color1temp);
			Serial.print("c1hsv me ");
			Serial.print(color1.h);
			Serial.print(" ");
			Serial.print(color1.s);
			Serial.print(" ");
			Serial.println(color1.v);
			Serial.print("c1hsv lib ");
			Serial.print(temp1.h);
			Serial.print(" ");
			Serial.print(temp1.s);
			Serial.print(" ");
			Serial.println(temp1.v);
			CRGB color2temp;
			color2temp.r = (ctoi(buffer[9]) * 100) + (ctoi(buffer[10]) * 10) + ctoi(buffer[11]);
			color2temp.g = (ctoi(buffer[12]) * 100) + (ctoi(buffer[13]) * 10) + ctoi(buffer[14]);
			color2temp.b = (ctoi(buffer[15]) * 100) + (ctoi(buffer[16]) * 10) + ctoi(buffer[17]);
			Serial.print("c2rgb ");
			Serial.print(color2temp.r);
			Serial.print(" ");
			Serial.print(color2temp.g);
			Serial.print(" ");
			Serial.println(color2temp.b);
			color2 = rgb2hsv_rainbow(color2temp);
			CHSV temp2 = rgb2hsv_approximate(color1temp);
			Serial.print("c2hsv me ");
			Serial.print(color2.h);
			Serial.print(" ");
			Serial.print(color2.s);
			Serial.print(" ");
			Serial.println(color2.v);
			Serial.print("c2hsv lib ");
			Serial.print(temp2.h);
			Serial.print(" ");
			Serial.print(temp2.s);
			Serial.print(" ");
			Serial.println(temp2.v);

Here is the output for  the HTML color codes FF0000 and FF00FF

c1rgb 255 0 0
c1hsv me 0 255 255
c1hsv lib 0 255 255
c2rgb 255 0 255
c2hsv me 208 255 255
c2hsv lib 0 255 255

Am I missing something obvious?  The output from rgb2hsv_approximate looks wrong for FF00FF, while my function seems to give a visually accurate result.

I tried the latest 3.1 beta with the same results.

focalintent, 2015-04-21:
See the notes on the library function:

```
// rgb2hsv_approximate - recover _approximate_ HSV values from RGB.
//
//   NOTE 1: This function is a long-term work in process; expect
//   results to change slightly over time as this function is
//   refined and improved.
//
//   NOTE 2: This function is most accurate when the input is an
//   RGB color that came from a fully-saturated HSV color to start
//   with.  E.g. CHSV( hue, 255, 255) -> CRGB -> CHSV will give
//   best results.
//
//   NOTE 3: This function is not nearly as fast as HSV-to-RGB.
//   It is provided for those situations when the need for this
//   function cannot be avoided, or when extremely high performance
//   is not needed.
//
//   NOTE 4: Why is this 'only' an "approximation"?
//   Not all RGB colors have HSV equivalents!  For example, there
//   is no HSV value that will ever convert to RGB(255,255,0) using
//   the code provided in this library.   So if you try to
//   convert RGB(255,255,0) 'back' to HSV, you'll necessarily get
//   only an approximation.  Emphasis has been placed on getting
//   the 'hue' as close as usefully possible, but even that's a bit
//   of a challenge.  The 8-bit HSV and 8-bit RGB color spaces
//   are not a "bijection".
//
//   Nevertheless, this function does a pretty good job, particularly
//   at recovering the 'hue' from fully saturated RGB colors that
//   originally came from HSV rainbow colors.  So if you start
//   with CHSV(hue_in,255,255), and convert that to RGB, and then
//   convert it back to HSV using this function, the resulting output
//   hue will either exactly the same, or very close (+/-1).
//   The more desaturated the original RGB color is, the rougher the
//   approximation, and the less accurate the results.
```

Swap-File, 2015-04-21:
Note: If I replace this code in rgb2hsv_approximate:

          h = (HUE_PURPLE + HUE_PINK) / 2;
          h += scale8( r - 128, FIXFRAC8(48,128));

with

          h = map((r - b), 0, 255, 208, 255);

I get to within 1 of the expected hue output for #FF00FF, but I am not sure of what other effects this may have, and I know it is not an optimized solution.

kriegsman, 2015-04-21:
Thanks for the ticket and for helping improve this by finding some problems; as you can see it's a little bit tricky to get right in cases. 
I'll take a look at it when I can, hopefully in the next few days. Thank you again!

kriegsman, 2015-05-14:
Thanks for the input.  Updated rgb2hsv_approximate in https://github.com/FastLED/FastLED/commit/2bc428082854e81d6d88f24a9c243e236a96136f 

If there are further issues with this function, please open a new different ticket so we can track it separately.

NicoHood, 2015-07-16:
The function seems quite complicated. You said its best to use with full saturation.
I am only using sat = 255 every time. I only move the hue and dim the leds down and up again.

I dont know how much overhead the "approximate" takes to also convert 255, 255, 0 successfully. But if there would be a way to specify "I am only using HSV input and want to convert them back" that'd be nice. This could maybe save the "approximate" part. Even if its not 100% accurate I guess.

It would also be nice to be able to add CHSV arrays to FastLED. I am more using HSV values, because they are easier to use with visualizations (in my case). The problem I currently have is I want to make them go up to the beat and then fade down. But converting back and forth is not the best solution and keeping a second copy of the whole strip is also bad for longer strips (doubles ram). Any ideas how to solve this?

The only Idea I currently have is to generate the RGB values before running show(), change the pointer of CRGB leds to the new generated array and then destroy the array again. Not sure if that all worth it...

kriegsman, 2015-07-16:
Hey Nico- the idea of somehow doing an HSV-based implementation is one that we've kicked around for quite a while; as far as I know there's no trivial solution.

For general questions and discussion like this, please use the G+ discussion group OR the "gitter" room at https://gitter.im/FastLED/public; Dan and I really try to keep the github Issue list for just bug reports only.  Re-ask the HSV question in one of those forums, and we can continue the conversation there.

Thanks again for the really active engagement (and for your fascinating work with dual-AVR boards)!

NicoHood, 2015-07-17:
@kriegsman Understood, but meh, dont want to have those accounts.

Thinking about this issue and also this https://github.com/FastLED/FastLED/issues/106 gave me some new ideas.

The problem:
You have a color (RGB or HSV) and want to blink it. What you have to do, is to save the current color (or at least the hue), iterate the brightness and then copy it to the leds array every time.

This blows the ram for RGB leds (7b per led, 1 for brightness) and 5b per HSV led (RGB +hue+ brightness).

The Question is if it would be possible to generate the Led data at runtime and then write it to the strip? Malloc() will not work on longer strips, I know. But wouldnt it be possible to take a HSV or RGB + V color and convert them at runtime (show() function) to pure RGB and then write them to the strip? The other idea would be to support HSV nativ (as addLeds input) or something like RGBV would also work.

An example (more like RGB + Val):
```cpp
#include "FastLED.h"

// How many leds in your strip?
#define NUM_LEDS 9

// Define the array of leds
CRGB leds[NUM_LEDS];
CRGB newLeds[NUM_LEDS];
uint8_t brightness[NUM_LEDS];

void setup() {
  FastLED.addLeds<WS2801, RGB>(leds, NUM_LEDS);

  // Choose a new color for every led
  for (int i = 0; i < NUM_LEDS; i++)
    newLeds[i].setHue(random8());
}

void loop() {
  // Change brightness
  static uint8_t c = 0;
  c++;
  for (int i = 0; i < NUM_LEDS; i++)
    brightness[i] = cos8(c);

  // Update leds
  FastLED_write(newLeds, brightness, NUM_LEDS);

  // Attention with FastLED.delay() !
  delay(5);
}

void FastLED_write(CRGB l[], uint8_t b[], int num) {
  // This is what FastLED could do under the hood
  for (int i = 0; i < num; i++) {
    CRGB color = l[i];
    color %= b[i];
    leds[i] = color;
  }
  FastLED.show();
}
```

kriegsman, 2015-07-17:
You already have a gitter login if you're using github. Please take the conversion over there https://gitter.im/FastLED/public and let's continue it there.