FastLED/FastLED #133: Arduino due with 1.5.8

Cardedon, 2015-02-08:
Using Arduino ide 1.5.8 and Arduino due the library doesn't work.
Rolling back to ide 1.5.6r2 the library work correctly.

focalintent, 2015-02-09:
What version of the library are you using?  You might need to use the 3.1 branch.  Also, what OS are you using?

trunet, 2015-04-05:
Same issue here. It doesn't work with 1.5.7+. FastLED3.1 branch on MacOSX. Works perfectly on 1.5.6r2.

focalintent, 2015-04-05:
Weird - 1.5.8 is what I developed the due support against.   What leds are you trying to work with?  (Since there's nearly 2 dozen types of leds that the library works with)

trunet, 2015-04-05:
I didn't test on 1.5.8, only 1.5.7 and 1.6.1.

I'm using a WS2811 led strip.

focalintent, 2015-04-05:
I need a little bit more information then on how it isn't working - because I'm running a due, with arduino 1.6.1, using WS2812 strips here without any problems at all.  Without a way to reproduce the problem here, there's going to be a limit to what I can do.

trunet, 2015-04-05:
I'll do more tests on monday and I'll come back here. Thanks Daniel...

focalintent, 2015-04-15:
Closing, not enough information given to recreate the issues reported here (for future reference, supplying more info than "doesn't work" as well as sample code for the sketches used would help - I'm not clairvoyant! :)

closed 2015-04-15 by focalintent