FastLED/FastLED #136: FastLED 3.0 + Arduino 1.6.0 + Mega 2560 - red color is on all times (black is red).

artur3, 2015-02-16:
Using CRGB::Black gives red color.
Removing serial port and IR functions does not help.

FastLED 3.0  
Arduino 1.6.0 
Mega 2560

Code:
#include <FastLED.h>
#include "IRLremote.h"

#define LED_PIN     6
#define COLOR_ORDER RGB
#define CHIPSET     WS2812
#define NUM_LEDS    5

#define BRIGHTNESS  200

CRGB leds[NUM_LEDS];

const int interruptIR = digitalPinToInterrupt(21);

uint8_t IRProtocol = 0;
uint16_t IRAddress = 0;
uint32_t IRCommand = 0;

int dot = 0;

unsigned long dalayTimerReload;
unsigned long runTime;

byte bri;
boolean dir;

//*********************************************************************************
//
//*********************************************************************************

void setup() 
{

  dalayTimerReload = 100;
  runTime = 0;

  bri = 0;
  dir = false;

  delay(3000); // sanity delay
  FastLED.addLeds<CHIPSET, LED_PIN, COLOR_ORDER>(leds, NUM_LEDS).setCorrection(TypicalLEDStrip);
  FastLED.setBrightness(BRIGHTNESS);
  FastLED.clear(true);

  Serial.begin(115200);
  Serial.println("Startup");

  // choose your protocol here to reduce flash/ram/performance
  // see readme for more information
  IRLbegin<IR_ALL>(interruptIR);

}

//*********************************************************************************
//
//*********************************************************************************

void loop()
{

  unsigned long delayTimer;
  uint8_t oldSREG;

  oldSREG = SREG;
  cli();
  if(IRProtocol) 
  {
    // print as much as you want in this function
    // see source to terminate what number is for each protocol
    Serial.println();
    Serial.print("Protocol:");
    Serial.println(IRProtocol);
    Serial.print("Address:");
    Serial.println(IRAddress, HEX);
    Serial.print("Command:");
    Serial.println(IRCommand, HEX);
	
    // reset variable to not read the same value twice
    IRProtocol = 0;
  }
  SREG = oldSREG;

  delayTimer = millis() - runTime;
  if(delayTimer >= dalayTimerReload)
  {
    leds[dot] = CRGB::Blue;
    FastLED.show();
    // clear this led for the next time around the loop
    leds[dot] = CRGB::Black;
            
    FastLED.setBrightness(bri);

    if(bri >= 255)
      dir = true;

    if(bri <= 0)
      dir = false;

    if(dir)
      bri--;
    else
      bri++;
      

    dot++;
    if(dot >= NUM_LEDS) 
      dot = 0;

    runTime = millis();
  };

}

focalintent, 2015-02-16:
Do me a quick favor?  Can you try pin 10?  I don't have a mega here to test with right now.

pklopping, 2015-02-16:
Do you think you could trim that code down to a minimal form that still replicates the issue?

focalintent, 2015-02-16:
Actually, @pklopping, what he has for submitted code is fine enough for diagnosing this issue.  (As it has nothing to do with his code, but rather the internal timing code for WS2812 output)

@artur3 Also, now that I've fixed #135 can you also switch over to the FastLED3.1 branch and re-test (both with and without changing to pin 10).

artur3, 2015-02-16:
Pin 10 work's  for FastLED3.0
Now I will try FastLED3.1

Thanks.

pklopping, 2015-02-16:
@focalintent fair enough.

If you have time (which you may not, I understand), could you briefly explain why pin 10 is/was a better option than pin 6? Does the FastLED library require the pin to have an interrupt associated with it?

focalintent, 2015-02-16:
The mega has so many io ports that not all of them can use single cycle I/O registers.  Some of them are more than 1 cycle, most likely 2, but possibly more.  

Pin six is on io port H, which is high enough to have the multi-cycle port write, while pin 10 is on port B, which is low enough to be single cycle.  

So, by asking him to try pin 10 on the mega, I'm able to isolate the problem down to the multi-cycle pin access.  

I did do some tweaking around this code in the 3.1 branch - so I'm hoping that now that 3.1 should compile for the mega again, that will solve the problem.

pklopping, 2015-02-16:
Ah, understood. 

Thank you for the great explanation!

artur3, 2015-02-16:
@focalintent  FastLED3.1 works fine with pin 10. With pin 6 also works but blue dot is barely visible (is not completely off).

focalintent, 2015-02-16:
Comment out the "FastLED.setBrightness(bri)" and see what that does to the brightness of the dot.

artur3, 2015-02-16:
@focalintent  no change for pin 6. 
There is no problem with pin 10.

focalintent, 2015-02-17:
Ok - try the FastLED 3.1 branch now - I found the timing issue with the first bit for hight ports and believe I have fixed it.

closed 2015-02-17 by focalintent