FastLED/FastLED #137: Parallel output support on Arduino

vincpad, 2015-02-18:
Hi,
I'm on a project that requires to display images at a high framerate on a 144 WS2812B strip (for a lightpainting stick).
But writing 144 leds in serial takes a long time (like 10ms), so I divided this strip in four 36 leds strips to drive them in parallel and get a higher framerate, but I cannot find any library that can do this on Arduino (it seems that yours can do this but only on Teensy).
The arduino version of your library currently writes the strips one after the other when calling the "show()" function.
Is it possible to drive multiple strips at the same time on Arduino? 
If so, do you plan to add this feature to your library?

focalintent, 2015-02-18:
I have roughly worked out how to do it, but it'd be a level of asm trickery that would make the current output look sane.  

However, if you want to be doing light painting the 400hz refresh cycle of the ws2812 is going to be terrible for you, I would highly recommend moving to the apa102, which in addition to having a data rate up to 24Mbps (vs 2812's 800kbps, though realistically on avr you'd be limited to 8Mbps) has a pwm refresh rate of 20kH, far far better for POV light painting type things.

focalintent, 2015-04-05:
Note: this is unlikely to ever get implemented in FastLED at this point.

closed 2015-04-05 by focalintent