FastLED/FastLED #138: Doesn't compile right with Arduino 1.6 IDE

patrickday, 2015-02-19:
Anything I compile on 1.6 just lights my TM1803 strip from my Arduino UNO as a blue/cyan color it won't blink or change colors. I tried an older IDE and the same example worked as intended.

focalintent, 2015-02-19:
What OS are you running the arduino ide on?  Also, try using the most recent FastLED3.1 branch (I've already put some fixes on there) and let me know if that helps.

thompson-jeff, 2015-02-27:
I too have the same error colors sort of appear but are very light variations.

patrickday, 2015-02-28:
I'm on Mac Os X 10.10.1. 
It didn't work on 1.6 but worked on 1.0.5

i'll try FastLED3.1

spbolton, 2015-03-01:
I am on windows 8.  Same issue with TM1803.  Failed on 1.6 and 1.59 but worked on 1.0.6.  It failed on 1.6 with FastLED3.1 also

focalintent, 2015-04-05:
Fixed on the 3.1 branch.

commit 7991b046 (2015-04-05): Fix #138 - get TM1803 working again in a post arduino 1.5.8 world.
closed 2015-04-05 by focalintent