FastLED/FastLED #139: Add Hardware SPI for u2 Series (16u2)

NicoHood, 2015-02-21:
Referring to #125 Hardware SPI on 16u2 currently seems not to work.
Bitbang is working fine though.

focalintent, 2015-07-14:
Unlikely to actually support the 16u2 hardware SPI, as the world is moving to arm.  (I will continue to support the major AVR based platforms for a while longer, but the esoteric edge stuff is unlikely - I just don't have the time).

closed 2015-07-14 by focalintent

NicoHood, 2015-09-24:
You may want to add the 8/16/32u2/at90usb82/162 (via HoodLoader2) to the 3.1 release notes. I've only tested Ws2801 but that works great via software spi.

NicoHood, 2015-09-24:
Also a documentation of how to use every n millis functions would be nice. This sounds like something very useful. I just stumbled across this now in the release page.