FastLED/FastLED #140: Setting pixel to CRGB::Black generates blue light

ultdev, 2015-02-22:
I am currently using 4 Pimoroni Unicorn Hats (8x8 NeoPixel matrices), and if I try to use more than two at a time, n-2 panels will always have a blue tinge to all colors displayed (even in the off state (CRGB::Black), the panel will be illuminated blue). This issue is similar to bug #136.

Test Environment:
FastLED 3.0.3, FastLED 3.1 (revision 8ee392ace1)
Arduino IDE 1.0.5, 1.6.0
4 Pimoroni Unicorn Hats
1 5V 10A power supply
1000 uF decoupling capacitor

Arduino sketch is available here: http://pastebin.com/aZgbQ7PT

Workaround: If 2 of the 4 "FastLED.addLeds" statements are commented out, this issue goes away

focalintent, 2015-04-05:
A handful of questions/things to recommend trying (I'm not in a position to exactly test what you're seeing here):

* Move the show out of setup and into loop.  There've been problems before with attempting to do showing from setup and not doing anything in loop.
* Which two pins show blue?  Is it the same two every time or does it change each time you run it?  What if you change the ordering of the pins?

Also there have been a number of changes to FastLED3.1 branch since this was opened, try that out and let me know what you end up getting.

focalintent, 2015-04-15:
Closing this issue as I can't recreate it with my setups here, and the questions for trying to narrow down what's going on remain unanswered.

closed 2015-04-15 by focalintent

ultdev, 2015-04-17:
I just tested this bug with revision 64a2ea7fea; all matrices are showing black now; thanks!