FastLED/FastLED #141: LED corruption with FastLED 3.1 and Teensy audio library

cneuzil, 2015-02-22:
This is a continuation of the issue i posted on the google plus page.

I get WS2812 LED corruption when I combine FastLED 3.1 and the audio library.  I get no corruption when I use FastLED 3.0.  Below is a simple sketch that can show the corruption:

https://github.com/cneuzil/FastLED-audio-debugging/blob/master/FastLED-audio-debugging/FastLED-audio-debugging.ino

Decreasing the frames per second makes the corruption go away.

I'm using Arduino 1.06 on windows along with teensyduino 1.21-test2.

cneuzil, 2015-04-12:
I have found that the following combination works:

Teensy 3.1
Arduino 1.0.6 - windows 7
Teensyduino 1.22
Latest FastLED 3.1 - pulled 2015-04-12
Latest Audio library - pulled 2015-04-12

The corruption has gone away.  

I do get an error compiling with FastLED master branch.  This is the output:

In file included from C:\Program Files (x86)\Arduino\libraries\FastLED/fastspi.h:50:0,
                 from C:\Program Files (x86)\Arduino\libraries\FastLED/FastLED.h:31,
                 from RGBShades_teensy_fft2_BT.ino:40:
C:\Program Files (x86)\Arduino\libraries\FastLED/fastspi_arm_k20.h: In member function 'void ARMHardwareSPIOutput<_DATA_PIN, _CLOCK_PIN, _SPI_CLOCK_DIVIDER, pSPIX>::init()':
C:\Program Files (x86)\Arduino\libraries\FastLED/fastspi_arm_k20.h:194:24: error: 'SPI0' was not declared in this scope
   if((SPI_t*)pSPIX == &SPI0) {

focalintent, 2015-04-12:
Yeah - that's because of a change made in a recent version of Teensyduino that I fixed on 3.1 but not on master (but 3.1 is shortly going to become master).  Is it ok to close this issue, then?

cneuzil, 2015-04-12:
Sounds good to close this issue.  One question, should we be using Arduino 1.6.3 or should we stay with the older 1.06?

Clint

focalintent, 2015-04-12:
If you're usng Teensyduino, I think you have to stay with 1.0.6 - I don't believe Paul has ported that over to the 1.6 branch of arduino yet.

closed 2015-04-12 by focalintent