FastLED/FastLED #142: ARM STM32F

sfranzyshen, 2015-02-27:
Core: ARM 32 Cortex-M3 CPU. Like the STM32F103C8T6 ARM STM32 Development Board

focalintent, 2015-02-27:
There is preliminary stm32 support on the sparkcore branch, as the sparkcore uses the stm32.

evq, 2015-03-10:
@sfranzyshen 

A little while back I made some [additions on top of the sparkcore branch](https://github.com/evq/FastLED/tree/stm32f103) so that it worked with the [Arduino_STM32](https://github.com/rogerclarkmelbourne/Arduino_STM32) project to use these types of dev boards with FastLED. Thanks to the work by Dan, it was easy enough :)

I've tested it with the APA102, P9813, WS2801 and WS2811 LED types with a Maple mini clone, so I'm reasonably confident things are more or less working. Please give it a try and let me know if things work for you.

I'm really excited by these dev boards, it's kind of rediculous how cheap they're going on aliexpress ($4!)

The one thing that I did notice is that when combined, FastLED and Arduino_STM32 have very large sketch sizes, on the order of 68% of program space (74k/108k) for the FirstLight example. By [overwriting the unknown exception handler](https://github.com/evq/Arduino_STM32) I was able to get it down to 24% of program space.

@focalintent 

I would make a PR for my changes, but I think they need some cleanup and streamlining. The Arduino_STM32 project is derived from the old libmaple library, and seems to have completely different register header files etc. In some places I just wholesale copied from SparkCore header files since I couldn't find the corresponding definitions in libmaple. Any direction you could give you would great :)

Evey

hackstage, 2015-05-03:
Beautiful! Just what I was looking for. Would like to see if it is possible to use parallel output with this. Thanks a lot evq. There's a brand new forum for the STM32 over at http://www.stm32duino.com FYI. Couldn't be more excited

Jorgen-VikingGod, 2015-06-22:
How is the progress?
I really want to have support for the STM32F401CDU6
is it hard to move from STM32F103 to the F401 or perhaps the F411

focalintent, 2015-07-14:
Closing this off for now.  Outside of photon support there isn't a lot of stm32 platform work coming up (there's a few other platforms in line before that).

closed 2015-07-14 by focalintent

RicardoHipp, 2016-05-14:
what exactly do you mean with  "By overwriting the unknown exception handler" 
i have the same problem. but with your code i get an error with my <DS3232RTC.h> libary.
with this https://github.com/rogerclarkmelbourne/Arduino_STM32 it works but the compiled code is very very large. i hope you can help me.

evq, 2016-05-16:
https://andriidevel.blogspot.com/2016/05/size-cost-of-c-exception-handling-on.html

Might be worth trying to apply this commit to the origin fork (rogerclarkmelbourne), it wasn't exactly a high LOC change:
https://github.com/evq/Arduino_STM32/commit/14fa24c71103e0fe334a0ed4ec920eb253158dd7

agn0stico, 2016-10-02:
Hi guys!
I noticed that this issue has been closed still a year ago.
Is there any plan to add support for RedBear Duo board (STM32F205 ARM Cortex-M3)?

kolyshkin, 2018-02-15:
https://github.com/FastLED/FastLED/pull/568