FastLED/FastLED #143: Error compiling for Teensy 3.1 with FastLED 3.1, Arduino 1.6.1 (Windows 8.1), TD: 1.21-beta12

jasoncoon, 2015-03-14:
Looks like more merge noise in color.h, line 9:

```
>>>>>>> c0c8f765f1b01655116cc0c95979203ba05c80ef
```

With that fixed, I get the following error.  Can't tell if it's a FastLED or Teensyduino 1.21-beta12 issue.

```
Arduino: 1.6.1 (Windows 8.1), TD: 1.21-beta12, Board: "Teensy 3.1, Serial, 96 MHz (overclock), US English"

Using library SmartMatrix in folder: C:\Users\Jason\Documents\Arduino\libraries\SmartMatrix (1.0.x format)

Using library FastLED in folder: C:\Users\Jason\Documents\Arduino\libraries\FastLED (1.0.x format)

Using library IRremote in folder: C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\IRremote (1.0.x format)

Using library SPI in folder: C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\SPI (1.0.x format)

Using library SD in folder: C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\SD (1.0.x format)

Using library Wire in folder: C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\Wire (1.0.x format)

Using library Time in folder: C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\Time (1.0.x format)

Using library DS1307RTC in folder: C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\DS1307RTC (1.0.x format)

Using library aJson in folder: C:\Users\Jason\Documents\Arduino\libraries\aJson (1.0.x format)

C:\Program Files (x86)\Arduino/hardware/tools/arm/bin/arm-none-eabi-g++ -c -Os -g -Wall -ffunction-sections -fdata-sections -MMD -nostdlib -fno-exceptions -felide-constructors -std=gnu++0x -fno-rtti -mthumb -mcpu=cortex-m4 -D__MK20DX256__ -DTEENSYDUINO=121 -DARDUINO=10601 -DF_CPU=96000000 -DARDUINO_ARCH_AVR -DUSB_SERIAL -DLAYOUT_US_ENGLISH -IC:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy3 -IC:\Users\Jason\Documents\Arduino\libraries\SmartMatrix -IC:\Users\Jason\Documents\Arduino\libraries\FastLED -IC:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\IRremote -IC:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\SPI -IC:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\SD -IC:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\Wire -IC:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\Time -IC:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\DS1307RTC -IC:\Users\Jason\Documents\Arduino\libraries\aJson C:\Users\Jason\AppData\Local\Temp\build1376878341043998385.tmp\Aurora.cpp -o C:\Users\Jason\AppData\Local\Temp\build1376878341043998385.tmp\Aurora.cpp.o 

In file included from Aurora.ino:37:0:

C:\Users\Jason\Documents\Arduino\libraries\FastLED/FastLED.h:12:2: warning: #warning FastLED version 3001000 (Not really a warning, just telling you here.) [-Wcpp]

 #warning FastLED version 3001000  (Not really a warning, just telling you here.)

  ^

In file included from C:\Users\Jason\Documents\Arduino\libraries\FastLED/platforms/arm/k20/fastled_arm_k20.h:8:0,

                 from C:\Users\Jason\Documents\Arduino\libraries\FastLED/platforms.h:8,

                 from C:\Users\Jason\Documents\Arduino\libraries\FastLED/FastLED.h:42,

                 from Aurora.ino:37:

C:\Users\Jason\Documents\Arduino\libraries\FastLED/platforms/arm/k20/fastspi_arm_k20.h: In member function 'void ARMHardwareSPIOutput<_DATA_PIN, _CLOCK_PIN, _SPI_CLOCK_DIVIDER, pSPIX>::init()':

C:\Users\Jason\Documents\Arduino\libraries\FastLED/platforms/arm/k20/fastspi_arm_k20.h:195:24: error: 'SPI0' was not declared in this scope

   if((SPI_t*)pSPIX == &SPI0) {

                        ^

Error compiling.
```

PaulStoffregen, 2015-03-14:
Opps, sorry about this.  A recent change turned "SPI0" (which is too short and too likely to conflict with Arduino naming conventions) to KINETISK_SPI0.

Just add "KINETISK_" in front of "SPI0" and it'll work as before.

jasoncoon, 2015-03-14:
That fixed it, thanks!

closed 2015-04-05 by focalintent