FastLED/FastLED #145: Possible Interference from SoftwareSerial, TinyGPS+ and/or FastLED? (WS2801)

beniroquai, 2015-03-14:
Hello,

I started using your great library to control LEDs via a BUS connection. Everything worked great, but now I had to move from one Strip setup (WS2801) to a multiple array setup. 
I hooked up two strips with 4 WS2801 to the arduino nano 328.

I'm reading GPS Data using the SoftwareSerial and then getting NMEA Data using Tinygps. Everytime the GPS (Neo- 6m) Blinks, the LED-Strip shows weired behaviour. Each of the Strips has its own Clock and Data Pin (Pin 7/8, 5/6) on the arduino. Colours and everything works great, but since I cut the one strip in two pieces I'm running into problems. Is the wiring incorrect? Is there a possible interference comming from the SoftwareSerial Connection running on (3,4) ?

Thank you very much!!

focalintent, 2015-03-14:
Repost this on the g+ group - there's a much wider range of people there who can help you out with what may be a wiring/grounding/power/interference issue than a library issue.  (I'm trying to keep this issues list to things that are specifically bugs with the library itself).

beniroquai, 2015-03-15:
Ok. Thank you! Do you have a resource which tells me the correct wiring for the multi-array approach using thw WS2801? You're providing detailed description for the NEOPixel, but I'm missing the wiring part for others ;)..

beniroquai, 2015-03-15:
Found the Issue: FASTLed works better with AltSerialSot. Alternative Library for Serial Communication!

kriegsman, 2015-03-31:
Closing this as "fixed enough".

closed 2015-03-31 by kriegsman