FastLED/FastLED #146: Problem with TM1803 strip

Boo1098, 2015-03-19:
I just tried using FastLED and when I ran the RGBCalibrate and chose the TM1803 chipset the strip just lights up pink. no variances in the pink at all either

focalintent, 2015-03-19:
What version of the library are you using?

What version of arduino and on which os?

What are you building for (arduino uno, teensy 3, etc?)

Boo1098, 2015-03-19:
Oh sorry
I'm using 3.0.1 of the library.
Arduino 1.6.1 on Windows 8.1 building for Arduino Uno

oakandsage, 2015-03-24:
I've got the same issue.

Versions 3.0.1 and 3.1 of the library. (I'm not sure what version the 3.1 branch is on - the release notes don't seem to be up to date - but I just downloaded the files from github 3/24/2016 at 12:30 AM) I'm also using the TM1803 chipset (the Radio Shack 1M strip). Arduino 1.6.1 on Windows 7 building for Arduino Uno. I was able to run the Radio Shack sample code, which runs a few test patterns, so I know the strip's good, but when I run the blink sample program, the whole strip is colored solid magenta.

I tried plugging the green data wire into pins 6, 7, and A0 in the Uno board (A0 because it's what the Radio Shack demo sketch uses). I set DATA_PIN to 6, 7, and 0 respectively. For pins 6 and 7 I get solid magenta; for pin 0 I get the most recent pattern that the Radio Shack sample code sent.

#define NUM_LEDS 10
#define DATA_PIN 7

CRGB leds[NUM_LEDS];

void setup() {

CRGB leds[NUM_LEDS];

void setup() {
    FastLED.addLeds<TM1803, DATA_PIN, RGB>(leds, NUM_LEDS);
}

void loop() { 
  // Turn the LED on, then pause
  leds[0] = CRGB::Red;
  FastLED.show();
  delay(500);
  // Now turn the LED off, then pause
  leds[0] = CRGB::Black;
  FastLED.show();
  delay(500);
}

Boo1098, 2015-03-24:
I'm using the same strip as you oakandsage

focalintent, 2015-04-05:
I just checked in a fix for the TM1803 on the FastLED3.1 branch.

closed 2015-04-05 by focalintent