FastLED/FastLED #147: HSV && uint8_t ?

rodrigograca31, 2015-03-21:
Hi! 
I have question about this code snippet:
```C
#include "FastLED.h"
CRGB leds[60];
void setup() { FastLED.addLeds<NEOPIXEL, 6>(leds, 60); }
void loop() { 
  static uint8_t hue = 0;
  FastLED.showColor(CHSV(hue++, 255, 255)); 
  delay(10);
}
```

You are using HSV.... 
HSV range for Hue is 0-360 and you and ```uint8_t``` variable that has 8 bits, 0-255.... so it will not work...

focalintent, 2015-03-21:
Fast led uses a 0-255 range for HSV for better performance.

closed 2015-03-21 by focalintent

kriegsman, 2015-03-21:
There's a special wiki page for this topic that explains more about the 255 situation for Hue.
https://github.com/FastLED/FastLED/wiki/FastLED-HSV-Colors

rodrigograca31, 2015-03-21:
Ok, @focalintent @kriegsman now I get it.

I'm sending it values from a web app where I'm using HSV.

**I will need to convert 0-360 to the corresponding 0-255 for the Hue value before sending it?**

NicoHood, 2015-03-21:
http://arduino.cc/en/pmwiki.php?n=Reference/Map

rodrigograca31, 2015-03-21:
Thanks @NicoHood but I think I will do it in the web app (Javascript) instead of in the arduino....