FastLED/FastLED #150: Undefined timer0_millis when using Teensy 2 core and FastLED 3.1

adammhaile, 2015-03-28:
I'm trying to compile 3.1 into the AllPixel firmware which uses a modified Teensy 2.0 core but it fails with this every time:

Firmware.cpp.o: In function `L_9302':
G:\Arduino\libraries\FastLED/platforms/avr/clockless_trinket.h:143: undefined reference to `timer0_millis'
G:\Arduino\libraries\FastLED/platforms/avr/clockless_trinket.h:143: undefined reference to `timer0_millis'
G:\Arduino\libraries\FastLED/platforms/avr/clockless_trinket.h:143: undefined reference to `timer0_millis'
G:\Arduino\libraries\FastLED/platforms/avr/clockless_trinket.h:143: undefined reference to `timer0_millis'
G:\Arduino\libraries\FastLED/platforms/avr/clockless_trinket.h:143: undefined reference to `timer0_millis'
Firmware.cpp.o:G:\Arduino\libraries\FastLED/platforms/avr/clockless_trinket.h:143: more undefined references to `timer0_millis' follow

Which is really weird, because I'm certainly NOT using an ATTiny! 
Why would it even be pulling in clockless_trinket.h at all?

I've tried compiling with the official Teensy 2.0 and ++2.0 cores and same failure. Compiling with the Leonardo or Pro Micro cores work fine, however.

Also, it works fine when I use the Master 3.0.3 branch. So maybe this answers my G+ question of if 3.1 was production ready :P

Any thoughts? Feel like it's pulling in the wrong platform...

focalintent, 2015-03-28:
The problem is your build process isn't defining CORE_TEENSY - which is what is necessary for the timer adjustment code to be using timer0_millis_count (which is what the teensy core uses) instead of timer0_millis (which is what the avr core uses).  

(Note that with the exception of the new platform stuff and some arduino 1.6.1 related issues folks have been using the 3.1 branch for well over 6 months now without issue - including on teensy 2.0/++2.0).

adammhaile, 2015-03-28:
Shouldn't the Teensy core be defining that? Or do I just #define it
somewhere?
Good to know it's been stable for everyone :) Certainly want to get updated
to it.

focalintent, 2015-03-28:
I believe it gets defined by the compiler command line - turn on verbose compiling in the arduino environment and you'll see it there.

adammhaile, 2015-03-28:
Nope:

C:\Program Files (x86)\Arduino\hardware\tools\avr\bin\avr-g++ -c -g -Os
-Wall -fno-exceptions -ffunction-sections -fdata-sections -mmcu=atmega32u4
-DF_CPU=16000000L -MMD -DUSB_VID=0x1D50 -DUSB_PID=0x60AB -DARDUINO=106
-DTEENSYDUINO=118 -felide-constructors -DUSB_SERIAL -DLAYOUT_US_ENGLISH
-IG:\Arduino\hardware\ATUSB_Core\cores\atusb -IG:\Arduino\libraries\FastLED
-IC:\Program Files (x86)\Arduino\libraries\EEPROM
C:\Users\admin\AppData\Local\Temp\build7325498756785215293.tmp\Firmware.cpp
-o
C:\Users\admin\AppData\Local\Temp\build7325498756785215293.tmp\Firmware.cpp.o

Using Arduino 1.0.6 IDE and latest Teensyduino extension.

adammhaile, 2015-03-28:
Weird... if I manually define CORE_TEENSY it fails saying that function is already declared.

In file included from Firmware.ino:2:
G:\Arduino\libraries\FastLED/FastLED.h:12:2: warning: #warning FastLED version 3001000 (Not really a warning, just telling you here.)
In file included from G:\Arduino\hardware\ATUSB_Core\cores\atusb/wiring.h:33,
                 from G:\Arduino\hardware\ATUSB_Core\cores\atusb/WProgram.h:19,
                 from G:\Arduino\hardware\ATUSB_Core\cores\atusb/Arduino.h:1,
                 from G:\Arduino\libraries\FastLED/led_sysdefs.h:32,
                 from G:\Arduino\libraries\FastLED/FastLED.h:34,
                 from Firmware.ino:2:
G:\Arduino\libraries\FastLED/platforms/avr/led_sysdefs_avr.h:31: error: previous declaration of 'volatile long unsigned int timer0_millis_count' with 'C++' linkage
G:\Arduino\hardware\ATUSB_Core\cores\atusb/core_pins.h:1858: error: conflicts with new declaration with 'C' linkage

focalintent, 2015-03-28:
Either way - which timer definition to use is gated by the presence of the define CORE_TEENSY - the fact that you're getting complaints about references to the arduino version of timer tells me that you're missing something to get CORE_TEENSY defined

focalintent, 2015-03-28:
Also the code you pasted is teensyduino1.18 - you know that teensyduino is up to 1.21, right?

adammhaile, 2015-03-28:
oh... that's because the modified core I use us based on 1.18, though I
tried compiling with the official 1.21 and same result

focalintent, 2015-03-28:
You kids and your modified cores.  I'll take a look when I get home.

adammhaile, 2015-03-28:
Apparently CORE_TEENSY is defined in core_id.h from the teensy core... so
it *should* be in there. Just wonder why this works in FastLED 3.0 but not
3.1

adammhaile, 2015-03-28:
LOL... It's only modified so that the USB VID/PID pair is not hard coded so
that  I can set it to the pair we own for the AllPixel

focalintent, 2015-03-28:
Here's some more fun - 3.0 uses the same definitions (and 3.0 uses clockless trinket as well).  I'll be back home in about 90 minutes and can sit down with a compiler then.

adammhaile, 2015-03-28:
Ok... that's REALLY weird. I haven't changed anything since I last built
all this with 3.0 (which still works fine) except for install the updated
teensy core. Thanks for looking!

adammhaile, 2015-03-28:
Out of curiosity, I added the following to the beginning of the main .ino file:

```c
#  if defined(CORE_TEENSY)
#warning "Teensy Loaded"
#endif
```

Sure enough, it printed that warning.
But then I put a similar warning in led_sysdefs_avr.h inside the "#  if defined(CORE_TEENSY)" section:

```c
#  if defined(CORE_TEENSY)
#warning "Teensy Loaded"
extern volatile unsigned long timer0_millis_count;
#    define MS_COUNTER timer0_millis_count
#  else
#warning "NO TEENSY"
extern volatile unsigned long timer0_millis;
#    define MS_COUNTER timer0_millis
#  endif
```

And "NO TEENSY" was printed instead... several times actually.

So that #define seems to be getting lost.. if that's even possible.

closed 2015-04-05 by focalintent

adammhaile, 2015-04-05:
Ahh...Teensy itself broke it. Really weird. Might have to poke around the teensy code and see what has changed... thanks!

adammhaile, 2015-04-06:
Able to compile fine now... weird thing is that it's not from the Teensy
core itself but seems to be from whatever Teensy does to modify the IDE.
The modified core I use is based on Teensy 1.18 (as mentioned previously)
but it stopped working after running the Teensyduino installer again. Kind
of wish it didn't modify so much in the environment.

commit 49e70b7b (2015-04-05): Fix #150 - not sure why the CORE_TEENSY define is getting lost in more recent teensyduino versions.  Work around it anyway.