FastLED/FastLED #152: WS2801 hanging with 3.1

adammhaile, 2015-04-08:
Trying to update the AllPixel firmware to 3.1 and most things work except WS2801. When I configure for that chipset and try to run, the firmware hangs. I have not changed the AllPixel firmware in any way. Is there something that has changed with WS2801 or SPI that might have started to cause issues? Other SPI chipsets seem to work fine...

I wish I could give a better idea of where the problem is happening, but all I'm calling is:

```
pLed = new WS2801Controller<SPI_DATA, SPI_CLOCK, RGB>();
FastLED.addLeds(pLed, _fastLEDs, numLEDs);
```

And then trying to initiate serial communication... it accepts the connection but never returns. Seems to me like either an memory issue or it's using some timers maybe that's screwing with stuff? I don't know... thoughts?

adammhaile, 2015-04-11:
A bit of an update...
I diffed the newer Teensyduino cores with the one I was using. Differences were negligable and I moved them over to my code. So, the real differences between 1.18 and 1.21 must be in the Teensy helper programs... which I'm still using since 1.21 is what I have installed.

I also tried creating a *very* basic sketch that would confirm my suspicions. Here's what I used (note, none of my AllPixel firmware insanity):

```c
#include <FastLED.h>

#define numLEDs 3
CRGB _fastLEDs[numLEDs];

void setup()
{
	Serial.begin(115200);
	Serial.setTimeout(1000);
	delay(5000);

	//This does NOT work
	//FastLED.addLeds<WS2801, SPI_DATA, SPI_CLOCK, RGB>(_fastLEDs, numLEDs);

	//This DOES work
	FastLED.addLeds<LPD8806, SPI_DATA, SPI_CLOCK, RGB>(_fastLEDs, numLEDs);
	
	FastLED.clear();
	FastLED.show();
}

void loop()
{
	Serial.write("Start!\n");
	_fastLEDs[0] = CRGB::Red;
	_fastLEDs[1] = CRGB::Green;
	_fastLEDs[2] = CRGB::Blue;
	FastLED.show();
	Serial.write("On!\n");
	delay(500);
	// Now turn the LED off, then pause
	_fastLEDs[0] = CRGB::Black;
	_fastLEDs[1] = CRGB::Black;
	_fastLEDs[2] = CRGB::Black;
	FastLED.show();
	Serial.write("Off!\n");
	delay(500);
}
```

When setup for LPD8806 (or any other) it outputs to the strip just fine and you get the serial output just fine (which I added to be absolutely sure it wasn't just my strip connection).

But when running WS2801, the whole sketch just hangs... and it hangs the second you try to call FastLED.show(). I confirmed this with the serial outputs in the code above. The 5 second delay in setup() is so I have time to open the serial monitor. Using WS2801, I see "Start" printed to the monitor and then... nothing. But every other chipset keeps going.

The final way I tested this was on an Arduino Pro Mini using the standard bootloader and Arduino pro that comes with the IDE. Not funny business. Just FastLED 3.1 (synced today) and the code you see above. So I'd say WS2801 is totally hosed, for whatever reason... I diffed as much of the FastLED WS2801 related code as I could find and didn't see any huge changes. Wish I could point you to something more specific.

Thanks!

focalintent, 2015-04-11:
What happens if you add 

#define FASTLED_FORCE_SOFTWARE_SPI

before the #include?   (trying to pre-check ideas before i get back to a system where I can do testing) - curious if this is something in the hardware SPI handling or at the WS2801 driver level.

(Folks still use WS2801s?  Wow)

adammhaile, 2015-04-11:
That makes it work :) Weird. Wish it was an option to just use SoftSPI... Hopefully that at least helps narrow it down.
I also just tried on Linux and Mac, both with totally fresh Arduino installs and no Teensy (just Pro Micro).

I like WS2801... not the best, but better than some. And it's on the AllPixel supported list, so gotta have it :P My LED supplier said he still sells a ton of that type, also APA102 has started cutting into that.

focalintent, 2015-04-11:
Yeah, I've been pushing people to go to lpd8806 before the apa102came out.  Thanks for the check, I now know roughly where to look.

adammhaile, 2015-04-11:
Any time... I like the LPD8806 as well, especially for the speed. But the color quality is kinda crap (21-bit and all...) and I've found them *really* prone to interference from RF noise. So used to fall back to the 2801 when I didn't need the speed. Not any more with the APA102 :)

focalintent, 2015-04-11:
Pull 3.1 and try again - it's looking over the code I saw something missing in the avr hardware SPI implementation that, ironically enough, may not trip up APA102/LPD8806 because of the extra overhead of writing out the clocks post-frame.  It's a long shot, but i'm not going to be able to check with hardware until tomorrow at this point.  (And whether or not my change fixes your issue, it should still be done)

adammhaile, 2015-04-11:
Bingo! That worked great. What exactly does that waitFully() do? Just wait until all the bits are pushed?
I tested with all the AllPixel chipsets and didn't see any issues so far. I'll do some further stress tests to be sure, but looks like that did it.

focalintent, 2015-04-15:
Yeah - waitFully waits until the hardware SPI buffers have finished flushing out before control is returned.  This is extra important now because the library turns off the SPI hardware after it is finished with it.

closed 2015-04-15 by focalintent

adammhaile, 2015-04-15:
Ah... Why does it turn it off? I would think that would add more overhead
turning it on and off each time.

focalintent, 2015-04-15:
The overhead is minor, it opens the door for someone to use the hardware in different ways (e.g. by multiplexing with other SPI harwdare), on some platforms (remember, AVR isn't the only world out there, but I do want things to work consistently across the board) the SPI hardware eats resources when it's enabled but not running, releasing the SPI hardware is a requirement for supporting the multiple sets of pins on the teensy 3.0 that share the same hardware SPI engine, etc... etc... etc...