FastLED/FastLED #154: Very slow to compile on Arduino Mega 2560

jpro56, 2015-04-15:
After loading the latest version of the Arduino IDE (from 1.0.6 to 1.6.3), I tried to compile a known working sketch that uses the FasLED library 3.1 (the only external library in my sketch) but the Arduino IDE just appeared frozen in the compilation. That lasted a few minutes.

Note that I am running the Arduino IDE on a PC with Win-7 64-bits.

I then successfully compiled and  loaded the simple Arduino blink program.

I fetched the latest version of the 3.1 branch but the Arduino IDE still froze during compilation.

I left the Arduino in it's locked-up state and found that eventually (more than 5 minutes...) the sketch managed to get compiled and loaded to my Mega !?

For now, I solved the issue by going back to the Arduino IDE 1.0.6 as I did not need anything specific out of the latest IDE releases.

focalintent, 2015-04-15:
Can you supply the sketch that you were compiling?  With IDE 1.6.3 I'm having things compile here without too much issue with multiple sketches from the examples.  

I do notice that things seem to build more slowly in arduino 1.6.3.

focalintent, 2015-04-15:
(I'm suspecting that this is a bug with Arduino, and not with FastLED)

jpro56, 2015-04-15:
I suspect that you are right about this. This is why I reverted back to a more 'stable' IDE version.

Here's the sketch that I initially tried...  it is an incomplete work but I have loaded it my Mega and it works fine with my 16X25 WS2812 array.   

https://gist.github.com/77d0612c38fa44216593.git

Note that I have tried a few more sketches but with the same results but all of them use multiple pins (16) on my MEGA !?

jpro56, 2015-04-15:
If I can help you with reloading the 1.6.3 IDE and performing any checks, just ask !

focalintent, 2015-04-15:
Ah!  No wonder your compile times are so long - the Avr code ws2812 is stupidly complex for the compiler, and instantiates on pin - so you're doing that 16 times over.  

Seriously, if you are doing 16-way output get thee to a teensy 3.1 or due!

jpro56, 2015-04-15:
hmmm...  but it compiles fine and fast enough with IDE 1.0.6 !?

And yes, I just received my first 2 Teensy's 3.1 today...     :-)

focalintent, 2015-04-15:
Sure but the compiler versions and optimization options being  given to them are different.

jpro56, 2015-04-15:
Are these compiler options something I should have specified somewhere in the IDE ?

focalintent, 2015-04-15:
No, arduino typically doesn't give you any insight into/control over compiler optimizations and options.

focalintent, 2015-04-15:
(closing this because slow compiler is slow, and there's not likely anything i can do in the library to help it)

closed 2015-04-15 by focalintent

jpro56, 2015-04-15:
OK so I guess I've taken the appropriate action by reverting back to 1.0.6 for now...

I was just reading the very limited release notes for the IDE and do not see anything that strikes me as necessary for my limited usage.

focalintent, 2015-07-16:
FastLED 3.1 branch now builds this stuff faster - 10 lines takes ~20s on my rMBP.