FastLED/FastLED #155: Blue data on some MEGA pins

jpro56, 2015-04-15:
Arduino IDE 1.0.6
FastLED 3.1 
PC running WIN-7 64 bits.
Arduino MEGA 2560
Ws2812b strips
5Vdc 20 Amps PSU

I include some pictures of my setup...

It is normally used for my Xmas tree that uses 34 different pins, one for each branch.
Temporarily, it is used to drive an array of 16 strips X 25 WS2812b, each strip on a different pin.

I noticed some color discrepancies between some of the strips when the brightness was set very low.
Did some testing to define the problem and here's what I found...

1) The problem is only associated with the blue color, red and green are OK.

2) The problem is visible When blue is set to a low even value (2, 4, 6, 8...  possibly higher values also but it is not easy to see)

3) Included picture shows all strips set to CRGB (0,0,2) and the problem 'strips' appear OFF.

4) With all strips set to CRGB (0,0,4), the problem 'strips' appear dimmer than the other strips, as if set to CRGB (0,0,3).

5) With all strips set to CRGB (0,0,6), the problem 'strips' appear dimmer than the other strips, as if set to CRGB (0,0,5)... etc...

6) Found by moving cables around., the problem is not associated to any physical strip(s) but to specific Arduino MEGA pins. (Pins 42, 43, 44, 45, 46, 47, 48, 49).
   These pins also correspond to MEGA 2560 pins PL0 thru PL7. All other digital pins used in my setup appear OK. 

[image]
[image]

jpro56, 2015-04-15:
Here's the sketch I used to test this problem...

https://gist.github.com/jpro56/9afc5af460940e09447c

focalintent, 2015-04-15:
I thought I had updated the README - i'll do that now - you can't use pins 42,43,44,45,46,47,48, or 49 (aka PORT L) for clockless leds on the mega.  This is because the timing for accessing the given port is too high for the code that I have written, and it would be a significant amount of code re-working to accommodate that.

focalintent, 2015-04-15:
Actually, README.md already states: NOTE: On the mega, you can't use pins 42,43,44,45,46,47,48, and 49 with 3-wire chipsets.

closed 2015-04-15 by focalintent

focalintent, 2015-07-16:
The FastLED 3.1 branch now works on all Mega2560 pins.

NicoHood, 2015-07-16:
Which commit fixed this issue? Any further information what you did to solve this?
You said some ports are slower in i/o access. Does the LED writing speed also slows down on those pins, so you should avoid them for longer strips?

focalintent, 2015-07-16:
I don't recall which one because this got fixed as a side effect of some asm reorganization.  The slower pin access will not affect timing much, as there is some overhead at 16mhz (and other recent work that I did to make the intra-pixel overhead 0 cycles will more than compensate)