FastLED/FastLED #156: Blue data on some 'other' MEGA pins

jpro56, 2015-04-15:
Following the closing comments from my previous issue, I moved all connections from pins 42,43,44,45,46,47,48,49 to unused pins 4,5,6,7,8,9,10,11

I now find that I get the exact same behaviour out of pins 6,7,8 and 9 as described in my previous issue.

Pins 4,5,10 and 11 are OK.

focalintent, 2015-04-15:
The problem may be with PORTH, PORTI, PORTJ, and PORTK as well then.

focalintent, 2015-04-15:
It's not surprising, not all the ports are accessible via the in/out opcodes - and while I had an adjustment for not using in/out, the timing was still off and I don't know how much I want to go mucking around in that asm to support the higher ports (I just wasn't sure offhand where the line was between ports that can use the in/out opcodes and ports that can't).

jpro56, 2015-04-15:
OUCH!  At this time I have the following pins working just fine...

2,3,4,5
10,11  
22 thru 41
50 thru 53

jpro56, 2015-04-15:
For my usage, I would only need to re-wire 6,7,8,9 to 4 other pins any suggestions ?

focalintent, 2015-04-15:
This is the set of pin definitions for the mega:

```
_DEFPIN_AVR(0, 1, E); _DEFPIN_AVR(1, 2, E); _DEFPIN_AVR(2, 16, E); _DEFPIN_AVR(3, 32, E);
_DEFPIN_AVR(4, 32, G); _DEFPIN_AVR(5, 8, E); _DEFPIN_AVR(6, 8, H); _DEFPIN_AVR(7, 16, H);
_DEFPIN_AVR(8, 32, H); _DEFPIN_AVR(9, 64, H); _DEFPIN_AVR(10, 16, B); _DEFPIN_AVR(11, 32, B);
_DEFPIN_AVR(12, 64, B); _DEFPIN_AVR(13, 128, B); _DEFPIN_AVR(14, 2, J); _DEFPIN_AVR(15, 1, J);
_DEFPIN_AVR(16, 2, H); _DEFPIN_AVR(17, 1, H); _DEFPIN_AVR(18, 8, D); _DEFPIN_AVR(19, 4, D);
_DEFPIN_AVR(20, 2, D); _DEFPIN_AVR(21, 1, D); _DEFPIN_AVR(22, 1, A); _DEFPIN_AVR(23, 2, A);
_DEFPIN_AVR(24, 4, A); _DEFPIN_AVR(25, 8, A); _DEFPIN_AVR(26, 16, A); _DEFPIN_AVR(27, 32, A);
_DEFPIN_AVR(28, 64, A); _DEFPIN_AVR(29, 128, A); _DEFPIN_AVR(30, 128, C); _DEFPIN_AVR(31, 64, C);
_DEFPIN_AVR(32, 32, C); _DEFPIN_AVR(33, 16, C); _DEFPIN_AVR(34, 8, C); _DEFPIN_AVR(35, 4, C);
_DEFPIN_AVR(36, 2, C); _DEFPIN_AVR(37, 1, C); _DEFPIN_AVR(38, 128, D); _DEFPIN_AVR(39, 4, G);
_DEFPIN_AVR(40, 2, G); _DEFPIN_AVR(41, 1, G); _DEFPIN_AVR(42, 128, L); _DEFPIN_AVR(43, 64, L);
_DEFPIN_AVR(44, 32, L); _DEFPIN_AVR(45, 16, L); _DEFPIN_AVR(46, 8, L); _DEFPIN_AVR(47, 4, L);
_DEFPIN_AVR(48, 2, L); _DEFPIN_AVR(49, 1, L); _DEFPIN_AVR(50, 8, B); _DEFPIN_AVR(51, 4, B);
_DEFPIN_AVR(52, 2, B); _DEFPIN_AVR(53, 1, B); _DEFPIN_AVR(54, 1, F); _DEFPIN_AVR(55, 2, F);
_DEFPIN_AVR(56, 4, F); _DEFPIN_AVR(57, 8, F); _DEFPIN_AVR(58, 16, F); _DEFPIN_AVR(59, 32, F);
_DEFPIN_AVR(60, 64, F); _DEFPIN_AVR(61, 128, F); _DEFPIN_AVR(62, 1, K); _DEFPIN_AVR(63, 2, K);
_DEFPIN_AVR(64, 4, K); _DEFPIN_AVR(65, 8, K); _DEFPIN_AVR(66, 16, K); _DEFPIN_AVR(67, 32, K);
_DEFPIN_AVR(68, 64, K); _DEFPIN_AVR(69, 128, K);
```

for each _DEFPIN_AVR cal, the third number is the port, the second number is the bit on that port, and the first number is the pin.  You want anything on port A,B,C,D,E,F, and G

(Seriously, though, you're going to be so happy when you get onto a platform where the 16-way output is done in parallel, instead of serial :)

jpro56, 2015-04-15:
Yeah I know but this is already built around that MEGA, and tis a working setup, I prefer not to have to redo it.

Note that the behaviour is not easy to notice, only the blue component for very low brightness settings on just a few pins.

Struggling with that table but could I use some of the analog pins, define them to be digital outs ?

focalintent, 2015-04-15:
If the pins are listed in that list up there, you can use them.  I'll see if I can sort that out into something ordered by port for you.  (Honestly, I have to admit, I'm a bit surprised that 16-way output, given the amount of code duplication that would cause, fits in the mega's flash!)

focalintent, 2015-04-15:
A 22
A 23
A 24
A 25
A 26
A 27
A 28
A 29
B 10
B 11
B 12
B 13
B 50
B 51
B 52
B 53
C 30
C 31
C 32
C 33
C 34
C 35
C 36
C 37
D 18
D 19
D 20
D 21
D 38
E 0
E 1
E 2
E 3
E 5
F 54
F 55
F 56
F 57
F 58
F 59
F 60
F 61
G 39
G 4
G 40
G 41
H 16
H 17
H 6
H 7
H 8
H 9
J 14
J 15
K 62
K 63
K 64
K 65
K 66
K 67
K 68
K 69
L 42
L 43
L 44
L 45
L 46
L 47
L 48
L 49

jpro56, 2015-04-15:
Thanks a lot, got it !! 

I will rewire them last 4 to pins 18, 19, 20 ,21 and test it again ! 

With this temporary setup I have 400 LEDs on 16 pins.
I actually got 34 pins running in parallel quite fine on the tree but only driving a total of 332 LEDs.
The tree sketch compiles in less than 54K.

jpro56, 2015-04-15:
Just reopened this briefly to mention that I have changed from the pins 6,7,8,9 to pins 18,19,20,21 with success !!!!

Thanks a lot Daniel for the time taken to help me fix this issue !!

closed 2015-04-15 by jpro56

focalintent, 2015-07-16:
The FastLED 3.1 branch now works on all Mega2560 pins.