FastLED/FastLED #157: Q. about 'lost' bit

knut2307, 2015-04-24:
I try to pass color 0xFFFFFF to WS2812B. 
Seems like last bit is masked off.  Scope tells me 0xFEFEFE as result to LEDs.  (I know this means 'nothing' to experienced functionality) .. anyway:
is this intended or 'bug'... or don't I understand the protocol?
Knut N

kriegsman, 2015-04-24:
You're correct in what you're seeing. It's a side-effect of how we implement the master brightness control (FastLED.setBrightness) with nondestructive dimming. It's essentially imperceptible (except for POV arrays), so overall it's a good trade off. 

Are you seeing any problems from this, or are you just noting something interesting?

knut2307, 2015-04-24:
Thank you for your time and work. I'm dont dare think of the houres you put in to this.. Tjank ypu, once more.

closed 2015-04-24 by knut2307