FastLED/FastLED #158: Problems with Arduino pro mini(3.3V 8Mhz) + WS2812B

SeanGau, 2015-04-28:
I'm using Arduino pro mini(3.3V 8Mhz) and WS2812B strip.
It works fine when I use the Adafruit library, but it doesn't work with FastLED library.
There are only front few leds twinkling but it's wrong.

focalintent, 2015-05-14:
Right now the only 8Mhz platform that I've done testing with is the ATTiny platform, not the ATMega328.  I have some attiny related issues to fix, it is possible they'll help with this.

focalintent, 2015-07-14:
A bunch of 8Mhz and tiny related fixes have gone in.

closed 2015-07-14 by focalintent

TomasHubelbauer, 2018-03-04:
@SeanGau has the fixes resolved your problems? I have the same setup - Arduino Pro Mini 3.3 V 8 MHz ATMega 328P. I can't get even the simplest program to run, the WS2812B LED strip just won't light up for me no matter what.
I tried using `clock_prescale_set(clock_div_2);` and tried with AdaFruit Neopixel library with 800 Mhz and 400 Mhz settings, too, but nothing happens. I don't have a scope to diagnose the digital PIN to determine if any signal is being sent at all unfortunately.
The version of FastLED that I am using is 3.001.007.