FastLED/FastLED #159: Please add your library to Arduino Library Manager

ffissore, 2015-05-04:
See FAQ: https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

kriegsman, 2015-05-04:
Good suggestion!

Looking at their FAQ though, it looks like it might not be able to handle libraries with subfolders, which is how we're trying to keep the (many!) files organized. 

Do you have experience with the library manager, and how it deals with subdirs?

ffissore, 2015-05-04:
Sub folder limitation is related to the location of `library.properties`. You can't have a repo structured [this](https://github.com/wayoda/LedControl/tree/1.0.1) way: you need it to be structured [this](https://github.com/wayoda/LedControl/tree/1.0.6) way.
As far as I can see from FastLED, it only misses a `library.properties`. See [this](https://github.com/arduino-libraries/Bridge/blob/master/library.properties) for example.

kriegsman, 2015-05-04:
Thank you!
We'll take a look!

NicoHood, 2015-05-04:
So the only thing is you have to move the .h/.cpp files into src/ and add a library.properties?

focalintent, 2015-05-04:
No, because I'll also want to do a bunch of testing to make sure things work in all the environments that we currently work with, as well as platforms and such.  So it's not the only thing that needs to be done (and is also why I haven't done this yet).

ffissore, 2015-05-04:
Only `library.properties` is needed. Moving files into `src` is optional. See this for example: https://github.com/adafruit/Adafruit_NeoPixel

focalintent, 2015-07-14:
lib.prop file added, will actually submit to the library manager when we push 3.1.

NicoHood, 2015-07-16:
Before you push 3.1 you should try to move all files into /src if possible.
Just a reminder because you said 3.1 will come soon ;)

focalintent, 2015-07-16:
I need to first find out whether or not that is actually needed (from another comment above it looks like it isn't).  I'd like to avoid doing a complete tree re-organization if I don't have to (at least not until I'm doing other major re-org work for other reasons).  Part of why I'm holding off is I want to find a way to do this that will allow me to keep the spark core tree in the main tree as well (instead of in a separate repo which is what's necessary now).

NicoHood, 2015-07-16:
Its not needed. But its the new schema. If you change it 1.0.x will also be incompatible.
It will clean up the github page and the folder though.

focalintent, 2015-07-16:
>  If you change it 1.0.x will also be incompatible.

This right here is a reason to not go out of my way to change it just yet.

I'm not going to do the layout changes until I absolutely have to.  It was problematic enough doing a version of a layout change for spark core support.  When I work on Photon support and revisit the spark tree I will see if there's a single layout that will work both for the new library layout and for the spark core's setup, and then i'll consider it.  (Even then, though - for the longest time, I was supporting 1.0.x and 1.5.x because there were some platforms/tools only available under 1.0.x, so I'm not really in any rush to drop support for 1.0.x).

focalintent, 2015-10-03:
Submission made to include the library - https://github.com/arduino/Arduino/issues/3917 - I'm not going to re-layout the library until I work out how to do a layout that will work for both the new arduino layout and environments like mbed and particle's spark/photon.

closed 2015-10-03 by focalintent