FastLED/FastLED #160: Signal issues with FastLED and Teensy Audio

jbergler, 2015-05-04:
I have an issue similar to #141, I'm trying to use the pjrc Audio library in combination with FastLED on my teensy 3.1. 

With current versions of the Audio and FastLED (3.1) libraries, as soon as I enable any audio processing, the LED's immediately flicker almost randomly, at least 50% of the time they are showing things they shouldn't be.

My theory was that something the audio library is doing messes with the timing of the bit bashing for the 2812B LED's I'm using.

@focalintent pointed out in chat that I should try `#define FASTLED_ALLOW_INTERRUPTS 0` which seems to have done the trick. I'm opening this as an issue for 2 reasons.

1. To have this on record that someone may hopefully find in the future with their googling.
2. To ask if there is anything that can be done to document this / possibly explain the consequences of disabling this option.

If there's no value in either, please feel free to close the issue. Also, for the record, here is a minimal version of the code I could reproduce the issue with. 

```c
// Audio Setup
#include <Wire.h>
#include <SD.h>
#include <SPI.h>
#include <Audio.h>

AudioInputAnalog         audio_in(A9);
AudioAnalyzeFFT1024      audio_fft;
AudioConnection          patchCord1(audio_in, 0, audio_fft, 0);

// LED Setup
#include <FastLED.h>
#define LED_PIN     10
#define COLOR_ORDER GRB
#define CHIPSET     WS2812B
#define NUM_LEDS    68
#define BRIGHTNESS  40
#define FRAMES_PER_SECOND 60

CRGB leds[NUM_LEDS+1];

void setup() {
  Serial.begin(9600);
  Serial.println("Restarting...");
  delay(2500); // sanity delay

  FastLED.addLeds<CHIPSET, LED_PIN, COLOR_ORDER>(leds, (NUM_LEDS+1));
  FastLED.setBrightness( BRIGHTNESS );
  FastLED.show();
  
  AudioMemory(10);
}

void loop() {
  leds[random(0,NUM_LEDS)] = CRGB::Green;
  leds[random(0,NUM_LEDS)] = CRGB::Blue;
  leds[random(0,NUM_LEDS)] = CRGB::Red;
  FastLED.show();
  FastLED.delay(1000/FRAMES_PER_SECOND);
  
  if (audio_fft.available()) {
    Serial.println("fft results available");
  } else {
    Serial.println("fft results NOT available");
  }
}
```

cneuzil, 2015-05-05:
I wanted to double check to see if my sketch still worked from #141 , and I found that it worked at 72MHz but not 96MHz.  I then pulled the latest audio and FastLED libraries, and now it doesn't work at 72MHz.

It does work if I add the line #define FASTLED_ALLOW_INTERRUPTS 0.  I didn't have to add that before.  My libraries were pulled at 2015-04-11, so I need to figure out how to get back to that state.

focalintent, 2015-05-14:
Using leds like the WS2811 require the library to disable interrupts.  There is, however, a small window of time between each led in which interrupts can be re-enabled to allow them to be handled.  If they take too long, however, either it will corrupt the data output, or it will cut your output frame short.  In which case, if you see that happening you have to disallow interrupts or switch to an led chipset that doesn't require disabling of interrupts to push out data, such as the APA102.

This is now documented on this page - https://github.com/FastLED/FastLED/wiki/Platform-limitations

closed 2015-05-14 by focalintent

jwhendy, 2017-06-11:
Does this apply to the teensy 3.2 and WS2812b as well? I'm guessing *yes*, but just thought I'd confirm for the benefit of others.  The limitations page doesn't show these, but this board might not have been out 2 years ago. If you google "teensy audio fastled audio interrupts" this is in the top hits but most of the other hits are on the teensy forums.

I had an issue with a spectrum analyzer, but didn't know where to start troubleshooting. I assumed it was my audio circuit, so I [posted on the teensy forums](https://forum.pjrc.com/threads/44761-Spectrum-analyzer-low-high-fft-read()-bins-triggering-at-wrong-frequencies) and someone mentioned interrupts. That enlightened me so now I know.