FastLED/FastLED #162: Taking long time to compile

willie68, 2015-05-08:
I switch from FastSPI_LED2 to FastLED and i found out that the compile time is about 10x than before. If i switch back to the old lib compiling is fast again. 
In the windows task manager i see that the task cc1plus.exe is the "problem".
Mainly i'm using the Arduin IDE 1.6.3, but i have testet this with the older 1.5.7 too. Same problem.

focalintent, 2015-05-14:
FastLED does a lot more, and is far more aggressive in it's use of inlining and templates to produce optimized code, which results in slower compile times.  This is a thing that is unlikely to change anytime soon.  (Also, the library itself is 15x larger - having gone from about 900 lines of code in FastSPI_LED2 to about 15,000 lines of code in FastLED 3.1)

closed 2015-05-14 by focalintent

willie68, 2015-05-14:
The problem mainly is, that the lib will recompiled every time, not just once a session like other libs.
At this point the 3.1 is useless for me. The optimisations are not so appreciable for my application, that i can slow down my development that much. But this is only my personal opinion. So i will use the old one. Thanks for your effort.

focalintent, 2015-05-19:
How long are you seeing build times take?  (in actual minutes/seconds, not just as a comparison to older build times).

If you can supply me the code that you're building I can see if there's any recommendations that I can make for your code to make it build faster.  While the build is mildly slower for me, it is still in the realm where the upload to your standard uno/due takes far longer than the build itself.  In fact, until this ticket, the only time I had a complaint about build times was someone who was basically making 30+ instantiations of the output classes, and well, no shock there that they were basically increasing by nearly 20x the total build time by doing that.

willie68, 2015-05-20:
Newly installed arduino 1.6.4. 
On first compile i have measured
FastSPI_LED2.h -> 12 seconds
FastLED.h -> 1:28 min. (88 seconds) 

on recompile (just changing nothing but adding a CRLF)
FastSPI_LED2.h -> 3 seconds
FastLED.h -> 1:13 min. (73 seconds) 

Here is my code. Quite simple.
```c++
#include "FastSPI_LED2.h"
//#include "FastLED.h"

///////////////////////////////////////////////////////////////////////////////////////////
// How many leds are in the strip?
#define NUM_LEDS 7

// Data pin that led data will be written out over
#define DATA_PIN 11

// This is an array of leds.  One item for each led in your strip.
CRGB leds[NUM_LEDS];

const byte LED = 13;

void setup() {
  Serial.begin(9600);

  // This function sets up the leds 
  // and tells the controller about them
  FastLED.addLeds<WS2812B, DATA_PIN, GRB>(leds, NUM_LEDS);

  Serial.println("SkypeStatus V0.1");
}

const word WAIT_DELAY = 10;
// This function runs over and over, and is where you do the magic to light
// your leds.
CRGB color;

byte index = 0;
byte prevSensorValue = 0;
long time = 0;
boolean blink = false;
boolean connect = false;

byte red = 64, green = 0, blue = 0;

void loop() {

  if (Serial.available() > 0) {
    if (Serial.read() == '#') {
      int index = Serial.parseInt();
      int red = Serial.parseInt();
      int green = Serial.parseInt();
      int blue = Serial.parseInt();
      byte myChar = Serial.read();
      if (myChar == 'b') {
        blink = true;
        myChar = Serial.read();
      } else {
        blink = false;
      }
      if (myChar == '*') {
        index = constrain(index, 0, NUM_LEDS);
        red = constrain(red, 0, 255);
        green = constrain(green, 0, 255);
        blue = constrain(blue, 0, 255);

        Serial.print(index);
        Serial.print(':');
        // print the three numbers in one string as hexadecimal:
        Serial.print(red, HEX);
        Serial.print('.');
        Serial.print(green, HEX);
        Serial.print('.');
        Serial.println(blue, HEX);
        clear();
        FastLED.setBrightness(255);
        showColor(index, CRGB(red, green, blue));
        time = 0;
        connect = true;
      } else {
        clear();
      }
    }
  } else if (!connect) {
    if (red == 64) {
      red = 63;
      green = 1;
      blue = 0;
    } else if (red > 0 && green < 64 && blue == 0) {
      red--;
      green++;
    } else if (green == 64) {
      green = 63;
      blue = 1;
      red = 0;
    } else if (green > 0 && blue < 64 && red == 0) {
      green--;
      blue++;
    } else if (blue == 64) {
      blue = 63;
      red = 1;
      green = 0;
    } else if (blue > 0 && red < 64 && green == 0) {
      blue--;
      red++;
    }
    blink = false;
    showColor(0, CRGB(red, green, blue));
    delay(100);
  } else {
    if (blink) {
      if ((time % 2) == 0) {
        FastLED.setBrightness(255);
      } else {
        FastLED.setBrightness(0);
      }
    }
    FastLED.show();
    time++;
    if (time > 60) {
      time = 0;
      connect = false;
    }
    delay(1000);
  }
}

void setLED(byte index, CRGB color) {
  index = index % NUM_LEDS;
  index = (NUM_LEDS) - index;
  index = index % NUM_LEDS;
  leds[index] = color;
}

void showColor(int index, CRGB color) {
  if (index == 0) {
    for (int i = 0; i < NUM_LEDS; i++) {
      setLED(i, color);
    }
  } else {
    setLED(index - 1, color);
  }

  // Show the leds (only one of which is set to white, from above)
  //  leds[0] = CRGB::DarkGreen;
  FastLED.show();
}

void clear() {
  for (int i = 0; i < NUM_LEDS; i++) {
    setLED(i, CRGB::Black);
  }
}
```

focalintent, 2015-05-21:
So, on the mac, these compile times are more like 4 seconds and 12 seconds for me (both of which are well within the range of acceptable compile times for me).  I'm curious what the hell the compiler is doing under windows that you are seeing over 1 minute compile times.  I'm setting up a windows environment here to test with to see how it comes out.  That type of time difference between the mac and the windows screams of an issue with how the compiler was built, and is more likely a bug for the arduino folks than it is for me.  

I'll update this once I get some running done on the windows side of things.

focalintent, 2015-05-21:
On my windows system it is more like 20-26 seconds to build with the FastLED.  Specifically, the issue appears to be the windows version of the compiler that they're using is having significant issues with the asm code that FastLED uses for the WS2812 (from some other testing, it appears that the mac version is having some of these issues as well, but they fly under the radar because it's building so much faster across the board, period).  If avr-gcc on windows is using the cygwin (mingw may have this issue as well, though) library to provide posix compatibility, that might be one possible source of issue, as cygwin and windows file i/o has always been a hellish combination.

Unfortunately, there isn't a whole lot that I can do with/about the asm code that's in there for the WS2812 and friends.  It's really the only way to get the inline scaling, color correction, and dithering support while sticking with the 2812's strict timing requirements.  The compile times drop to under 2-4 seconds on my windows machine when building for an SPI/4 wire led chipset like the APA102 that doesn't require the asm code.  

Of course, the times that i'm seeing are still something like four times faster than what you're seeing.  I don't know if that is a function of the OS that i'm using (I'm running Windows 10) or the hardware that I have (Surface Pro 3 w/core i5 and an SSD drive) or if there's some other issue lurking within the depths of avr-gcc.

I don't have a linux system to test with here, but I suspect compile times on the linux system are likely to be more in line with the mac than they are with windows.

willie68, 2015-05-21:
That are the times i would have expected. My first maschine is an core I5 with SSD, but i try it on another maschine with an core I7 and SSD. There i get nearly the same results as you. 25 seconds for a full rebuild. I don't mind about the 1:28 mintues, but i find it strange that a recompile (without changing the lib sources) always takes the same time. That's the main problem for me.

focalintent, 2015-05-21:
What version of windows are you using on the two machines?  Also - the reason why you have a full rebuild without changing the library sources is because the template code is in header files, not source files, which means they're re-built when you build, whether or not the files have changed.

willie68, 2015-05-21:
Both machines runnning Windows 8.1 64bit. 
So could i change the template location?

focalintent, 2015-05-21:
No, because then gcc won't instantiate it properly.  There isn't a way around this.

davidbrenner, 2015-07-08:
It'd be some work, but since you only have approximately <20 template definitions, couldn't you move them to the source files and then use explicit instantiation?

I was going to use this library in a workshop teaching kids programming, but waiting ~30 seconds for the code to compile when the only change is the color of the LED is too long (and that's on modern machines. Some of the laptops the kids have access to are older and will certainly take >1min per compilation). It's a great library, but I'd encourage you to reconsider looking into a fix for this issue.

For others experiencing this, until it is resolved (or if it remains a wontfix), I'd suggest taking a look at the Adafruit neopixel library at: https://github.com/adafruit/Adafruit_NeoPixel.

focalintent, 2015-07-08:
>It'd be some work, but since you only have approximately <20 template definitions, couldn't you move them to the source files and then use explicit instantiation?

For one, it'd be more than "some work" - Mark and I each have already put a stupid amount of time into juggling and dealing with this.

The reason why the explicit instantiation wouldn't work is basically one of math.  There is one problematic template, ClocklessController for the AVR platform.  This is because there is some pretty complex asm code in there to handle the various timing requirements for the 12 3-wire led chipsets the library supports, while also rolling in the 0-overhead, non destructive scaling, dithering, color correction, etc...

This template has 3 parameters it can be instantiated on - digital pin, rgb ordering, and the explicit timings.  On an arduino, there's 19 digital pins (with the megas having far more).  There are 6 possible rgb orderings.  Finally, there are, as mentioned above, 12 3-wire led chipsets, all with different timings.  So, that comes out to 1368 explicit instantiations that would be needed to cover everything the library supports.  And, you would still need to compile the .cpp file with those 1300 instantiations at least once per arduino session (because of how it cycles temp directories from run to run, at least on my system here) - and each instantiation of CClocklessController takes 5-60 seconds to compiler (depending on platform, it's fastest on the mac, by almost an order of magnitude compared to windows).  

This actually makes the compilation time far far far worse, with the added benefit of penalizing everyone, not just folks compiling for the WS2812 on AVR. 

Ultimately, the problem here is one of two sets of bugs/issues with gcc.  The first is that avr-gcc is significantly slower to compile on windows than it is on OS X.  The second is that gcc doesn't play well in environments where you have a lot of heavy mixing of asm and C++ code, and working around those issues has contributed to the complexity of the asm and the compiling here.

If someone can actually get the compile time down in a meaningful way, without breaking the combination of chipsets and platforms that are supported, and without losing the functionality/performance that the library has, I will happily accept the pull. Until I get through much more of my own backlog of both issues and features for the library (including at least 3 new hardware platforms that I have sitting here that need support), I don't have any more time to put into this one issue.

NicoHood, 2015-07-09:
@davidbrenner He is totally right and you know the library is great. If you want to use it for education buy proper Led chipsets. Apa102 is way faster and in general better because you then dont have any interrupt problems which you may have to explain some beginners. Or ws2801 or those lpdxxxx also work. And they are not much more expensive.

kriegsman, 2015-07-09:
Short version:
It's driving us crazy too. 
We've put a ton of time into hiring down the root cause. 
The root cause is gcc asm block arg performance sucking (suddenly in the version of gcc that Arduino bundles).

If there were an easy solution, we'd already be using it. There isn't one. All the solutions are a lot of work. 

That said, this is open source land. If you can make it better, we'll totally take the pull request!

willie68, 2015-07-09:
@NicoHood: It makes no difference in compile time, if you use WS2812 or other LED's. What's slow is the lib itself...
For me, i just need a lib for the WS2812B, so i switch to Adafruit Neopixel lib.

kriegsman, 2015-07-09:
That is not correct. 

gcc is ONLY slow compiling the three-wire code: WS2811/WS2812/etc, and ONLY for AVR CPUs. 

gcc is fast compiling the three-wire LED code for ARM CPUs. 

gcc is fast compiling the four-wire LED code for all platforms. 

It's just WS28xx LEDs on AVR that gcc is really sucking at right now. Switching to four-wire LEDs like APA102 totally makes the problem disappear.

kriegsman, 2015-07-09:
No further comments here please unless you have startling new data that you've discovered or a pull request. gcc has a problem, it's making us all miserable, and that's the whole story right now.

focalintent, 2015-07-15:
You're welcome - https://github.com/FastLED/FastLED/commit/e74caebaab54ffa82b4c8393fcde02d5da05edd2 (and the speedup may actually be closer to 2-4x )