FastLED/FastLED #163: ATmega168PA with internal 8Mhz clock on a breadboard

strohne, 2015-05-10:
First things first: thank you for the great work!

I try to use FastLED on an 'Arduino on a breadboard' with ATmega168**PA** and internal 8Mhz clock. Compiling with Arduino IDE 1.6.3 for a ATmega168 works as expected, but compiling for ATmega168**p**  throws errors. For example:

```
FastLED/platforms/avr/clockless_trinket.h:234:36: note: in expansion of macro 'ROR1'
#define SCROR04(B, N) SCALE02(B,N) ROR1(B) CLC1

[...]

FastLED/platforms/avr/clockless_trinket.h:226:47: error: impossible constraint in 'asm'
 #define CLC1 asm __volatile__("clc" ASM_VARS );
```

To reproduce create a board definition file hardware\breadboard\avr\boards.txt with the following content:

```
### ATmega168PA ###
atmega168pabb.name=ATmega168PA on a breadboard (8 MHz internal clock)

atmega168pabb.upload.protocol=arduino
atmega168pabb.upload.maximum_size=14336
atmega168pabb.upload.maximum_data_size=1024
atmega168pabb.upload.speed=19200
atmega168pabb.upload.tool=arduino:avrdude

atmega168pabb.bootloader.tool=arduino:avrdude
atmega168pabb.bootloader.low_fuses=0xE2
atmega168pabb.bootloader.high_fuses=0xDE
atmega168pabb.bootloader.extended_fuses=0x05
atmega168pabb.bootloader.file=ATmegaBOOT_168_pro_8MHz.hex
atmega168pabb.bootloader.unlock_bits=0x3F
atmega168pabb.bootloader.lock_bits=0x0F

atmega168pabb.build.mcu=atmega168p
atmega168pabb.build.f_cpu=8000000L
atmega168pabb.build.core=arduino:arduino
atmega168pabb.build.variant=arduino:standard

### ATmega168 ###
atmega168bb.name=ATmega168 on a breadboard (8 MHz internal clock)

atmega168bb.upload.protocol=arduino
atmega168bb.upload.maximum_size=14336
atmega168bb.upload.maximum_data_size=1024
atmega168bb.upload.speed=19200
atmega168bb.upload.tool=arduino:avrdude

atmega168bb.bootloader.tool=arduino:avrdude
atmega168bb.bootloader.low_fuses=0xE2
atmega168bb.bootloader.high_fuses=0xDE
atmega168bb.bootloader.extended_fuses=0x05
atmega168bb.bootloader.file=ATmegaBOOT_168_pro_8MHz.hex
atmega168bb.bootloader.unlock_bits=0x3F
atmega168bb.bootloader.lock_bits=0x0F

atmega168bb.build.mcu=atmega168
atmega168bb.build.f_cpu=8000000L
atmega168bb.build.core=arduino:arduino
atmega168bb.build.variant=arduino:standard
```

This defines two boards, the only difference is build.mcu=atmega168p vs. build.mcu=atmega168.

The following code compiles with the second board, but not the first:

```
#include "FastLED.h"

#define LEDS_NUM 30
#define LEDS_TYPE    WS2812B
#define LEDS_COLOR_ORDER GRB
#define PIN_LED_STRIP 3

CRGB leds[LEDS_NUM];

void setup() {
  FastLED.addLeds<LEDS_TYPE,PIN_LED_STRIP,LEDS_COLOR_ORDER>(leds, LEDS_NUM).setCorrection( TypicalLEDStrip );
}

void loop() {
}
```

It would be great if you could add support for the ATmega168PA or give me some hints on how to compile it.

focalintent, 2015-05-14:
Ok - added a define for the 168p -- assuming there's no major differences between the 168p and the 168, this should make things fine.  (Note - the library has issues on some avr platforms at 8Mhz right now - depending on the leds you are using, you may run into some of these).

closed 2015-05-14 by focalintent

strohne, 2015-05-14:
Works very well, thank you!