FastLED/FastLED #165: WS2812B LED coloring issue

mattwever, 2015-05-15:
Running Arduino 1.6.3 with FastLED 3.0, Arduino Nano 3, 328 atmega, WS2812B leds.

LEDs, when setup with RGB ordering, output green for red, and red for green. Blue is fine.

Switching to GRB ordering, all the LEDs in my strip turn green, with hints of other colors (ie a flashing "red" LED will be a slightly pulsing fully lit green LED. 

The coloring of the strip does not occur in the RGB ordering - but obviously, the colors are backwards.

Any thoughts? Using pins 4 and 6 if it matters.

focalintent, 2015-05-19:
Are you running the nano at 8 or 16Mhz?

mattwever, 2015-05-19:
I'm not quite sure. What's the easiest way to check? Should it be one or
the other?

Update: when I use one of the beta IDE (1.5.6) the lights work as coded.

focalintent, 2015-05-19:
Check out https://github.com/FastLED/FastLED/issues/164 -- let me know if that works for you.  (If it does, i'll be over here in the corner setting fire to gcc)

hamsafar, 2015-05-27:
Hi
I have a same problem with Arduino DUE.
I want to use ParallelOutput with DUE. when I use this code:

42.// Set the first n leds on each strip to show which strip it is
43.  for(int i = 0; i < NUM_STRIPS; i++) {
44.    for(int j = 0; j <= i; j++) {
45.     leds[(i*NUM_LEDS_PER_STRIP) + j] = CRGB::Red;
46.   }
47. }

Color of led is not correct(Is not Red!) 
In first strip, first LED is a bit blue and the second LED is Green! and when I change line 45 to Green, the first led become red and blue and the second LED become off, and when I change line 45 to blue, the first LED become off and the second become red! I think in shift data is a problem or my DUE frequency isn't match with this library!

please help me....
Thanks﻿

michaelmarriott, 2015-05-27:
I'm not sure how you declared your leds array, but you can change the config and state explicitly
EG:
FastLED.addLeds<WS2812, 4, GRB>(leds, NUM_LEDS_PER_STRIP);
vs
FastLED.addLeds<WS2812, 4, RGB>(leds, NUM_LEDS_PER_STRIP);
will switch the green and red around.

focalintent, 2015-05-27:
@hamsafar please use the fastled3.1 branch for parallel output with the due - if you are still having problems please open a separate issue as the due is a wildly different platform than avr (and in the ticket please include the version of arduino that you are using and the os you are using.

hamsafar, 2015-05-27:
Ok,thanks.

focalintent, 2015-07-14:
#164 - which was the root of the issue in here, is now fixed.

closed 2015-07-14 by focalintent