FastLED/FastLED #166: FastLED on Photon

jasoncoon, 2015-05-20:
Copied from https://community.particle.io/t/fastled-on-photon/11933

No worries, and no rush.  Thanks!

```
In file included from /spark/compile_service/shared/workspace/6_hal_12_0/firmware-privafastled-apa102-60.cpp:1:0:
/spark/compile_service/shared/workspace/6_hal_12_0/firmware-privaFastLED/FastLED.h:12:2: warning: #warning FastLED version 3001000 (Not really a warning, just telling you here.) [-Wcpp]
#warning FastLED version 3001000 (Not really a warning, just telling you here.)
^
In file included from /spark/compile_service/shared/workspace/6_hal_12_0/firmware-privaFastLED/FastLED.h:34:0,
from /spark/compile_service/shared/workspace/6_hal_12_0/firmware-privafastled-apa102-60.cpp:1:
/spark/compile_service/shared/workspace/6_hal_12_0/firmware-privaFastLED/led_sysdefs.h:19:43: fatal error: platforms/avr/led_sysdefs_avr.h: No such file or directory
#include "platforms/avr/led_sysdefs_avr.h"
^
compilation terminated.
make[1]: *** [../build/target/user/platform-6fastled-apa102-60.o] Error 1
make: *** [user] Error 2

Error: Could not compile. Please review your code.
```

m-mcgowan, 2015-05-20:
I'm happy to help out here where needed, such as getting a STM32F2XX define in place for the photon/electron.

stevesparks, 2015-07-01:
Bump, this is happening to me as well.

Jerware, 2015-07-02:
I would love to see this solved too.

focalintent, 2015-07-08:
The Photon isn't yet supported by FastLED.  I have one, but haven't had the time to get it up and running and tested on there.

kasperkamperman, 2015-08-22:
Just came across the update for Photon of the DotStar library. It didn't seem too different. The (SPARK) definition changed to platform id. Maybe this helps in some way:
https://github.com/wesner0019/SparkCore-DotStar/blob/master/firmware/dotstar.cpp
More on the pin-mapping macro:
https://community.particle.io/t/touch-4wire-library-not-yet-ported-for-photon/12372/25

focalintent, 2015-08-22:
@kasperkamperman the dotstar library uses higher level libraries to talk to the gpio pins and hardware - while I'm directly using the low level gpio and hardware registers. These have changed between the spark core and the photon, so no, it isn't just about adding a platform id.

kasperkamperman, 2015-08-22:
Clear. I've missed that higher level layer in between. I looked at your code and I thought to see some similarities (with my non-expert eye).

rschanafelt, 2015-10-06:
Is the Particle (Spark) Photon supported in the latest 3.1 release? Or is only the Particle (Spark) Core supported still?

focalintent, 2015-10-06:
No, only the spark core still - I will update this ticket and make an announcement when the photon is supported.

beaubrewer, 2015-10-08:
Happy to see APA102 support is coming. Getting FastLED working with the Photon would amazing.
@Daniel, keep up the excellent work!

keavin7, 2015-10-13:
Photon User Here.
@focalintent I'm really excited to try the FastLED library! I've heard its second to none! I will squeal like a little girl when this feature hits my project :)

beaubrewer, 2015-10-21:
@focalintent, I would love to help add support for the Photon.
My C/C++ is rusty, but if you could point me in the right direction I'm ready and willing to do what it takes. I'm guessing it will entail more than [updating the platforms.h](https://github.com/FastLED/FastLED/blob/master/platforms.h#L17) to match the Photon's STM32F205.

eklex, 2015-10-22:
Hi everyone!

I added support for the Photon to FastLED-Sparkcore library.
I didn't re-test every exemples since I'm doing that for one of my projects. However I've been using the lib for 2 weeks now, which makes me pretty confident.
I run the lib in ParticleDev IDE with Photon firmware 0.4.6, I didn't try 0.4.7 yet.
You can find my port at [eklex/FastLED-Sparkcore](https://github.com/eklex/FastLED-Sparkcore). 
Please give feedback, and if everything's good, I will request a pull to the master.

As a side note, don't try to import my custom library into the web IDE, it won't work because of the json file.

focalintent, 2015-10-22:
Nice, and thanks!  I'll take a swing at it with my photons later tonight - it looks like the right set of register changes.  I'll deal with the hardware SPI bits later on (since it falls back to bit banging that's fine).

jasoncoon, 2015-10-22:
Fantastic, thank you!  I hope to be able to try this out tonight.

causticforms, 2015-10-22:
Great news! I've just tried to get up and running with this with the ParticleDev IDE but no joy... it seems to compile, upload and run on the Photon ok but all I get is 1 LED lit white. I'd guess I'm on the latest Photon firmware as it just spent a while updating?...

eklex, 2015-10-22:
@causticforms Try to downgrade your Photon to 0.4.6. I had some issues yesterday when they pushed the new release.
I think it's also worth to mention that I tested the library with WS2812B strips only which are not controled by the SPI. So if you're using the SPI, it may not work (as mentioned by @focalintent).

focalintent, 2015-10-22:
Actually they should work - fast led falls back to bit banging pins if it can't find spi hardware - and if you can talk to ws2812's then the pin library is working :)

MarkEMarkEMark, 2015-10-25:
Hi Eklex, thanks for taking the initiative with this. I am trying to add your library to my new Photon. Please how do I add all the library files? (I know my question isn't specific to your library, sorry.) All I can see is adding a file at a time, and pasting in each of your file's code one at a time. Is there an easier way? I even tried Particle Dev (the offline version), but was frustrated as all the files seem to want to be in the same folder, but the library has sub-folders. I am used to Arduinos and FastLED, so I'm not a total thicky, but would really appreciate a hand-up. Thanks

keavin7, 2015-10-25:
@MarkEMarkEMark The Particle Dev takes a bit of getting used to, but overall Its starting to grow on me!

One key caveat of the IDE is that you have to have all of the files in one project directory. You can specify which items get compiled by following this discussion right mere: https://community.particle.io/t/selecting-which-file-gets-compiled-in-particle-dev/14016

MrRocketman, 2015-10-26:
Tested on 0.4.7 with my [E1.31 code](https://github.com/MrRocketman/E131PhotonPixel) (pre-alpha code) and everything is working great so far! Thank you so much!! Christmas Lights here I come!

MarkEMarkEMark, 2015-10-26:
Hi Keavin7. Thanks for the response and encouragement towards P Dev.

So all the subfolders of eklex's git need to be flattened into one folder? Then I need to search out any paths in the declarations too? I tried that, but there doesn't seem to be a clue as to where to find everything.

May I ask, how does Eklex use his library untouched? Does he not use P Dev? If not, how do you import your own libraries?

Sorry for all the questions - this is frustrating to me, as I just want to get going :-)

Thanks.

keavin7, 2015-10-26:
@MarkEMarkEMark in all honest I'm pretty new to the IDE. I was wondering about the issue of nested files as well... judging by @MrRocketman 's link you have to put everything in the same directory.... think he may have done that for you X)

https://github.com/MrRocketman/E131PhotonPixel

I've yet to try this... deadlines. looking forward to it though!

jasoncoon, 2015-10-26:
You just need to copy the contents of the [firmware](https://github.com/eklex/FastLED-Sparkcore/tree/master/firmware) directory into your sketch directory.  You should have your .ino file, any .cpp, .h, etc files, and then all of the FastLED files such as FastLED.h, FastLED.cpp, lib8tion.cpp, etc.  You'll also have the platforms directory.  You do not need to flatten the contents of the platforms directory.

jasoncoon, 2015-10-26:
Finally got around to testing your changes, @eklex, and everything seems to work fine on the Photon.  Thanks!

eklex, 2015-10-27:
Sorry @MarkEMarkEMark, I just saw your comment. As suggested by @jasoncoon and @keavin7, you don't need to flatten the library. Personnaly, I copy the **firmware** directory and rename it **FastLED**, then I remove the **docs** and **examples** directories. I place my project files (.ino or .cpp) at the top level where I include the library as follow: ```#include "FastLED.h"```.
Hence, the library stays in its own directory, and you don't mix your project files with the library. This should give you something similar:
```
| FastLED
| | FastLED.cpp
| | FastLED.h
| | fastled_arm_stm32.h
| | fastled_config.h
| | fastpin.h
| | fastpin_arm_stm32.h
| | ...
| strips.cpp
| strips.h
```
Once I pushed my changes into the [master branch](https://github.com/focalintent/FastLED-Sparkcore) (which is managed by @focalintent), you will be able to enjoy FastLED and Photon from the Particle Web IDE.

Thanks @jasoncoon and @MrRocketman for testing the modifications.

beaubrewer, 2015-10-27:
You guys rock. FastLED + Photon + APA102 = Awesome.

MarkEMarkEMark, 2015-10-27:
Thanks Keavin7. Thanks Eklex, Thanks JasonCoon, 

I will try this later this week when I get a chance.

MarkEMarkEMark, 2015-11-01:
Hi Sorry for relying on you guys just to get up and running.

What I've done:

1. downloaded eklex FastLED zip
2. Unzipped, renamed firmware folder to just FastLED
3. removed examples / documentation folders
4. put blink.ino in FastLED folder
5. changed include to just be: #include "FastLED.h"
6. opened folder in ParticleDev
7. compiled

Thanks to you all, at least it attempts to compile now, but I just get a load of errors about the first line of any importance: "FastLED.addLeds<LPD8806, DATA_PIN, CLOCK_PIN, RGB>(leds, NUM_LEDS);"

- parse error in template argument list
- no matching function for call to 'NSFastLED::CFastLED::addLeds(NSFastLED:CRGB [1], int)'
- template argument 4 is invalid
- wrong number of template arguments (4, should be 3)

(these were hard to write down, because the error list disappears if you click anywhere else)

Please see screenshot for any info i have missed.

If I can just get blink working, I can do anything! :-)

Thanks

[image]

jasoncoon, 2015-11-01:
No problem, glad to try and help.  I get the same error when I try to use that same line.  If I remove the last template argument (color order), or change it to another color order (GRB, BRG) it compiles fine.  I think RGB is the default color order anyway.  Must be some conflict with Particle libraries.  This line compiles file:

FastLED.addLeds<LPD8806, DATA_PIN, CLOCK_PIN>(leds, NUM_LEDS);

MarkEMarkEMark, 2015-11-01:
Thanks Jason. Edging closer - now I get "build didn't produce binary" error. Do you know how to see proper errors? Thanks again!

jasoncoon, 2015-11-01:
Hrm, not sure on that one, but a quick Google search turned up this: https://community.particle.io/t/particle-dev-build-didnt-produce-binary/14809/13

jasoncoon, 2015-11-01:
I can't see the top of your sketch, but do you have these two lines at the top?

jasoncoon, 2015-11-01:
Here's a working Blink example:

https://github.com/jasoncoon/FastLED-Photon-Blink-Example

Thanks to the tip from eklex, it's much easier/cleaner to include FastLED in a sub-directory.  I'm not sure why, but you don't need to include the directory name in the #include statement.

MarkEMarkEMark, 2015-11-01:
Hi Jason - thank you so much! That compiles! I did have the lines you suggested, but I had my .ino in the FastLED folder (I misunderstood about that). Now to get cracking with my Xmas lights!

(PS - checkout my android app, which is a wallpaper but which will be a controller for real LED lights this year... I hope... 

https://play.google.com/store/apps/details?id=uk.co.kicktechnic.christmaslights2014lwp

thanks again!)

al-zonke, 2015-11-02:
Great Thanks!
Works for APA102.

causticforms, 2015-11-02:
Just been trying again and with the help of this discussion I've done it! Thank you!

So just to really simplify it, what I did to get it working on my WS2812B strip was:

Install firmware 0.4.6 on my photon.
Download jasoncoons blink example.
Change the line FastLED.addLeds<WS2812B, DATA_PIN, **RGB**>(leds, NUM_LEDS); FastLED.addLeds<WS2812B, DATA_PIN, **GRB**>(leds, NUM_LEDS);

boy-blunder, 2015-11-08:
Just want to say thanks to eklex as I have tested the updated Fastled with Photon and APA102 leds and all appears to work well.

Of course it goes without saying, thanks to Daniel Garcia for creating, updating and maintaining Fastled.

benseitz, 2015-11-09:
When using der Blink-Example with a LPD8806 (or WS2812B) and uncommenting line 32 or 36 there are 13 Errors beginning with "Parse error in template argument list". 
When compiling for the Neopixel it succeeds.

Anyone knows where my problem is?

EmileAlberts, 2015-11-12:
Hi guys

I have been stuck with this problem that you seemed to solve but I cant get to solve it. I am using the Particle Build web IDE. When I include the FastLED library and compile the simple Blink program I still get the following error:

FastLED/led_sysdefs.h:19:43: fatal error: platforms/avr/led_sysdefs_avr.h: No such file or directory
 #include "platforms/avr/led_sysdefs_avr.h"

I have in also add the line after inlcude:

#include "FastLED.h"
FASTLED_USING_NAMESPACE;

But I still get the error. Can anyone help me out please?

causticforms, 2015-11-12:
EmileAlberts: for the moment I'm reasonably sure this will not work through the web IDE. You have to use the particle dev IDE.

EmileAlberts, 2015-11-12:
Thanks. I downloaded Particle Dev IDE and it compiles there successfully.

lessavyfav, 2015-11-17:
Hi there, a bit new here, eagerly hoping to get my APA102 + photon project running FastLED rather then Adafruit before my launch date. @al-zonke with Adafruit I used D4 and D4
#define DATAPIN   D5
#define CLOCKPIN  D4

In fastLED I tried:
#define DATA_PIN D5
#define CLOCK_PIN D4
and
#define DATA_PIN 5
#define CLOCK_PIN 4

for setup I've tried:
FastLED.addLeds<P9813, DATA_PIN, CLOCK_PIN, RGB>(leds, NUM_LEDS);
and
FastLED.addLeds<APA102, DATA_PIN, CLOCK_PIN, GRB>(leds, NUM_LEDS);
and both of the above with GRB

I can't get a blink...

Should I be using different pins? Any advice?

lessavyfav, 2015-11-17:
Sorry, got it running! 

used:
#define DATA_PIN 0
#define CLOCK_PIN 1

and
FastLED.addLeds<APA102, DATA_PIN, CLOCK_PIN, GRB>(leds, NUM_LEDS);

GRB!

Now to figure how to get CRGB colors mapped right, fastLED docs here I come!

ScruffR, 2015-11-23:
Would be nice if a working version could be pushed to Particle Build (Web IDE) too.

lessavyfav, 2015-12-01:
@jasoncoon or anyone else suddenly unable to get Photon to run this now? I wonder if it is a 0.4.7 vs 0.4.6 firmware thing...

ScruffR, 2015-12-01:
There is an updated fork here
 https://github.com/focalintent/FastLED-Sparkcore

It's just not yet pushed to Particle Build Web IDE, but the fixes are not too hard to do yourself either.

focalintent, 2015-12-08:
Pushed to the particle ide - thanks again, @eklex for doing the kicks!

closed 2015-12-08 by focalintent

beaubrewer, 2015-12-15:
Has anyone been successful in driving a large number of the APA102s via the Photon? I've been able to get it working perfectly on a 1 meter strip but scaling up (210 ft) has been a challenge. I have a 5v 350W power supply that I was able to modify and get the Photon inside. In the end I have the Photon inside, powered by the power supply and the clock and data line running out.

I can get the demo FastLED program working, but the leds have a (what appears to be somewhat ransom) flicker of red. I'm wondering if I've missed something basic.

I looked at the Photons pin out diagram and realized the SPI pins are A3 (clock) and A4 (data). I'm using D1 (clock) and D0 (data) at the moment. Should I bet using the SPI pins? I guess you just use the CLK, MOSI, and ignore the MISO, and SS?

The next thought I had was the Photon is 3v3 logic level and the APA102 is 5v. Maybe I need a level shifter?

Lastly, it could be some kind of interference with the power supply or back scatter from the lines. Maybe it needs to be terminated?

I would love some direction or thoughts from any of the electronic gurus out there.

focalintent, 2015-12-15:
See my comments about this elsewhere about 15 minutes ago: https://github.com/focalintent/FastLED-Sparkcore/issues/2#issuecomment-164650934 and  https://github.com/focalintent/FastLED-Sparkcore/issues/2#issuecomment-164651316

(including please don't use issues for divergent conversations, especially not closed issues)