FastLED/FastLED #167: TM1803 support doesnt appear to be working?

webdevbrian, 2015-05-20:
Hi, 

Testing things out with a radioshack 1m T1803 chip strip -- here is my code:

#include "FastLED.h"

// How many leds in your strip?
#define NUM_LEDS 30

#define DATA_PIN A0

// Define the array of leds
CRGB leds[NUM_LEDS];

void setup() { 
  // Uncomment/edit one of the following lines for your leds arrangement.
   FastLED.addLeds<TM1803, DATA_PIN, RGB>(leds, NUM_LEDS);
}

void loop() { 
  // Turn the LED on, then pause
  leds[0] = CRGB::Red;
  FastLED.show();
  delay(500);
  // Now turn the LED off, then pause
  leds[0] = CRGB::Black;
  FastLED.show();
  delay(500);
}

Strip just illuminates blue, doesn't do anything else ... any ideas?

focalintent, 2015-05-20:
What version of the library, what version of arduino, what is are you running arduino on?

webdevbrian, 2015-05-20:
Latest available library -- nano -- I dont understand the last question

focalintent, 2015-05-20:
Sorry, hit by autocorrect.  What version of the arduino _software_ and what operating system (windows, OS X, Linux, etc...)

webdevbrian, 2015-05-20:
Ah, the evil autocorrect!

Arduino 1.6.4 -- OSX Yosemite 10.10.3

focalintent, 2015-05-20:
Ok, there's some reported issues with the 1.6.4 compiler that I haven't had the time to dig into (short version, the gcc compiler is eating up the asm code for those strips).  There was a comment in, in think it was #164 (can't easily check at the moment, on phone) that had a possible fix (changing compiler flags, but that points me at some code changes I can make).

webdevbrian, 2015-05-20:
Gotcha -- would a work around be just using a different compiler?

webdevbrian, 2015-05-20:
Edit nevermind -- I guess everything obviously needs the IDE to compile, duh.

Let me know if you submit a tweak or something

webdevbrian, 2015-05-20:
FYI This issue apparently goes back until the first 1.6 version -- I can't get the FastLED lib to work with anything prior to 1.6 (like the 1.5 betas) ... so right now anyone using the TM1803's are SOL (and apparently have been for quite some time ... ?)

focalintent, 2015-05-20:
What's happening with the 1.5's for you?  I've done a lot of work w/1.5.8 (it's what I use for anything non-teensy 3 related, though now that teensyduino supports 1.6.x, I'm going to move everything to 1.6.x for testing/dev going forward) -- I haven't had a chance to do a lot of 1.6 work yet, the dayjob and life have been getting in the way (and the 1.6's were only released recently)

webdevbrian, 2015-05-20:
It said something about the fastLED lib not having special characters in the name, wouldn't include in my tests

focalintent, 2015-05-20:
Can you copy/paste the exact error message that you're getting?  That sounds really odd - and nothing that i've heard from anyone (and I have users using everything from 1.0.5 to 1.6.4 as well as unreleased betas).

webdevbrian, 2015-05-20:
http://d.pr/i/1iBFH/34r7IhjJ

focalintent, 2015-05-20:
ah - change the name of your folder to FastLED

webdevbrian, 2015-05-20:
Yeah I tried that -- thats the strange part, it still says the same thing regardless of folder name

focalintent, 2015-05-20:
do you have any other folders anywhere named FastLED-master?   (also - you have to restart arduino after changing the directory name)

webdevbrian, 2015-05-20:
Strangely enough I just tried it and it's working now (in the 1.5Beta) -- still randomly gives me the error but the library loads ... this is pretty odd....

Anyhow -- let me know if you get to figuring out the 1.6+ issue -- the 1.5beta client is pretty rough :)

focalintent, 2015-05-20:
Keep an eye on the g+ group - re-testing everything for 1.6 is one of my requirements before calling FastLED3.1 final and releasing it.  (And i'll also update tickets as well, so i'll leave this ticket open for now)

webdevbrian, 2015-05-20:
Will do - thanks for the replies Dan

DSeiford, 2015-05-29:
I also cannot get FASTLED working with the Radio Shack LED strips (TM1803) with the Arduino IDE after 1.5.7. I can get it to compile and send to the Arduino, but all the LEDs turn on bright white and never change. I found an old copy of the Arduino 1.0.1 IDE and it works fine with that. I have tried with an Uno, and a Nano. I also have an Yun, and a Wido, so I need the newer IDE for some of my boards.

Here is a sketch that works on the 1.0.1 IDE and fails on any of the newer IDEs. It's the Cylon example edited for the Radio Shack strips.

#include "FastLED.h"

// How many leds in your strip?
#define NUM_LEDS 10

// For led chips like Neopixels, which have a data line, ground, and power, you just
// need to define DATA_PIN.  For led chipsets that are SPI based (four wires - data, clock,
// ground, and power), like the LPD8806, define both DATA_PIN and CLOCK_PIN
#define DATA_PIN 12
// #define CLOCK_PIN 13

// Define the array of leds
CRGB leds[NUM_LEDS];

void setup() { 
	FastLED.addLeds<TM1803, DATA_PIN, GBR>(leds, NUM_LEDS);
}

void loop() { 
	// First slide the led in one direction
	for(int i = 0; i < NUM_LEDS; i++) {
		// Set the i'th led to red 
		leds[i] = CRGB::Red;
		// Show the leds
		FastLED.show();
		// now that we've shown the leds, reset the i'th led to black
		leds[i] = CRGB::Black;
		// Wait a little bit before we loop around and do it again
		delay(60);
	}

	// Now go in the other direction.  
	for(int i = NUM_LEDS-1; i >= 0; i--) {
		// Set the i'th led to red 
		leds[i] = CRGB::Red;
		// Show the leds
		FastLED.show();
		// now that we've shown the leds, reset the i'th led to black
		leds[i] = CRGB::Black;
		// Wait a little bit before we loop around and do it again
		delay(60);
	}
}

focalintent, 2015-05-29:
Check out what people are trying here - https://github.com/FastLED/FastLED/issues/164 - it seems to be a workaround for a bunch of folks (at least until I can figure out a way to force the compiler to behave from in the code)

focalintent, 2015-07-14:
Checked in fixes for TM1803 for avr using arduino 1.6.x.

closed 2015-07-14 by focalintent