FastLED/FastLED #171: Slower processing with multiple pins

timvancromvoirt, 2015-05-23:
Hi,

I am running just a simple code using Fastled 3.0.3. to control a single WS2812B LEDstrip (60 LEDS) on a Arduino Mega 2560, it works perfectly. 
My goal is to run 10 LEDstrips with 10 different animations on 10 different pins. 
I already wrote a code for that but the animations I programmed run very slow in comparison when using only 1 output pin and 1 animation. It seems that for each extra output pin I use that the whole of the code runs twice as slow.

I noticed that the more output pins I declare before set up (See code below) the slower the animations run, even when I only use 1 pin with 1 animation for 1 ledstrip in the loop.

My question is, is it true that the more pins you declare, the slower the whole of the script runs even when you are not using them in the void loop?

- FastLED.addLeds<WS2812B,5>(leds[0], NUM_LEDS);
- FastLED.addLeds<WS2812B,6>(leds[0], NUM_LEDS);
- etc....

Thank you in advance!

focalintent, 2015-05-23:
It takes 1.8ms to write out 60 leds worth of data.  If you call FastLED.addLeds 10 times, then it means each time you call show it will take 18ms to write out 600 leds worth of data.  (Whenever you call show it has to write out _all_ the led data, as it doesn't have any way of knowing what you have/haven't changed).

Likewise, your animation code, what you write/run for 60 leds, if you turn around and run it on 600 leds is likely to take about ten times as long to run (assuming your code scales linearly, if your code scales poorly, then it could conceivably take even longer to run).

In other words, yes, this is expected.  (The more work you do each time loop is called, the fewer times per second loop can be executed).

closed 2015-05-23 by focalintent

timvancromvoirt, 2015-05-23:
Thank you for your quick response, that's what I already thought.

I just found out I called FastLED.show(); two times in my code, don't know why I did that exactly. 
Now I use it only one time and it seems to run at a faster speed now.

Hokyjack, 2016-06-25:
@focalintent Hello, is there any way to speed it up from 1.8 ms? I am creating a POV display with use of full 144 RGB APA102 strip and arduino nano and I need to redraw all 144 pixels each lets say <3ms.
Is this possible somehow? Would teensy 3.2 do better work. Or some other strip?

focalintent, 2016-06-25:
Not with the WS2812's, no it isn't - without going to parallel output.  

If you're using APA102's you should be able to get faster rates - make sure you're using hardware SPI for writing LED data - and yeah, for POV I would suspect that a teensy 3.2 would be better for you.

romanlife, 2016-07-23:
Hi, I am creating a POV display with use of APA102 strip using ESP8266-12F (connecting GPIO13-DATA,14 GPIO - CLOCK). But when I begin to test the speed of refresh 37 LEDS APA102 - it takes about 750 us using 80MHZ ESP8266 and 450 us on 160 MHZ ESP8266. Even Arduino MEGA 2560 gives 500 us refresh for 37 APA102 leds. Is there any way to speed it up from 750 us to 150 us? When I change DATA_RATE_MHZ(24) at  - FastLED.addLeds<APA102, DATA_PIN, CLOCK_PIN, RGB, DATA_RATE_MHZ(12)>(leds, NUM_LEDS); - nothing is changing with timings of refresh?????? Is there any ideas?

focalintent, 2016-07-23:
I haven't implemented hardware spi support for the esp8266 yet.

romanlife, 2016-07-23:
Thanks for quick answer, and what about maple mini or arduino due? Can this boards work with APA102 using hardware spi support on 24 mhz??

focalintent, 2016-07-23:
I would recommend the teensy3.2 for pov work.  I don't support the maple mini.  The due also has hardware spi supports.

slootsky, 2016-07-24:
@romanlife There is a different library for ESP8266 that might work for you.   https://github.com/JoDaNl/esp8266_ws2812_i2s  (@focalintent : I hope this is ok here)

focalintent, 2016-07-24:
Except the question is about driving apa102 not ws2812's - anyone who is using ws2813's for POV is just asking for disappointment.

romanlife, 2016-07-24:
I completely agree about the ws2812 LEDs. they are not suitable for pov display. APA102 Leds are perfect, but I'd really like to have support APA102 for hardware spi esp8266 in my project.I will wait with great impatience