FastLED/FastLED #173: FastLED problem in DUE

hamsafar, 2015-05-27:
I want to use ParallelOutput with DUE. when I use this code:

42.// Set the first n leds on each strip to show which strip it is
43. for(int i = 0; i < NUM_STRIPS; i++) {
44. for(int j = 0; j <= i; j++) {
45. leds[(i*NUM_LEDS_PER_STRIP) + j] = CRGB::Red;
46. }
47. }

Color of led is not correct(Is not Red!) 
In first strip, first LED is a bit blue and the second LED is Green! and when I change line 45 to Green, the first led become red and blue and the second LED become off, and when I change line 45 to blue, the first LED become off and the second become red! I think in shift data is a problem or my DUE frequency isn't match with this library!

I use fastLed3.1 library and my Arduino version is 1.6.4 and my OS is windows7(32bit)

please help me....
Thanks﻿

focalintent, 2015-05-27:
Can you include the setup function that you are using as well?

hamsafar, 2015-05-27:
#include<FastLED.h>
#define NUM_LEDS_PER_STRIP 32
#define NUM_STRIPS 8
CRGB leds[NUM_STRIPS * NUM_LEDS_PER_STRIP];
// WS2811_PORTD: 25,26,27,28,14,15,29,11
void setup() {
   LEDS.addLeds<WS2811_PORTD,NUM_STRIPS>(leds,   NUM_LEDS_PER_STRIP).setCorrection(TypicalLEDStrip);
  LEDS.setBrightness(32);
}

void loop() {
  // Set the first n leds on each strip to show which strip it is
  for(int i = 0; i < NUM_STRIPS; i++) {
    for(int j = 0; j <= i; j++) {
      leds[(i*NUM_LEDS_PER_STRIP) + j] = CRGB::Red;
    }
  }
  LEDS.show();
  LEDS.delay(10);
}

focalintent, 2015-05-27:
I haven't tested the due's parallel output with anything other than arduino 1.5.8 yet.  What happens if you don't use the parallel output but just use individual pins?

hamsafar, 2015-05-27:
When I use individual leds don't operate correctly and blinking and don't work softly!
Before I test with avr and I have not any problem! I want to make a large display with Due and need to use parallel output(144*80 pixel - 144 column). With parallel output Due can send data to 15 output and I want each output controlling 10 column with controllable buffer. Data send with USB from a PC.
Excuse me for my bad English language.

Thanks a lot

hamsafar, 2015-05-27:
Can I use WS2811_PORTA and WS2811_PORTD simultaneously to have 15 parallel output?

focalintent, 2015-05-27:
Ok, it sounds like there's a problem with the due and arduino 1.6.x then.  I will have to spend some time debugging here to figure out what's going wrong with it.  When it's working you will be able to use PORTA and PORTD - but it won't be true parallel (the way 16-way parallel output on the teensy3.1 works) - as data will be written out in two batches.

focalintent, 2015-05-27:
However, if you are going to be doing led output driven by a PC i very very strongly recommend that you look at fadecandy - http://scanlime.org/2013/11/fadecandy-easier-tastier-and-more-creative-led-art/ - as that is optimized for PC based driving of parallel strips of WS2812's.

hamsafar, 2015-05-27:
Thanks for your help and your rapid reply.
I check it with arduino 1.5.8 and check fadecandy and search that can I find it in my country.

hamsafar, 2015-05-31:
Hi,
Excuse me, your library is OK! I use a buffer to change level of DUE output to 5 volt, then my problem solved. I test it in arduino 1.5.8 and 1.6.4.
Thanks a lot

hamsafar, 2015-05-31:
Excuse me I have problem when use WS2811_PORTA and WS2811_PORTD simultaneously to have 15 parallel output! 
#define NUM_STRIPS 15
void setup() {
   LEDS.addLeds<WS2811_PORTA,NUM_STRIPS>(leds, NUM_LEDS_PER_STRIP);
   LEDS.addLeds<WS2811_PORTD,NUM_STRIPS>(leds, NUM_LEDS_PER_STRIP).setCorrection(TypicalLEDStrip);
  LEDS.setBrightness(32);
}

focalintent, 2015-07-14:
In your next to last comment, you indicated that things are working - but in your most recent comment, you say you have a problem - what problem are you having?  (Actually, i'm going to close this ticket to clean things up - if you are still having a problem with the due and PORTA/PORTD - open a new ticket and please, in that ticket provide more information about what kind of problem you're having).

closed 2015-07-14 by focalintent