FastLED/FastLED #174: Slow compile times under Windows (at least)

kriegsman, 2015-05-29:
I think that this is, at heart, an "Arduino" issue, but I'm opening this ticket in FastLED to collect my own notes and findings "coherently."

Under Arduino 1.6.4 on Windows something dramatically bad has happened to compilation times (much slower).  Compiling the FastLED 3.1 "Blink" sketch takes over 30 seconds.  

Building FastLED 3.1 "Blink": first build 37 seconds.  Subsequent builds 29 seconds.

This single command (executed from a command shell) takes 28 seconds by itself:
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10604 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR -I"C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" -I"C:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\standard" -I"C:\Users\mkriegsman\Documents\Arduino\libraries\FastLED31" C:\Users\MKRIEG~1\AppData\Local\Temp\build7945551212190042949.tmp\Blink.cpp -o C:\Users\MKRIEG~1\AppData\Local\Temp\build7945551212190042949.tmp\Blink.cpp.o 

For contrast, compiling a basically empty sketch takes 7 seconds for the first build, and about 1 second for subsequent builds, and this command line takes about 3 seconds:
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -Wall -Wextra -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10604 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR -I"C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" -I"C:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\standard" C:\Users\MKRIEG~1\AppData\Local\Temp\build7945551212190042949.tmp\sketch_may29b.cpp -o C:\Users\MKRIEG~1\AppData\Local\Temp\build7945551212190042949.tmp\sketch_may29b.cpp.o 

Adding the -I{FastLED} argument to the empty sketch's compile command has no effect on timing.

Adding #include "FastLED.h" to the empty sketch causes the compile time to be high again (over 25 seconds) -- even though there's no invocation of any FastLED code in the main sketch.  So it's taking... 20+ seconds to find, load, parse and process something included from FastLED.h.

"g++ -- version" = 4.8.1

So something's going on that's causing that one invocation of g++ to take 25+ seconds.  Now to figure out what, why, and what to do about it...

MattMose, 2015-05-29:
Yes! Thank you for detailing this issue Mark. I just started getting into the FastLED library about a week ago and this problem has plagued me from the start. There hasn't been as much reporting on it as I expected, so I was concerned it was something specific to my setup.

This issue is especially troublesome when uploading code to an Adafruit Trinket which gives you a 10 second window for the IDE to upload the code. I've been noting compile times with a stopwatch, then using that time - sometimes up to 2:45 - to judge when to press the bootloader button on the Trinket to start the 10 second upload window. It's been trying to say the least ;)

kriegsman, 2015-05-29:
OK, so I have at least one new data point:if you tell Arduino 1.6.4 (Windows) to set compiler warnings to "ALL", it adds the -Wextra command to the g++ command line, and ... then the compile NEVER COMPLETES.

However, if I cut the command line from the status window, and paste it into a Windows command shell, it completes normally.  

Arduino 1.6.4 (Windows) has issues.  gcc 4.8.1 may have issues, too, but Arduino 1.6.4 (Windows) definitely does.

Mac version of Arduino 1.6.4 has no such issue.

Continuing to dig; any deep and specific data points appreciated, especially A/B cases where it is bad in one case, and fine in the other.

kriegsman, 2015-05-29:
Curiouser and curiouser:

I just compiled this exact code in Ard164Win in just a few seconds.

    #include "FastLED.h"
    void setup() {}
    void loop() {}

kriegsman, 2015-05-29:
Now confirming something we've seen before: something about G++ 4.8.1 really, _really_ sucks wind when it sees 'asm' blocks, such as we use for the Neopixel/WS2811/WS2812/WS2812B three-wire LEDs.  

Compile time for FastLED "Blink" sketch using Neopixel(or similar) = 28 seconds.
Compile time for FastLED "Blink" sketch using APA102(or similar) =  3 seconds.

Lovely.  Nice work, gcc.  You've made fast code slow again.

focalintent, 2015-05-29:
Well - to be fair the clockless asm code is pretty pathological, using over 25 registers and is actually more like 150 adjacent asm blocks.

kriegsman, 2015-05-29:
More data on where the time is (or rather, isn't) going.  I can it take the FastLED 3.1 Blink sketch, and run it through the C preprocessor (g++ -E), producing what we'll call "blink_pp.cpp".  This file is self-contained, in that all the includes are already included, etc.  I made two versions: one that uses APA102 (blink_pp_102.cpp), and one that uses WS2811 (blink_pp_2811.cpp).  The preprocessor runs in a couple of seconds, and the resulting cpp files are identical except for the one line with addLeds -- as I would expect.

Now, if I compile each of the two cpp files, the time difference shows up here: 3 sec for blink_pp_102.cpp, 26 sec for blink_pp_2811.cpp.

Since all of the file-includes and path-searching is already done, what this is measuring is (I think) just the raw time to compile the one single source file and generate executable code.

A quick glance at blink_pp.cpp shows all the asm blocks and related template-generated code.  It's not a monster, but it's not "hello world", either.  I think the complexity is probably necessary to achieve what we're shooting for.

Next up I want to spend a little time letting that all sink in a bit, and see if there's anything we can do to help gcc along speed-wise.

Also, I'm still mystified that the Windows version of gcc appears to be significantly slower than the OSX version -- even when all the 'including' and path-searching and really most of the I/O overall is all factored out.

MattMose, 2015-05-30:
Mark, I'm following your updates like Breaking News. Thanks for your efforts in the hunt and keeping us up to date!

kriegsman, 2015-05-30:
Next step will be a side-by-side breakdown of OSX vs WIndows times; that'll probably be later this weekend when I have access to side-by-side machines.

MattMose, 2015-06-02:
OK, this doesn't solve the heart of the slow compile time issue, but it offers a potentially permanent workaround for me. I just discovered CodeBender.cc and started using that and all of my problems were solved. A 3 minute compile on my Windows machine with local Arduino IDE install took literally 2 seconds on CodeBender. Using CodeBender also solved issue #164 for my Trinket sketches.

kriegsman, 2015-06-03:
My next goal is to see if I can figure out what's the main thing that affects compilation time.  I don't think we can easily change much, but it'll be good to understand what affects what.  Candidates in my head all relate to asm blocks, but I wonder what makes a difference.  E.g. lots of individual blocks? many blocks in a single C function? lots of args?  "volatile"?    Obviously, we can't make changes that break the functionality of the code, and we definitely also don't want to degrade the maintainability of the code, but if we understand which are the dominating factors in compilation time, we can make more informed decisions going forward.

baslentfert, 2015-06-08:
have the same issue here, somehow the compilation never finished, don't know what caused it.

kriegsman, 2015-06-08:
I had THAT problem when the compiler warnings were set to the maximum.

kriegsman, 2015-06-13:
After more digging, some more learnings about what gcc is fast at and what it's slow at:

Fast:
- preprocessor is quick, so #define's are cheap
- templates are quick, so using tons of templates is fine and doesn't really affect compilation speed
- many asm blocks in one procedure seems quick

Slow:
- gcc chokes on speed with something about long argument lists to asm blocks. Without reading the gcc code (nope), I can't be sure, but it seems like there's something in gcc that's N^2 or worse for asm block arg lists?

I've tinkered a bit and as we ready knew, there's no one quick easy answer, because fundamentally (1) we need exact cycle-by-cycle control of instructions, (2) there are benefits of using both C and asm, so 'mixed' is the best language, (3) gcc offers virtually no reliable control of how variables  move in and out of specific registers, and (4) given even a tiny bit of leeway, gcc generates code that adds extra instructions, borks the timing, and basically "screws the pooch". The only reliable tool we have to combat this are long, explicit asm variable parameter lists -- which gcc is wicked slow on. 

Still thinking about this, and at least now we know where the slowdown comes from inside gcc.

kriegsman, 2015-06-14:
It looks like if you remove _unused_ "use" vars from the asm param list, the resulting output machine code is unchanged, as you would hope and expect.

HOWEVER, if you remove _unused_ "write" vars from the asm parameter list, the generated machine code changes, and gcc breaks your toys.

This is going to come down to finding if there's ANYTHING we can do that keeps gcc toeing the line without making the code any harder to maintainable.  As it is, there's some pretty serious magic going on in there, and keeping that code maintainable is going to be the highest priority.

jbagg22, 2015-06-14:
Another data point: I'm running Mac Arduino 1.6.4 and latest FastLED (3.0.3) with a strand of NeoPixels. Compile time for the super simple Quick Example on the Documentation/Overview page took about 30 seconds. I can't get a more complicated program I wrote to compile at all. This more complicated program compiles in a few seconds sans FastLED.

focalintent, 2015-06-14:
@jbagg this issue is for tracking a windows compile time issue. If you are having a build issue on the Mac open a new issue (with more details). Also use the fastled3.1 branch for arduino 1.6.4.

NicoHood, 2015-06-15:
I want to note that non spi chipsets like ws2811 also compile slow under linux (the default 14.04 ubuntu x64) for me. So its probably a general problem.

kriegsman, 2015-06-15:
OK, I think the short version here is that "gcc is surprisingly slow at compiling some things" and there's not much we can do about it at this point other than report it as a gcc bug... and wait.  Frustrating, but it's clearly an serious performance issue _inside_ gcc.

I'm closing this ticket, and leaving it here for notes, but I'm not going to pursue it any more here at this point.  Filing it under "S", for "the Serenity to accept the things I cannot change", and "gcc Sometimes just Sucks".

closed 2015-06-15 by kriegsman

focalintent, 2015-07-15:
You're welcome - https://github.com/FastLED/FastLED/commit/e74caebaab54ffa82b4c8393fcde02d5da05edd2 (and the speedup may actually be closer to 2-4x )