FastLED/FastLED #177: Strobe function

macnl, 2015-06-04:
I think the fastLED lib is realy cool.
But i asked myself.. wouldent it be cool to also have some strobe effects??
That would be realy awsome!!
Its just a thought..

NicoHood, 2015-06-05:
The timings would be more interesting since it is a simple blink. I tried this a year ago and had not a good result. Not enough that I really liked it. But that is not an issue about FastLed. You should better ask at the google group. If someone has a good timing with a good example that'd be interesting though.

macnl, 2015-06-06:
why i wrote this.. is just that it would be great to see its intergrated IN fastled.. 
Since they already did so mutch magic.. mayby it issent that hard for them to intergrade it.

focalintent, 2015-07-14:
@kriegsman just shows off some strobe example code in the g+ community - see https://plus.google.com/112916219338292742137/posts/8GjYbFGyBGM and https://gist.github.com/kriegsman/e9d2e153af7bf4770d5c

closed 2015-07-14 by focalintent