FastLED/FastLED #178: Possible endframe issue with APA102

dmf3030, 2015-06-07:
Using the newest 3.1 library with a strip of 144 led strip APA102.  I've got a simple loop flipping all LEDs to red then all to blue.  When I declare 144 leds at start I get the last three leds flipping the wrong color each cycle.  When all the leds are red, the last leds are blue and vice versa.  If I only declare 32 LEDs I only get one LED flipping the wrong color.  I checked my code and pasted it below, I dont think theres any issue.

I also did some reading and it looks like there was some mystery around the APA102 endframe previously, but now it known that it needs to be long enough to push all the data out to the leds and that if it isnt long enough then the last bit of data might be stored and pushed in the next update.  Not sure if this is pertinent to the issue and I'll keep digging at it.  Here is a link to the APA102 info, specific endframe info at the bottom.

https://cpldcpu.wordpress.com/2014/11/30/understanding-the-apa102-superled/

My code for toggling the LEDs from red to blue, seems clean to me and not sure why the end few LEDs would be the wrong color, nor why more are wrong the more LEDs I declare.

void loop() {
      
        for(int x = 0; x < NUM_LEDS; x++) { 
            leds[x].setRGB(200,0,0);
        }
        
        delay(1000);
        FastLED.show();
        
        for(int y = 0; y < NUM_LEDS; y++) { 
            leds[y].setRGB(0,0,200);
        }
        
        delay(1000);
        FastLED.show();
     }

focalintent, 2015-06-07:
Weird - I've tested the end framing code with up to 768 leds without problems, and the size of the endframe is based on that article.  I'll look again.

dmf3030, 2015-06-07:
I'll look again and do some more testing to see if its something on my side and then update if I find out its just my implementation.

focalintent, 2015-06-07:
Your code looks good - also my endframe code should be toggling the clock 32 times for every 64 LEDs.  I might increase that number in a just to be safe sort of way.  (Eg a full clock toggle per led, vs half of one for each).

focalintent, 2015-06-07:
Try grabbing the latest FastLED3.1 - I doubled the number of endframe clock strobes that happen, let me know if that helps.  (Made it 1 per led instead of half per led).

dmf3030, 2015-06-07:
Thats seems to have done the trick, same code now is correctly toggling all the leds between red and blue without the last few being out of sync like before.  Thanks for the quick fix.

focalintent, 2015-06-07:
Excellent!

closed 2015-06-07 by focalintent

DavidEGrayson, 2015-06-24:
It looks like this is the commit where you fixed the problem: https://github.com/FastLED/FastLED/commit/3b13da9896face3f6cda41d876882d604870505f

The problem with the earlier code you had is that integer division in C++ rounds down by default, but you really want it to round up.  I think the C++ expression for the correct number of end frames to send is `(nLeds + 63)/64`.  For 144 LEDs, this means we should send 3 end frames, which would be 12 bytes.

focalintent, 2015-06-24:
I know how C++ does math, this is why I used a do/while loop intentionally.  Let's work through the math (since you are the N'th person I've had to point this out and explain it to).  144 / 64 is 2.25, nBytes will be 2.  The first time through the doWhile loop, we write out 4 bytes.  Then we get to the while(nBytes--);, since nBytes is 2, this value is 2, and we go through the loop again (decrementing nBytes, so it is now 1) writing out 4 more bytes.  Now we get to the while(nBytes--) again, and since the value is 1, we will go through the loop again (decrementing nBytes, so it is now 0) writing out 4 more bytes - therefore, 12 bytes is written out.

I think the actual problem is that the endframe spec/count is guessed at, not actually documented.  I've effectively doubled the size of the endframe to hopefully account for that (effectively making it 1 bit of end frame per led, instead of the half bit of end frame per led that initial documentation hints at).

DavidEGrayson, 2015-06-24:
Oops, you're right, I didn't read that while loop correctly.   `nBytes` really holds the number of frames you are planning to send minus one.