FastLED/FastLED #179: MBED support - NRF51822 etc

dmf3030, 2015-06-07:
Saw there might be work on either the NRF51822 platform or possibly mbed.  Want to put my vote in for mbed especially because it now has support for the NRF51822 and a wide variety of STM and other chips, by porting the library to an mbed native you would add a bunch of new chip support in one go.

Thanks for the great library.

focalintent, 2015-06-07:
There is already nrf51822 support.  Making this an mbed library would not automatically get me more chipset support 
- [ ]

focalintent, 2015-06-07:
The library is too low level for that - supporting more chipsets requires doing low level io implementations.

dmf3030, 2015-06-07:
Understood and that makes sense.  With existing nrf support I'll look at pulling fastled into an mbed project im working on based on the nrf and see if I can get it to work.

shlomozippel, 2016-01-26:
I got FastLED working on the nrf51822 using the RedBearLabs NRF arduino core which has mbed support: https://github.com/RedBearLab/nRF51822-Arduino

Note: I am using a module that doesn't have an external oscillator. If you are too then you'll have to remove the code that enables it from nrf_system.c