FastLED/FastLED #180: Better Chipset overview in examples

NicoHood, 2015-06-07:
In the blink example APA102 is not listed. It would be better to add all possible chipsets here so the user does not have to guess the names and see what is available. Also I'd rather use a fill all function so the user quickly can test all leds. I always have to look this up or write a loop for a quick'n dirty test of all leds.

https://github.com/FastLED/FastLED/blob/FastLED3.1/examples/Blink/Blink.ino

But I also like the order as in the fire2012 example which is way better to view.
That is how I currently do it:
```cpp
// FastLED
#include "FastLED.h"

#define LED_PINS    MOSI, SCK // DATA_PIN, or DATA_PIN, CLOCK_PIN
#define COLOR_ORDER GRB
#define CHIPSET     WS2801 // WS2811 LPD8806
#define NUM_LEDS    32

#define BRIGHTNESS  255  // reduce power consumption
#define DITHER      255  // try 0 to disable flickering
#define CORRECTION  TypicalLEDStrip

CRGB leds[NUM_LEDS]; // Define the array of leds

void setup() {
  // FastLED setup
  FastLED.addLeds<CHIPSET, LED_PINS, COLOR_ORDER>(leds, NUM_LEDS).setCorrection(CORRECTION);
  FastLED.setBrightness( BRIGHTNESS );
  FastLED.setDither(DITHER);
  FastLED.show(); // needed to reset leds to zero
}
```

I'd make it a bit more consistent in all exampled except the blink of course.

focalintent, 2015-06-07:
The problem with the setup in Fire2012 is it doesn't account for different chipsets needing different numbers of arguments, which is why the blink and first light examples are _very intentionally_ not using #define's for everything.  

I'll add APA102 to the examples, but the purpose of blink is to show off blinking an led, so I'm not going to change that.  The purpose of FirstLight is to show off manipulating all the leds and also conveniently behaves as a test of all the leds in your strip.

focalintent, 2015-06-07:
(And yes, I know you are abusing a quirk with #define's to make LED_PINS be either a single or double argument thing and guess what - that's a kind of advanced use/trick that I don't want to have to support questions about from new coders - there's a lot of twisted code tricks that I could be doing in the example code that would end up confusing more than educating).

closed 2015-06-07 by focalintent

NicoHood, 2015-06-07:
Sure.
At least add the missing chipsets (dont know if its more than just apa)

focalintent, 2015-06-07:
Already done (added apa102, apa104 and dot star definitions)