FastLED/FastLED #183: Black is green on TM1803

jasonturcotte, 2015-07-05:
I'm having a "black as green" issue. Using a TM1803 30 LED strip.

All colors appear to be working ok, except black. For example, if I try:

leds[1] = CRGB(255,0,0);

then only the red LED is lit. If I try:

leds[1] = CRGB(0,0,255);

then only the blue is lit. However:

leds[1] = CRGB(0,0,0); results in the green LED being lit very faintly.

Thoughts?

To add one more oddity to the mix, the first segment of LEDs works fine, meaning all pixels are off and true "black".﻿

jasonturcotte, 2015-07-05:
Ok. I shot a quick video of what's going on. I figured that would help.

I also created a test sketch that does nothing but set the strip to black and does that every .5 seconds.

Here's all the code:

```
#include <FastLED.h>
#include <SPI.h>

//strip info
#define NUM_LEDS     30
#define CHIPSET      TM1803
#define COLOR_ORDER  RBG
CRGB leds[NUM_LEDS];

#define DATA_PIN     5

void setup() {
  // power-up safety delay
  delay(3000);
  FastLED.addLeds<TM1803, DATA_PIN,RBG>(leds, NUM_LEDS);
  
  FastLED.setBrightness(64);
}

void loop() {
  FastLED.showColor(CRGB::Black);
  delay(500);
}
```

Link to the video: https://www.youtube.com/watch?v=IRzlZFW_cek

focalintent, 2015-07-08:
What version of the arduino software are you using, and what platform are you building on?  Green being stuck on usually means an issue of timing with the first of the 24 bits written out for each rgb led, which most often is because gcc is doing something stupid with my asm code.  I think there's another round of bugs/issues that I need to work around somehow to keep this from happening.

focalintent, 2015-07-14:
Again - what are you running this on, a trinket, a teensy 3, a leonardo?  All the platforms have their own custom code for handling these chipsets, and without knowing what you're building for - there really isn't a good way for me to test.  (I asked for this almost a week ago, if I don't hear anything in the next couple of days, i'm going to close this).

focalintent, 2015-07-14:
Checked in fixes for TM1803 for avr using arduino 1.6.x.

closed 2015-07-14 by focalintent

jasonturcotte, 2015-07-14:
Sorry... using it on an UNO R3 and version 1.6.7 of the IDE.

NicoHood, 2015-07-14:
There is no such version 1.6.7 yet. The current dev version is 1.6.6
Do not use the arduino.org IDE, they are just fake. You can read up on this on some online sites like hackaday.
Grab the latest IDE from arduino.cc

jasonturcotte, 2015-07-14:
I was using 1.7.5 from arduino.org.

Just installed 1.6.5 from arduino.cc.

jasonturcotte, 2015-07-16:
Just updated to 3.1 and the issue seems to be gone at this time.