FastLED/FastLED #185: Leds died with FastLed 3.1

NicoHood, 2015-07-11:
I downloaded the newest 3.1 Branch today and it seems my Leds died. I guess it is not a software problem, but just in case I want to mention and ask here for help.

So I used the nightly IDE of today with Ubuntu 64 bit, FastLed 3.1 of today, ws2801 with SPI pins and 9 leds, Arduino Uno and Nano. The GND of the Uno was also connected with the GND of the audio signal due to music visualization with my uno.

I did not exceed the 9 Leds but even if I did that shouldnt matter. Leds were connected to the PC via USB Cable and got Power from 3A 5V Power supply. The problem might be located in the power supply. I recently bought it from china and never used it before. It said it can be used for Leds (because other PSUs make the leds flicker). And it worked good so far. But then I noticed that the last Led flickers randomly. Then it turned off. Then I noticed that some leds wont work if I replug the Power cable. Also middle Leds turned off.

I then moved to my USB Arduino power supply (9 leds should work) and I got the same error. In case I am totally failing at programming (I used a blink) I grabbed my other Arduino which had a working RGB blink sketch and tested that. The last 2 leds are now turned off. With the PSU, with the arduino itself and without the audio gnd. And I cant get them lightning again (last 2 leds).

The PSU hat about 4.5V measured (with 3m extension cable), USB 4.8V.
But why are the leds dying and why so slow? Maybe the PSU outputs peeks?

Any suggestion how to fix the problem or at least find out whats the problem here?

NicoHood, 2015-07-11:
Found out that the PSU gives 7V if it has no load. Meaning blinking the leds to the music made them go 5-7V and die slowly.

Any suggestions for better PSUs with a normal plastic case? Any help appreciated.

focalintent, 2015-07-11:
Better to ask the psu question on the Google group.  (I usually just measure the voltage with no old to be safe, personally)

NicoHood, 2015-07-12:
Don't and won't have a Google Account.

I think I'll try this adafruit PSU next:
http://www.adafruit.com/product/1466

Maybe you could add some information about good PSUs in the wiki? How to use an ATX for example and what is important to avoid flickering or dead leds.

kasperkamperman, 2015-07-12:
I think you can read without a Google account. To calculate amperage us 60mA per RGB led. So 66 leds on this 4A PSU (at least when you make them bright and white).

broekrh, 2015-07-13:
I had exactly the same problem. I used an external power supply and my leds started flickering, and after a while did not burn at all. Turned out that I forgot to connect ground coming from the external power supply to a ground pin on the Arduino. After I corrected this, the leds came alive again :-)

focalintent, 2015-07-14:
Opened #186 to track adding power notes into the wiki, closing this because the library isn't killing your LEDs, you are :)

closed 2015-07-14 by focalintent