FastLED/FastLED #186: Collect power supply best practices into the wiki

focalintent, 2015-07-14:
Re: #185 - collect information about power supplies into the wiki for folks.  This may just be a set of links to articles at adafruit and other places on power supplies to use when working with leds.  (For folks who are unwilling to use the g+ community to ask what people are using).

NicoHood, 2015-10-03:
Just to let you know the end of my story:

I now bought both Adafruit power supplied, 4A and 10A. Both work as expected. They output very clean 5V. The 10A runs at a bit higher voltage (if no led is on) which is still totally okay for a 10A PSU. (talking about 5.2 V or so).

The 4A PSU has only a 2 pin plug and the 10A version a 3 pin plug (for ground etc. Dont know the english name). I had problems with other PSUs (even with different ATX PSUs) that my PC Audio makes very high noises when the GND of the Leds are connected with my Arduino (which is connected to the PC which is connected to the Music boxes). With both PSUs its gone.

My WS2801 Leds consume ~3.7V max with 50leds. I use 25 for my ambilight to I can even run a raspberry pi with this. I have not tested a raspi yet though (which is also very picky about a good PSU).

The conclusion is to just grab a good PSU before (slowly and silent) damaging your hardware. First you dont notice it, but then weird errors occur and they die. Its also worth for general safety since we are talking about 230V on the other end. Quality, you know.

Links:
https://www.adafruit.com/products/1466
http://www.adafruit.com/product/658

marmilicious, 2023-08-30:
The FastLED Reddit group also has a wiki with links to videos about power stuff and people are welcome to ask questions there.