FastLED/FastLED #188: Interest in porting FastLED to Cypress PSoC "Components"

RayHAz, 2015-07-15:
Is anyone interested in porting FastLED to a Cypress component?

I've recently started working with Cypress's PSoC devices, especially since the CY8CKIT-059 is available for $10 (M3 CPU; the M0-based CY8CKIT-049 is $4). The Cypress PSoC Creator is their IDE, and since the "P" in "PSoc" refers to programmable digital and analog blocks on-chip, the IDE has some graphical programming in terms of drag-and-drop components which allow you to create a wide range of "circuitry" within the chip without having to touch a soldering iron.

There is an existing PSoC component for WS281x strings, but I desire to interface to SM16716 strings.  I got to thinking it might be easier to just refactor the existing FastLED code over to Cypress components, and then provide the ability to interface to many LED controllers.  The additional functionality (beyond just interfacing to LED hardware) would also be a nice touch, and I think usable by a larger community.

I'd be willing to do much (or all) of the work... but if there's interest, I'd love to collaborate.

I'll make a similar post in the Cypress forums.

focalintent, 2015-07-15:
I have a cypress board here to play with - but there's a lot of platforms on my list to port to (lpc8xx for a contract job, nrf52, the stm32 used in the photon, a handful of m0 cores), not to mention led chipsets.  So, there's interest, it's just a question of time/priorities.

RayHAz, 2015-07-15:
Thanks... I know you're busy (who isn't these days!), and mostly put this out there to attract other Cypress users who might be interested.  (My attempt to post on Cypress forum was auto-rejected... apparently my Cypress cred isn't worth beans yet. *sigh*)

Like I said... I'd be happy to do the work.  Unfortunately I’m actually not that great with C or C++… I can do them, but it’s not easy for me like it is the younger guys.  (Years and years of Microsoft and Visual Basic… and at some point, Pascal and Ada – and Fortran and COBOL before that!)  If you can be available to answer the occasional question ("this code LOOKS like it's prepping a PWM to run at duty cycle X... but I'm not sure" sort of thing), then I'm happy to trade you some insight into Cypress PSoC. 

I've also used STMicro in the last 5-6 months (moving servos, flashing single LEDs in coordination with a Windows app), so might be able to help out there as well.  I'm probably moving all my device work over to Cypress though... I like their approach to devices, inexpensive development/reference kits, and the Digital Block approach to the flexible hardware.

As I tell folks, I'm semi-retired, and busier than I ever was when I actually "worked" for a living -- but willing to help if I can.  -> http://LinkedIn.com/in/RayHarwood

focalintent, 2015-07-15:
As far as interest goes - I believe you are exactly the second person to have asked me about the cyprus :)

RayHAz, 2015-07-17:
Fair enough.  There has been a little discussion over on the Cypress forums... but apparently I'm still lacking sufficient cred to post there. *sigh*

I opened a case with the Cypress support folks, hoping to spark an interest in broadening the line of LED strings easily integrated into the PSoC line. I suggested that FastLED might be a good resource, and with minimal effort they might be able to attract another market segment into their product line.

"It's always about the money".  -- Ray Harwood

Which of course doesn't preclude me from doing it myself, if I get motivated to do so.  And I found step-by-step instructions on integrating GCC into the PSoC Creator GUI.  At some point it'll be "too easy to pass up" and I'll lock myself away until it's done.

zackees, 2026-06-03:
## fbuild assessment (tracker: FastLED/fbuild#392)

**Status: not feasible at this time. Closing fbuild#392 as `not planned`.**

### fbuild backend
- **Board metadata**: no `psoc`/`cypress`/`cy8c*` entries in `crates/fbuild-config/assets/boards/json/`.
- **Build backend**: no `psoc`/`cypress` references anywhere in `crates/fbuild-build/` or the rest of the workspace.
- **Test fixture**: no `tests/platform/psoc*` or `cypress*`.
- **External lookup** (`ci/board_sources.py --search psoc|cy8c|cypress`): no matches across Arduino official, ESP32, ESP8266, Adafruit, SparkFun, Teensy, or Zephyr package indices.

### Upstream toolchain story
- Cypress/Infineon PSoC parts (PSoC 4/5LP/6) are developed against **PSoC Creator** or **ModusToolbox**, with proprietary Components / HAL/PDL drivers — not a standard Arduino core. PlatformIO once had a `cypress` platform; it is unmaintained and does not provide a usable Arduino-API surface for FastLED.
- There is no Arduino-style FastLED port for PSoC in the FastLED tree, so even a perfect fbuild backend would have nothing to compile against.

### What would be required
1. An upstream Arduino-compatible core for PSoC (or a FastLED platform shim against ModusToolbox/PDL).
2. A toolchain package (arm-none-eabi-gcc is fine, but the SDK headers/linker scripts/startup code must come from Infineon).
3. fbuild board JSON + a `fbuild-build` platform module to wire the SDK build flow.
4. A smoke fixture under `tests/platform/`.

None of (1) exists in a maintained form today. Re-open when an Arduino-compatible PSoC core lands upstream and there is concrete user demand.