FastLED/FastLED #189: Variation on black is green problem

broekrh, 2015-07-15:
Hello, 
Currently I'm running into the following problem. I am working on a project in which a Arduino Mega 2560 is controlling 9 led strips/rings. They are all controlled independent with pins 33 till 41. When I attach the first 5 ledstrips, all is working fine. But after connecting the 6th strip, all strips turn green at startup. This is regardles to which pin I'm using (and I'm not using pins 6,7,8,9,14,15,16,17,42,43,44,45,46,47,48,49,62,63,64,65,66,67,68,69)

Turning them black (eg 0,0,0) won't solve the problem, they stay green. I have tried the different example programs in the FastLed library, an run into the same problem. So I suppose there's nothing wrong with my code.
* All the leds are WS2812B
* The Arduino version is 1.6.2
* All Leds are ground with Arduino, and I'm using an external 5V power source (another power source didn't solve the problem)
* I'm using 470ohm resistors on each data pin
* I'm using FastLed 3.0.3

Is this a known issue ?

focalintent, 2015-07-15:
Switch to fastled3.1 branch - and also read the info on Megas here for pin limitations: https://github.com/FastLED/FastLED/wiki/Platform-limitations

focalintent, 2015-07-15:
Ah just saw that you are already controlling for pins - grab 3.1 - 3.0.3 and arduino 1.6.x don't get along well

focalintent, 2015-07-15:
(Also, thank you for prefilling all the platform info, makes my life saner - and 3.1 will hopefully become master soon - some last testing to do on it, first)

broekrh, 2015-07-15:
Thanks !. I will try this and let you know tomorrow (going to bed now).

focalintent, 2015-07-16:
(Also note that the most latest FastLED3.1 branch makes all arduino mega pins usable)

broekrh, 2015-07-16:
Success !! Downloaded Fastled 3.1 and all strips are running perfect ! Thanks !

closed 2015-07-16 by focalintent